ev3dev-lang-java/ev3dev-lang-java #658: LOGGER variable missing in some classes?

wuqs-net, 2019-03-24:
Hello! I've forked this project and I'm trying to make some improvement to the `EV3Led` class. When I try to build and test the project, it's giving me errors saying "cannot find symbol for variable `LOGGER`" for these classes:

- `ev3dev.hardware.display.LinuxFramebuffer`
- `ev3dev.hardware.display.BitFramebuffer`
- `ev3dev.hardware.display.ImageUtils`
- `ev3dev.sensors.EV3Key`
- `ev3dev.hardware.display.RGBFramebuffer`
- `ev3dev.hardware.display.SystemDisplay`
- `ev3dev.utils.Brickman`
- `ev3dev.utils.USBPort`
- `ev3dev.hardware.display.StolenDisplay`
- `ev3dev.hardware.display.OwnedDisplay`
- `ev3dev.hardware.EV3DevPlatforms`

What should I do?

JakubVanek, 2019-03-24:
Hi,

I think it is caused by the fact that these classes use the `@Slf4j` annotation from Logback. This automatically adds the `log` field to the class. However it is necessary to enable annotation processing in your IDE.

wuqs-net, 2019-03-24:
Thank you very much! Now I can build the project without problems.

closed 2019-03-24 by wuqs-net