ev3dev-lang-java/ev3dev-lang-java #662: FR: document how fast sensor reads should be

salamanders, 2019-05-09:
How fast does your ev3 actually run? I've got what I thought was a simple "repeat(1000) { read the gyro and motor sensors, store the values in a list}" loop. Granted, written in Kotlin, on Java, on ev3dev.  But still, just a loop. And it starts out taking 180ms **per loop**!  After a few seconds it gets down to about 70ms per loop, but still... 70 is a lot!  It doesn't leave much time left for calculations.

Does yours go faster?  I assumed reading sensors should be crazy fast.  I think I eliminated extra calls (like battery voltage readings) but may have missed a slow one. 

FR: some documentation note setting expectations around standard sensor performance.  (And ideally motor commands as well)

Maybe what I'm getting is what is expected from the brick...

salamanders, 2019-05-14:
[image]

I ran some trials.  There is a very noticeable "bake in" period for the first 30 seconds where sensor reads take much more time.  Then it smooths out with a much better average.

```kotlin
println("SCENARIO $testNum: Sampling both sensors to see if the speedup is java-wide warm-up")
val testEndTime = System.currentTimeMillis() + testDurationMs
while (System.currentTimeMillis() < testEndTime) {
    Profiler.trackTime(PROFILED.MOTOR) {
        motorTachos.add(motorLeft.tachoCount)
    }
    Profiler.trackTime(PROFILED.DISTANCE) {
        distanceProvider.fetchSample(distanceSample, 0)
        distances.add(distanceSample[0])
    }
}
```

https://docs.google.com/spreadsheets/d/1n6thK3HMuHTWT5SSSP1fWjvHKKLg8NOCreKFwq0z1qQ/edit?usp=sharing

JakubVanek, 2019-05-15:
Hi, this looks suspiciously similar to what could a running JIT compiler do. I'm not sure what does it, but I would speculate about some possible causes:
* Kotlin or Java runtime (but idk, maybe it's not the case) which is doing something or is being compiled in the background,
* non-compiled java code between the function call and its return,
* some strange form of memory buffering in the garbage collector.

I am unfamiliar with profiling and Kotlin, but could you create some sort of flame graph to see where the time is lost? However the time graph is also useful, as this shows that the issue is startup-related and that it slowly fades away.

The "not useful" solution to the program latency would be to rewrite it in C++ and cache the file descriptors used for accessing the ev3dev API.

The actual solution may be to tune the  JVM. This time I'd try a flag that I previously rejected: https://stackoverflow.com/a/46516509 (-XX:TieredStopAtLevel=n). If you are using the gradle plugin, it should ve possible to add it by using `jvmFlags += ["..."] to the main block in config.gradle.

jabrena, 2019-05-21:
Pretty interesting analysis @salamanders 

Do you have the project to test in local?

Juan Antonio

JakubVanek, 2019-05-30:
If anyone wants to speed the sensor reading up, I have created https://github.com/JakubVanek/elj-nativeio. It is not ready yet, there are some remaining issues. I'm going to focus on other issues in this project, so it is probably free to be maintained by someone other.
* unified build process for both Java and C++ worlds
* runtime loading of correct native library for given architecture (armel vs armhf vs amd64)
* integration to ev3dev-lang-java and its GenericMode and other classes.

jabrena, 2019-09-02:
In the future, we will create a specific benchmark for that purpose.

closed 2019-09-02 by jabrena