ev3dev-lang-java/ev3dev-lang-java #663: Large motors work great - medium motor won't move?

salamanders, 2019-05-18:
I've been happily driving my tank around using 2 large EV3 motors.
But when I tried to add in a medium motor, it locks up when it tries to rotateTo.  I tested the motor using the standard EV3 OS, in case it was a bad motor or bad cable, and it works just fine.

Anything I can do to debug why it stops running when I issue the rotateTo command?

```kotlin
val motorSweep = EV3MediumRegulatedMotor(MotorPort.D)
logger.info { "Created motor.  ${motorSweep.maxSpeed} ${motorSweep.position} ${motorSweep.tachoCount}" }
motorSweep.resetTachoCount()
motorSweep.rotateTo(45) // Locks up here
logger.info { "Finished 45: ${motorSweep.position}" }
```

robot@ev3dev:~$ uname -r
4.14.117-ev3dev-2.3.4-ev3

JakubVanek, 2019-05-21:
I can reproduce the issue with this kernel (not tested on previous ones.
Reproducer:
```java
package issues;

import ev3dev.actuators.lego.motors.EV3MediumRegulatedMotor;
import lejos.hardware.port.MotorPort;

public class Issue663 {
    public static void main(String[] args) throws Exception {
        EV3MediumRegulatedMotor m = new EV3MediumRegulatedMotor(MotorPort.D);
        m.resetTachoCount();
        int dstAngle = 90;
        for (int i = 0; i < 10; i++) {
            m.rotateTo(dstAngle);
            dstAngle = -dstAngle;
            Thread.sleep(1000);
        }
    }
}
```

Log:
```
---> Running "time java -Xms64m -Xmx64m -XX:+UseSerialGC -noverify -jar "/home/robot/java/programs/template_project_gradle-2.5.3.jar"" <---
2019-05-21 20:38:27 [main] INFO ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
2019-05-21 20:38:29 [main] INFO ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
2019-05-21 20:38:31 [main] INFO ev3dev.actuators.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: D
```

In the background, the connected motor in `/sys/class/tacho-motor` changes from `motor1` to `motor2`.

JakubVanek, 2019-05-21:
When the motor is manually configured from command line, this happens:
```
robot@ev3dev:/sys/class/tacho-motor/motor3$ echo 45 >position_sp
robot@ev3dev:/sys/class/tacho-motor/motor3$ echo run-to-abs-pos >command
robot@ev3dev:/sys/class/tacho-motor/motor3$ cat state 
running stalled
```

`speed` and `speed_sp` attributes are both zero. Setting `speed_sp` seems to have no effect. Therefore when the library waits for the move to end, it probably gets stuck waiting for the `running` state to disappear.

JakubVanek, 2019-05-21:
I see the problem: after reset, the speed request is set to zero. With zero speed, the kernel controller indeed sees the motor as stalled. Setting the motor speed fixes the problem:
```java
package issues;

import ev3dev.actuators.lego.motors.EV3MediumRegulatedMotor;
import lejos.hardware.port.MotorPort;

public class Issue663 {
    public static void main(String[] args) throws Exception {
        EV3MediumRegulatedMotor m = new EV3MediumRegulatedMotor(MotorPort.D);
        m.resetTachoCount();
        /* here */ m.setSpeed(360); /* here */
        int dstAngle = 90;
        for (int i = 0; i < 10; i++) {
            m.rotateTo(dstAngle);
            dstAngle = -dstAngle;
            Thread.sleep(1000);
        }
    }
}
```

JakubVanek, 2019-05-21:
On the other hand, it does not directly explain the difference between the medium and large regulated motors.
@salamanders Does setting the speed solve the issue?

JakubVanek, 2019-05-21:
It seems that it is a bug in this library in the functions `rotate()` and `rotateTo()`. In functions `forward()` and `backward()`, the speed_sp attribute is configured before issuing the command. This is not done in the problematic two.

JakubVanek, 2019-05-21:
If the workaround with `m.setSpeed(360)` works, the #666 PR should be a working solution for this.

salamanders, 2019-06-16:
I think it is something intertwined with motor states between app runs.  Possibly.

I tried a simple:

    val motorCenter: RegulatedMotor = EV3MediumRegulatedMotor(MotorPort.D)
    println("isStalled:${motorCenter.isStalled}")
    println("maxSpeed:${motorCenter.maxSpeed}")
    println("speed:${motorCenter.speed}")
    println("tachoCount:${motorCenter.tachoCount}")
    println()
    println("About to 45");
    motorCenter.rotate(45, true)
    delay(250)
    println("isStalled:${motorCenter.isStalled}")
    println("maxSpeed:${motorCenter.maxSpeed}")
    println("speed:${motorCenter.speed}")
    println("tachoCount:${motorCenter.tachoCount}")
    println()

And got odd results: 

```
isStalled:false
maxSpeed:1540.4999
speed:0
tachoCount:0

About to 45
isStalled:false
maxSpeed:1531.1399
speed:0
tachoCount:35
```

Almost like it *thought* the speed was 0, but it was leftover set to 360 from the previous run.

jabrena, 2019-09-01:
Reading the whole discussion, I don´t understand the issue.
I need a summary to get an idea.

JakubVanek, 2019-09-02:
Hi,

the way I remember it is the following: it is necessary to set the ev3dev speed_sp attribute (and maybe others) before sending the movement command (like `run to rel pos`) to the kernel driver, because sometimes the speed_sp attribute gets reset to zero (probably when resetting the tacho count, because our code sends `reset` command to the kernel). Without resending the speed, speed_sp will stay at zero and after the movement command is sent, the kernel regulator will show the motor as busy forever. This is because there is no movement towards the target point - it is literally moving at zero speed.

Regards,

Jakub

jabrena, 2019-09-02:
Tomorrow, I will check this example to raise the error:

```
import ev3dev.actuators.lego.motors.EV3MediumRegulatedMotor;
import lejos.hardware.port.MotorPort;

public class Issue663 {
    public static void main(String[] args) throws Exception {
        EV3MediumRegulatedMotor m = new EV3MediumRegulatedMotor(MotorPort.D);
        m.resetTachoCount();
        int dstAngle = 90;
        for (int i = 0; i < 10; i++) {
            m.rotateTo(dstAngle);
            dstAngle = -dstAngle;
            Thread.sleep(1000);
        }
    }
}
```

And I will review the possible solution:

https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/666/files

closed 2019-12-02 by jabrena