ev3dev-lang-java/ev3dev-lang-java #665: Brake, coast, float etc. probably do not work

JakubVanek, 2019-05-21:
Hi,

the functions in `BaseRegulatedMotor` which operate on `STOP_COMMAND` probably do not operate as intended. They only set what should be done once `stop()` is called. However this is not consistent with the behavior on lejOS, where `flt()` actually performs the release of the motor pins.

Regards,

Jakub

jabrena, 2019-05-21:
Hi @JakubVanek, all stop modes is working. In LeJOS only exist flt() but the most closed behaviour in EV3Dev is coast.

closed 2019-05-21 by jabrena

JakubVanek, 2019-05-21:
Hi, the issue I see is not that `float` mode does not exist. The issue is that the specified action is executed only once stop() is called. This is changed in the #667 PR, which adds a common `doStop()` function.

commit f55343df (2019-05-21): Trigger actual stop when calling flt() and similar functions

Previously only the stop mode was configured in some functions.
This however did not trigger the stop action itself.
This commit adds a stop action execution to flt(), coast(), brake(), hold() and stop().
Also, the immediateReturn argument is supported. There is also
a change against the leJOS default - the non-parameterized
stop() and flt() calls are non-blocking now.
This is documented in the leJOS ADR.

Issue: fixes #665
PR #667 (2019-12-02): Trigger stop action when calling flt() and similar functions
Previously only the stop mode was configured in some functions. This however did not trigger the stop action itself.
This commit adds a stop action execution to flt(), coast(), brake(), hold() and stop().
Also, the immediateReturn argument is supported. There is also a change against the leJOS default - the non-parameterized stop() and flt() calls are non-blocking now. This is documented in the leJOS ADR.

Issue: fixes #665