ev3dev-lang-java/ev3dev-lang-java #668: Project status review

jabrena, 2019-05-30:


JakubVanek, 2019-05-30:
Gradle plugin
- https://github.com/ev3dev-lang-java/gradle-plugin/issues/4 + https://github.com/ev3dev-lang-java/gradle-plugin/issues/5
  - these are about changing the way dependencies are specified in Gradle, so that it is how it is recommended
  - this depends only on merging the new plugin version to the rest of the repositories
- https://github.com/ev3dev-lang-java/gradle-plugin/issues/10
  - this changes the way splash is uploaded, so a temp-file is not used
  - this also depends only on merging the new plugin versions

Template project
- https://github.com/ev3dev-lang-java/template-project-gradle/pull/34
  - just a new plugin version

Examples
- https://github.com/ev3dev-lang-java/examples/pull/4
  - just a new plugin version

Main library
- recent issues
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/649
    - this is about what "unregulated motor" is
    - ev3dev-lang-java assumes that class UnregulatedMotor is only for a RCX motor
    - leJOS assumes that EV3 motors are also intended to be controlled through UnregulatedMotor
    - leJOS also contains class RCXMotor for this purpose
    - not fixed by anything yet
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/651
    - this is about the delay in BaseRegulatedMotor sometimes not being enough
    - this is partially fixed by #655
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/653
    - this is about cleaning up sysfs functions
    - #654 fixes this
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/661
    - SSLHandshakeException
    - I don't know what is the problem might be the absence of unlimited crypto, but I'm not sure
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/662
    - this is about sensor reads being slow
    - I'm working on a JNI C++ library for fast reading of files (no udev)
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/663
    - medium motor refused to move, it just stalled
    - #666 fixes this
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/665
    - basically, coast(), brake(), hold(), flt() don't act immediately
    - #667 fixes this
- pull requests
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/650
    - this changes the writeString() function so that only a single permission check is done (by the constructor itself)
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/654
    - this cleans up Sysfs a little bit
    - it also changes the way writeString() works, which could slightly lower the overhead
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/655
    - this fixes the BaseRegulatedMotor construction on EV3
    - this bypasses the mode switch when the port is already in auto mode, so no udev 
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/659
    - by wuqs-net; from quick look this cleans up EV3Led
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/660
    - just a new plugin version
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/666
    - this sets the speed_sp register before triggering a rotate()/rotateTo(), so that the kernel regulator does not stall
  - https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/667
    - this changes the way that stop is handled, so that when a function is called, it is triggered immediately

jabrena, 2019-08-18:
Related issue for Debian packages:
https://github.com/ev3dev/ev3dev-lang-python/issues/655

jabrena, 2019-12-02:
Fixed

closed 2019-12-02 by jabrena