ev3dev-lang-java/ev3dev-lang-java #670: BaseRegulatedMotor.getTachoCount() flips to negative?

salamanders, 2019-06-18:
I was continually polling the motor's getTachoCount, while steering my robot around using        

    bb.motorLeft.forward() // is a EV3LargeRegulatedMotor
    bb.motorRight.backward() // is a EV3LargeRegulatedMotor
    delay(1_000)

and noticed the tachoCount suddenly flip negative:

    ms	tachoRight    
    9853	786    
    9874	786    
    9897	786    
    9918	786    
    10003	786    
    10020	786    
    10039	786    
    10059	786    
    10080	-786    
    10100	-785    
    10127	-780    
    10143	-775    
    10164	-769    
    10184	-762    

Anyone else encounter this?

JakubVanek, 2019-06-18:
Hi,

this seems to be caused by the use of the polarity parameter:

https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/9d02a050b3baaf37a2e8ae7cce7a1c9d1afa2589/src/main/java/ev3dev/actuators/lego/motors/BaseRegulatedMotor.java#L134-L141

https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/9d02a050b3baaf37a2e8ae7cce7a1c9d1afa2589/src/main/java/ev3dev/actuators/lego/motors/BaseRegulatedMotor.java#L149-L156

Instead of doing the inversion (which should be set to 'normal'), a negative speed_sp/duty_cycle_sp should be written.

Regards,

Jakub

jabrena, 2019-09-01:
Sorry, this info is provided by EV3Dev.

```
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/master/src/main/java/ev3dev/actuators/lego/motors/BaseRegulatedMotor.java#L112-L118
```

I can do anything more.

jabrena, 2019-09-02:
Creating an example to raise the scenario

JakubVanek, 2019-09-02:
Hi @jabrena ,

the issue I think causes this is the usage of the inversion. This inverts not only the movement direction, but also the reported tacho count. From my point of view, the intent of the inversion parameter is to correct the situation when a motor is mounted oppositely, i.e. forwards motion of the motor would move the robot backwards.

My proposal is this: when the user calls the `backward()` or `forward()` function, the inversion would not be touched; rather the following setSpeed lines would be used:
```java
// for forward:
this.setSpeed(+this.speed);
// for backward:
this.setSpeed(-this.speed);
```

Regards,

Jakub Vanek

jabrena, 2019-09-02:
It is a Bug in the implementation. 
I have fixed it. I am going to verify the solution with a System Test with the EV3.

Thanks @salamanders & @JakubVanek 

I continue

salamanders, 2019-09-03:
Fantastic - thank you so much!

jabrena, 2019-09-03:
Defining a TestSample to review the changes in branch: #670

jabrena, 2019-09-03:
@salamanders 

Take a look mate:

``` 
2019-09-03 19:12:23 [main] INFO  example.MotorEncoderTest - MotorEncoderTest
2019-09-03 19:12:28 [main] INFO  e.a.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: A
2019-09-03 19:12:31 [pool-1-thread-2] INFO  example.MotorEncoderTest - FORWARD
2019-09-03 19:12:31 [pool-1-thread-1] INFO  example.MotorEncoderTest - 1 0
2019-09-03 19:12:32 [pool-1-thread-1] INFO  example.MotorEncoderTest - 2 135
2019-09-03 19:12:32 [pool-1-thread-1] INFO  example.MotorEncoderTest - 3 223
2019-09-03 19:12:32 [pool-1-thread-1] INFO  example.MotorEncoderTest - 4 312
2019-09-03 19:12:32 [pool-1-thread-1] INFO  example.MotorEncoderTest - 5 401
2019-09-03 19:12:33 [pool-1-thread-1] INFO  example.MotorEncoderTest - 6 489
2019-09-03 19:12:33 [pool-1-thread-1] INFO  example.MotorEncoderTest - 7 577
2019-09-03 19:12:33 [pool-1-thread-1] INFO  example.MotorEncoderTest - 8 665
2019-09-03 19:12:33 [pool-1-thread-2] INFO  example.MotorEncoderTest - BACKWARD
2019-09-03 19:12:33 [pool-1-thread-1] INFO  example.MotorEncoderTest - 9 748
2019-09-03 19:12:34 [pool-1-thread-1] INFO  example.MotorEncoderTest - 10 658
2019-09-03 19:12:34 [pool-1-thread-1] INFO  example.MotorEncoderTest - 11 572
2019-09-03 19:12:34 [pool-1-thread-1] INFO  example.MotorEncoderTest - 12 395
2019-09-03 19:12:35 [pool-1-thread-1] INFO  example.MotorEncoderTest - 13 292
2019-09-03 19:12:35 [pool-1-thread-1] INFO  example.MotorEncoderTest - 14 203
2019-09-03 19:12:35 [pool-1-thread-1] INFO  example.MotorEncoderTest - 15 116
2019-09-03 19:12:35 [pool-1-thread-1] INFO  example.MotorEncoderTest - 16 28
2019-09-03 19:12:36 [pool-1-thread-1] INFO  example.MotorEncoderTest - 17 -61
2019-09-03 19:12:36 [pool-1-thread-1] INFO  example.MotorEncoderTest - 18 -149
2019-09-03 19:12:36 [pool-1-thread-1] INFO  example.MotorEncoderTest - 19 -235
2019-09-03 19:12:36 [pool-1-thread-1] INFO  example.MotorEncoderTest - 20 -323
2019-09-03 19:12:37 [pool-1-thread-1] INFO  example.MotorEncoderTest - 21 -422
2019-09-03 19:12:37 [pool-1-thread-1] INFO  example.MotorEncoderTest - 22 -523
2019-09-03 19:12:37 [pool-1-thread-1] INFO  example.MotorEncoderTest - 23 -626
2019-09-03 19:12:38 [pool-1-thread-1] INFO  example.MotorEncoderTest - 24 -727
2019-09-03 19:12:38 [pool-1-thread-1] INFO  example.MotorEncoderTest - 25 -814
2019-09-03 19:12:38 [pool-1-thread-1] INFO  example.MotorEncoderTest - 26 -899
2019-09-03 19:12:38 [pool-1-thread-1] INFO  example.MotorEncoderTest - 27 -990
2019-09-03 19:12:39 [pool-1-thread-1] INFO  example.MotorEncoderTest - 28 -1078
2019-09-03 19:12:39 [pool-1-thread-1] INFO  example.MotorEncoderTest - 29 -1165
2019-09-03 19:12:39 [pool-1-thread-1] INFO  example.MotorEncoderTest - 30 -1253
2019-09-03 19:12:39 [pool-1-thread-1] INFO  example.MotorEncoderTest - 31 -1342
2019-09-03 19:12:39 [pool-1-thread-1] INFO  example.MotorEncoderTest - 32 -1430
2019-09-03 19:12:40 [pool-1-thread-1] INFO  example.MotorEncoderTest - 33 -1519
2019-09-03 19:12:40 [pool-1-thread-1] INFO  example.MotorEncoderTest - 34 -1611
2019-09-03 19:12:40 [pool-1-thread-1] INFO  example.MotorEncoderTest - 35 -1699
2019-09-03 19:12:40 [pool-1-thread-1] INFO  example.MotorEncoderTest - 36 -1787
2019-09-03 19:12:41 [pool-1-thread-1] INFO  example.MotorEncoderTest - 37 -1873
2019-09-03 19:12:41 [pool-1-thread-1] INFO  example.MotorEncoderTest - 38 -1961
2019-09-03 19:12:41 [pool-1-thread-1] INFO  example.MotorEncoderTest - 39 -2049
2019-09-03 19:12:41 [pool-1-thread-1] INFO  example.MotorEncoderTest - 40 -2137
2019-09-03 19:12:42 [main] INFO  example.MotorEncoderTest - End
2019-09-03 19:12:42 [pool-1-thread-1] INFO  example.MotorEncoderTest - 41 -2224
```

jabrena, 2019-09-03:
Creating a PR.

jabrena, 2019-09-03:
Example refactored with Java 8 syntax:

``` java
@Slf4j
public class MotorEncoderTest {

    Consumer<RegulatedMotor> readMotor = motor -> {

        IntStream.rangeClosed(1,50).boxed()
            .forEach(i -> {
                LOGGER.info("{} {}", i , motor.getTachoCount());
                Delay.msDelay(200);
            });
    };

    Consumer<RegulatedMotor> writeMotor = motor -> {

        IntStream.rangeClosed(1, 2).boxed()
            .forEach(i -> {
                LOGGER.info("FORWARD");
                motor.forward();
                Delay.msDelay(2000);
                LOGGER.info("BACKWARD");
                motor.backward();
                Delay.msDelay(2000);
            });
    };

    Consumer<RegulatedMotor> processTasks = motor ->  {

        final ExecutorService executor = Executors.newFixedThreadPool(2);

        List<CompletableFuture<Void>> motorTasks = Arrays.asList(
                CompletableFuture.runAsync(() -> this.readMotor.accept(motor), executor),
                CompletableFuture.runAsync(() -> this.writeMotor.accept(motor), executor)).stream()
                .collect(Collectors.toList());

        motorTasks.stream().forEach(CompletableFuture::join);

        executor.shutdown();

        motor.stop();

        LOGGER.info("End");
    };

    public static void main(String[] args) throws Exception{

        LOGGER.info("MotorEncoderTest");

        MotorEncoderTest example = new MotorEncoderTest();

        example.processTasks.accept(new EV3LargeRegulatedMotor(MotorPort.A));

        System.exit(0);
    }
}
```

JakubVanek, 2019-09-04:
Note to thread pools: I think we will have to cache and/or share them, because native threads could be quite expensive regarding both their startup time and memory footprint. However I have no data yet to back this up. Maybe the JVM is clever and caches this behind the scenes and the effect also may not be dramatic.

closed 2019-12-02 by jabrena