ev3dev-lang-java/ev3dev-lang-java #671: Template project gradle - Task:deploy FAILED

floydbloke, 2019-07-28:
Using Eclipse v2019-06 (4.12.0), with the template project gradle imported throws a FAILURE when executing the gradle deploy task
`Working Directory: G:\BrickPi\Java\template_project_gradle
Gradle user home: C:\Users\Eddie\.gradle
Gradle Distribution: Gradle wrapper from target build
Gradle Version: 4.7
Java Home: C:\Users\Eddie\AppData\Local\Programs\AdoptOpenJDK
JVM Arguments: None
Program Arguments: None
Build Scans Enabled: false
Offline Mode Enabled: false
Gradle Tasks: deploy

> Task :clean
> Task :compileJava
> Task :processResources
> Task :classes
> Task :jar
> Task :templateWrapper

> Task :deploy FAILED
Uploading file: template_project_gradle-2.5.3.jar
Uploading file: launcher.sh
[Using default splash, copying to a temporary file.]
Uploading file: splash-4968094004747586544.txt

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':deploy'.
> Upload of program files failed.`

Drilling down into the error under Execute doLast {} action for :deploy shows
`
Root cause: java.nio.file.FileSystemException: C:\Users\Eddie\AppData\Local\Temp\splash-4968094004747586544.txt: The process cannot access the file because it is being used by another process.`

JakubVanek, 2019-07-28:
Hi,

this is a known bug (fixed in newer gradle-plugin): https://github.com/ev3dev-lang-java/gradle-plugin/issues/10
It is fixed in version 1.6.8. If you manually change your project according to https://github.com/ev3dev-lang-java/template-project-gradle/pull/34/files, it should start working.

Regards,

Jakub

floydbloke, 2019-07-29:
Thank you.  That works.

closed 2019-07-29 by floydbloke