ev3dev-lang-java/ev3dev-lang-java #672: OpenCV errors in Eclipse

floydbloke, 2019-08-03:
Some more beginner's help needed please.
Running Eclipse on Win10, I successfully installed OpenCV to the brick (BrickPi+ in my case) using the gradle task.
In the eclipse editor however, any lines trying to import OpenCV could not resolve, e.g.
```
import org.opencv.core.Core;
import org.opencv.highgui.Highgui;
import org.opencv.highgui.VideoCapture;
```
I manually downloaded and ran the OpenCV2.4.9 exe and added the jar to Build Path ->User Libraries
this solved the error in the editor, but when attempting to build it fails as follows:
```
> Task :compileJava FAILED
G:\BrickPi\Java\template_project_gradle\src\main\java\sudsolver\SudSolver.java:3: error: package org.opencv.core does not exist
import org.opencv.core.Core;
                      ^
G:\BrickPi\Java\template_project_gradle\src\main\java\sudsolver\SudSolver.java:4: error: package org.opencv.highgui does not exist
import org.opencv.highgui.Highgui;
                         ^
G:\BrickPi\Java\template_project_gradle\src\main\java\sudsolver\SudSolver.java:5: error: package org.opencv.highgui does not exist
import org.opencv.highgui.VideoCapture;
                         ^
```
Do I need to add OpenCV to the gradle as a dependency  (how?? please) ...or have I missed something else.

JakubVanek, 2019-08-03:
Hi @floydbloke,

it should be enough to add a `libOpenCV = true` line to `config.gradle` file like this:
```groovy

// Project configuration
brick.pref {
    // Main class //
    mainClass = "example.programming.HelloWorld"

    // Brick connection parameters //
    sshHost = "10.0.0.89"
    sshUser = "robot"
    sshPassword = "maker"

    libOpenCV = true // <- here
}
```

Regards,

Jakub

JakubVanek, 2019-08-03:
Hmmm, I have just looked into the plugin and you will need to add a dependency to `build.gradle` as well (see [examples](https://github.com/ev3dev-lang-java/examples/blob/master/opencv/build.gradle) repository):
```gradle
dependencies {
    compileOnly("nu.pattern:opencv:2.4.9-7")
   // ...
}
```

floydbloke, 2019-08-03:
Thank you once again.  Up and running now.

closed 2019-08-03 by floydbloke