ev3dev-lang-java/ev3dev-lang-java #673: Build Java 13 for EV3

jabrena, 2019-08-18:
It is necessary to review the whole process to release a JVM for EV3 with support for OpenJDK13

jabrena, 2019-08-18:
@JakubVanek, what is the difference between both builds:

- https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk13_build_ev3_linux/
- https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk13_build_ev3_linux_native/

JakubVanek, 2019-08-18:
Hi @jabrena,

the concept behind the native builds is that they are not building on x86 machines. Instead, an ARMv8 machine capable of running 32bit binaries is used. This makes it possible to run bootcycle builds - the built JDK can run on the building system too. Bootcycle builds firstly build an intermediate JDK and that JDK is then used to build the final JDK.

The build type is given to Jenkinsfile and from there it also goes to the containers:
https://github.com/ev3dev-lang-java/openjdk-ev3/blob/master/Jenkinsfile

I also thought that with native builders, it may be possible to run some JTreg tests directly after the build. However I have not investigated it further.

Regards,

Jakub

jabrena, 2019-08-18:
> I also thought that with native builders, it may be possible to run some JTreg tests directly after the build. However I have not investigated it further.

This idea could be awesome!

Next week, I will review how to build OpenJDK 13 for Debian Stretch. Currently, it was built for Debian Buster. I will dive in the project to discover how to build for current stable EV3Dev distro.

jabrena, 2019-08-19:
I have upgraded Jenkins job in the CI in order to accept 2 possible distros:

- Debian Stretch
- Debian buster

In the last builds, we built for Debian Buster, but users need builds for `Debian Stretch`
https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk13_build_ev3_linux_native/

jabrena, 2019-08-19:
I have received an Execution error for the version:

https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk13_build_ev3_linux_native/27/console

``` 
checking if packaged modules are kept... yes (default)
checking for version string... 14+10-ev3-unstable
configure: Found potential Boot JDK using configure arguments
configure: Potential Boot JDK found at /build/jdk-ev3 is incorrect JDK version (Error: dl failure on line 597); ignoring
configure: (Your Boot JDK version must be one of: 12 13 14)
configure: error: The path given by --with-boot-jdk does not contain a valid Boot JDK
configure exiting with result code 1
```

jabrena, 2019-08-19:
Testing current build on the latest snapshot for Debian Buster:
https://www.ev3dev.org/downloads/
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2019-08-18/
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2019-08-18/snapshot-ev3dev-buster-ev3-generic-2019-08-18.img.xz

JakubVanek, 2019-08-19:
> I have upgraded Jenkins job in the CI in order to accept 2 possible distros:
> 
>     * Debian Stretch
> 
>     * Debian buster
> 
> 
> In the last builds, we built for Debian Buster, but users need builds for `Debian Stretch`
> https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk13_build_ev3_linux_native/

This is an interesting point - the builds should be unambiguous and do always the same thing. This is because the periodic weekly trigger uses only the default parameter values. The current naming convention didn't consider building all JDK versions for both distros.

JakubVanek, 2019-08-19:
> I am trying to build OpenJDK 13 for Debian Stretch. In the execution, I have received an Execution error for the version:
> 
> https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk13_build_ev3_linux_native/27/console
> 
> ```
> checking if packaged modules are kept... yes (default)
> checking for version string... 14-internal+0-adhoc..jdk
> checking for javac... no
> checking for java... no
> configure: Could not find a valid Boot JDK. OpenJDK distributions are available at http://jdk.java.net/.
> configure: This might be fixed by explicitly setting --with-boot-jdk
> configure: error: Cannot continue
> configure exiting with result code 1
> [BUILD] Configuring Java for target 'ev3'
> 
> ...
> ...
> 
> checking if packaged modules are kept... yes (default)
> checking for version string... 14+10-ev3-unstable
> configure: Found potential Boot JDK using configure arguments
> configure: Potential Boot JDK found at /build/jdk-ev3 is incorrect JDK version (Error: dl failure on line 597); ignoring
> configure: (Your Boot JDK version must be one of: 12 13 14)
> configure: error: The path given by --with-boot-jdk does not contain a valid Boot JDK
> configure exiting with result code 1
> ```

I have quickly looked at it and it seems it is not caused by building for Stretch. The 14+9 tag (previous successful build) succeded with Boot JDK ver 12. The 14+10 tag in the latest build failed; this means that we need to get a valid JDK13 build before building "JDK14".

I will have to implement the non-GA-release (=latest commit) building also for the stabilization branch, as I see no other solution of breaking the cycle until the 13 GA version is released.

JakubVanek, 2019-08-19:
OpenJDK 12 build should on the other hand succeed - there are multiple GA tags available.

JakubVanek, 2019-08-19:
Hmmm, maybe selecting Stretch as the target system did have an effect:
```
checking for version string... 14+10-ev3-unstable
configure: Found potential Boot JDK using configure arguments
configure: Potential Boot JDK found at /build/jdk-ev3 is incorrect JDK version (Error: dl failure on line 597); ignoring
configure: (Your Boot JDK version must be one of: 12 13 14)
```
The Boot JDK used on ARM (native) builders for the build is the one built for Debian Buster in openjdk12_build_ev3_linux_native. As the builder container uses the same distribution version as the target one, this could prevent the JDK12 from operating. If this is the problem, it is necessary to first build JDK12 for Stretch. Using that JDK, it should be then possible to build JDK13/14 for Stretch too.

JakubVanek, 2019-08-19:
There is also an extra step that must be taken before the build can be packaged: the changes in the ev3dev packaging codenames should be integrated to the packaging container, if it is necessary.

JakubVanek, 2019-08-19:
I have some ideas about the new naming convention.
* Folders may be a powerful tool to use. Moreover, there can be duplicate job names in different folders (they create their own namespace).
* The `open`-jdk prefix could be omitted, as everything there is OpenJDK.
```
structure under ev3dev view

/ev3dev-build/                         - for openjdk builds
             /ev3-stretch-native/        - builds for ev3dev-stretch for EV3, building on ARMv8
                                /jdk11     - OpenJDK 11 build
                                /jdk12     - OpenJDK 12 build
                                /jdk13     - OpenJDK 13 build
                                /master    - OpenJDK Master build
             /ev3-buster-native/         - builds for ev3dev-buster for EV3, building on ARMv8
                               /jdk11      - OpenJDK 11 build
                               /jdk12      - OpenJDK 12 build
                               /jdk13      - OpenJDK 13 build
                               /master     - OpenJDK Master build
/ev3dev-test/                          - for openjdk tests
            /ev3-stretch/                - for stretch
                        /TBD               - ???
            /ev3-buster/                 - for buster
                       /TBD                - ???
```
* (optionally) the `ev3` platform part could be removed, as no other platforms are currently used.
* I think the `ev3dev-` prefix on the top level may be necessary. This is because views do not create a separate namespace (AFAIK); therefore having an ev3dev prefix should prevent collisions. Optionally, we could just create an `/ev3dev/` folder and make it like `/ev3dev/build/` and `/ev3dev/test/`.

JakubVanek, 2019-08-19:
More in #674.

JakubVanek, 2019-08-21:
New JDK13 (not 14) snapshot builds after https://github.com/ev3dev-lang-java/openjdk-ev3/pull/38 and https://github.com/ev3dev-lang-java/openjdk-ev3/pull/39 are ready:
* stretch: [openjdk13_build_ev3_linux_native#30](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk13_build_ev3_linux_native/30/)
* buster: [openjdk13_build_ev3_linux_native_next#3](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk13_build_ev3_linux_native_next/3/)

jabrena, 2019-08-22:
# Testing OpenJDK 13 for Debian Stretch

- [x] Build snapshot
- [ ] Install OpenJDK 13 on EV3
- [ ] Testing some samples on the Brick

Build:
https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk13_build_ev3_linux_native/30/

``` 
wget -N https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk13_build_ev3_linux_native/30/artifact/build/jri-ev3.tar.gz
sudo tar -zxvf jri-ev3.tar.gz -C /opt
sudo mv /opt/jri/ /opt/jri-13
update-alternatives --install /usr/bin/java java /opt/jri-13/bin/java 1013
java -Xshare:dump
```

jabrena, 2019-08-22:
We should review to clean old builds like Java 9 & Java 10.
What do you think @JakubVanek ?

Juan Antonio

JakubVanek, 2019-08-22:
Yup, I agree.

Currently the jobs for JDK9 and JDK10 are empty and they don't do anything. However I do not have Jenkins permissions for removing builds, so we will have to ask someone from AdoptOpenJDK.

Additionally, I would like to suggest #674 as part of the cleanup.

JakubVanek, 2019-08-22:
I have added a `[for deletion]` prefix to `Display Name` of the old JDK9/10 related builds. I have also added a `obsolete: ` prefix to the old JDK11/12/13 builds, as they are hopefully going to be replaced with the builds in the new structure.

jabrena, 2019-09-02:
This week, I will store Java 9/10 to the archive:
https://github.com/ev3dev-lang-java/openjdk-ev3-archive

JakubVanek, 2019-09-06:
I have recreated the old JDK9 / JDK10 jobs, they're now here:
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-9/
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-10/

JakubVanek, 2019-09-06:
As all working builds are now replicated [in the new layout](https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/) (including the to-be-archived ones) and https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/674 needs only a PR merge, it should be now possible to request the deletion of the other jobs.

Everything for deletion is marked as `[for deletion]` in https://ci.adoptopenjdk.net/view/ev3dev/ (everything except three folders at the bottom - `Builds`, `Pipelines` and `Tests`).

This currently includes the following:
 * old builds (openjdkXX_build_ev3_*)
 * old folder structure attempt (single `ev3dev` folder; too long URLs)
 * my recent attempt at creating a proof-of-concept of using Job DSL for job autoconfiguration (failed due to groovy sandbox configuration)
 * old test jobs; they all hopefully have their corresponding branches in [openjdk-ev3-test](https://github.com/ev3dev-lang-java/openjdk-ev3-test)
    * [`openjdk09_openjdktest_ev3_linux`](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk09_openjdktest_ev3_linux)
    * [`openjdk10_openjdktest_ev3_linux`](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk10_openjdktest_ev3_linux)
    * [`openjdk10_openjdktest_ev3_linux_full`](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk10_openjdktest_ev3_linux_full)
    * [`openjdk10_openjdktest_systemtest_ev3_linux`](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk10_openjdktest_systemtest_ev3_linux)
    * [`openjdk10_openjdktest_systemtest_ev3_linux_full`](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk10_openjdktest_systemtest_ev3_linux_full)
    * [`openjdk11_openjdktest_ev3_linux`](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk11_openjdktest_ev3_linux)
    * [`openjdk11_openjdktest_ev3_linux_full`](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk11_openjdktest_ev3_linux_full)
    * [`openjdk11_openjdktest_systemtest_ev3_linux`](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk11_openjdktest_systemtest_ev3_linux)

jabrena, 2019-09-06:
It is a very good job @JakubVanek

JakubVanek, 2019-09-25:
JDK13 was released; starting GA builds:
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-13/5/
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster-13/5/

I won't build/upload Debian packages yet.

jabrena, 2020-04-18:
Built

closed 2020-04-18 by jabrena