ev3dev-lang-java/ev3dev-lang-java #674: Restructure CI jobs

JakubVanek, 2019-08-19:
Using folders, it should be possible to simplify the layout of the jobs in Jenkins.
```
/ev3dev/                             - root folder, see below
       /build/                         - for openjdk builds
             /ev3-stretch-native/        - builds for ev3dev-stretch for EV3, building on ARMv8
                                /jdk11     - OpenJDK 11 build
                                /jdk12     - OpenJDK 12 build
                                /jdk13     - OpenJDK 13 build
                                /master    - OpenJDK Master build
             /ev3-buster-native/         - builds for ev3dev-buster for EV3, building on ARMv8
                               /jdk11      - OpenJDK 11 build
                               /jdk12      - OpenJDK 12 build
                               /jdk13      - OpenJDK 13 build
                               /master     - OpenJDK Master build
       /test/                          - for openjdk tests
            /ev3-stretch/                - for stretch
                        /TBD               - ???
            /ev3-buster/                 - for buster
                       /TBD                - ???
       /pipelines/                     - TBD, could be used for building & testing together
```
* (optionally) the `ev3` platform part could be removed, as no other platforms are currently used.
* I think the `ev3dev` folder on the top level may be necessary. This is because views do not create a separate namespace (AFAIK); therefore having an ev3dev prefix should prevent collisions.

JakubVanek, 2019-08-19:
To address the issue with the changing distro versions in the latest jdk12 builds, I have changed the configuration to the following model:

* [openjdk12_build_ev3_linux_native][1]: 12 for Stretch
* [openjdk12_build_ev3_linux_native_next][2]: 12 for Buster
* [openjdk13_build_ev3_linux_native][3]: 13 for Stretch
* [openjdk13_build_ev3_linux_native_next][4]: 13 for Buster

[1]: https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk12_build_ev3_linux_native/ 
[2]: https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk12_build_ev3_linux_native_next/ 
[3]: https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk13_build_ev3_linux_native/ 
[4]: https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk13_build_ev3_linux_native_next/

jabrena, 2019-08-22:
I liked it.

Can we close this issue?

JakubVanek, 2019-08-22:
Not yet - the changes made so far are only temporary. The target state is sketched in the first post and it is not yet implemented; I will try to do so now.

JakubVanek, 2019-08-22:
I have built the structure here: https://ci.adoptopenjdk.net/view/ev3dev/job/ev3dev/.
There are two views in that folder: `All` and `Build Summary`. `All` shows the folder structure; `Build Summary` shows summary of builds in the subfolders.
The URL has doubled ev3dev because the `ev3dev` folder (which creates namespace isolation) is located in `ev3dev` view (which creates the tab on the main page). However, the view part is not mandatory; https://ci.adoptopenjdk.net/job/ev3dev/ will work too.

OpenJDK-EV3 will however have to be adjusted, as the move will change the URLs from where the host/boot JDKs are taken.

jabrena, 2019-08-23:
Maybe, so much folders in the last implementation:

https://ci.adoptopenjdk.net/view/ev3dev/
https://ci.adoptopenjdk.net/view/ev3dev/job/ev3dev/
https://ci.adoptopenjdk.net/view/ev3dev/job/ev3dev/job/build/

What is the idea for pipeline folder and test?
https://ci.adoptopenjdk.net/view/ev3dev/job/ev3dev/job/pipelines/
https://ci.adoptopenjdk.net/view/ev3dev/job/ev3dev/job/test/

jabrena, 2019-08-23:
Now, all builds appear for deletion, but we didn´t talk about the archive process to not loose any build from 9 & 10.

JakubVanek, 2019-08-23:
Yup; however there are already no build artifacts for deletion. I have cleaned the jobs a few months ago to save space on the Jenkins master.

I will reenable the jobs and try to run a new build. Do we want to build only for Stretch?

JakubVanek, 2019-08-23:
The test folder is intended to contain some automatic ARM-native tests, if we add some.
The pipelines folder is intended for orchestration of multiple builds & tests together.
- Now that the builds depend on each other, I think the periodic builds could be ordered from JDK11 to JDK-Master.
- Also, the tests that would be run should be chained to the build via a top-level pipeline job.

JakubVanek, 2019-08-23:
The new URLs are quite long; however it is not all that bad.
* When the view is omitted, the URL is actually shorter - see https://github.com/ev3dev-lang-java/openjdk-ev3/pull/41/files.
* The displayed paths of the jobs are now nicer: [image]

JakubVanek, 2019-08-23:
Hmmm, we have one build of JDK9/JDK10 archived: https://github.com/ev3dev-lang-java/openjdk-ev3/releases/tag/v0.5.0. However it's probably a bit out of date.

JakubVanek, 2019-08-23:
I have some ideas regarding the URL length & complexity:
* We could omit the extra `ev3dev` folder and instead use a shorter prefix (like `elj`) to the `build`, `test` and `pipelines` folders. This would be to prevent collisions with folders from AdoptOpenJDK
   ```
   From:
   https://ci.adoptopenjdk.net/view/ev3dev/job/ev3dev/job/build/job/ev3-buster-native/job/jdk11/
   To:
   https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/ev3-buster-native/job/jdk11/
   ```
* \+ we could omit the `ev3` and `native` part, as everything we now have is ARMv8 -> EV3.
   ```
   From:
   https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/ev3-buster-native/job/jdk11/
   To:
   https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster/job/jdk11/
   ```
* \+ we could remove the separate folder for buster and stretch:
   ```
   From:
   https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster/job/jdk11/
   To:
   https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster-jdk11/
   ```
* \+ we could omit the `jdk` prefix:
   ```
   From:
   https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster-jdk11/
   To:
   https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster-11/
   ```
   Alternatively, without the view prefix:
   ```
   https://ci.adoptopenjdk.net/job/eljbuild/job/buster-11/
   ```

jabrena, 2019-08-23:
I understand the refactoring, I like this schema:

https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster-jdk11/

do you like if we evolve like:

https://ci.adoptopenjdk.net/view/ev3dev/job/build/job/buster-jdk11/
https://ci.adoptopenjdk.net/view/ev3dev/job/test/job/buster-jdk11/

Do you like?

JakubVanek, 2019-08-23:
Hi,

there is a problem with that: the `view/ev3dev` part is optional and it does not isolate job names. The following two URLs would point to the same item:

https://ci.adoptopenjdk.net/view/ev3dev/job/build
https://ci.adoptopenjdk.net/job/build

The only difference is that the `ev3dev` view wouldn't show up in the upper navigation bar.
By doing this, we would take the generic `build` folder/job just for ev3dev and that feels strange to me.

On the other hand, this was the reason why I have introduced the second level `ev3dev` folder - for namespace isolation. The following two links would point to different items:

https://ci.adoptopenjdk.net/job/myFolder/job/myProject
https://ci.adoptopenjdk.net/job/myProject

I chose the `elj` alternative prefix to take the global `eljbuild`, `eljtest` and `eljpipelines` folders as they are unlikely to be taken and they refer to `ev3dev-lang-java`.

Regards

JakubVanek, 2019-08-23:
Hmm, I have clicked on some other builds in Jenkins and the URLs are long as well:
https://ci.adoptopenjdk.net/view/arm/job/build-scripts-pr-tester/job/build-test/job/jobs/job/jdk12u/job/jdk12u-linux-arm-hotspot/
https://ci.adoptopenjdk.net/job/build-scripts/job/jobs/job/jdk11u/job/jdk11u-linux-x64-openj9/

JakubVanek, 2019-08-23:
JDK9 build was successful: [openjdk09_build_ev3_linux#72](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk09_build_ev3_linux/72/)
JDK10 build should be OK too: [openjdk10_build_ev3_linux#147](https://ci.adoptopenjdk.net/view/ev3dev/job/openjdk10_build_ev3_linux/147/)

JakubVanek, 2019-09-06:
I have created the described folder structure:
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/
https://ci.adoptopenjdk.net/view/ev3dev/job/eljtest/
https://ci.adoptopenjdk.net/view/ev3dev/job/eljpipes/

JakubVanek, 2019-09-06:
After https://github.com/ev3dev-lang-java/openjdk-ev3/pull/46 and https://github.com/ev3dev-lang-java/openjdk-ev3/pull/45 are merged, I would consider this issue finished.

JakubVanek, 2019-09-06:
_Originally posted by @jabrena in https://github.com/ev3dev-lang-java/openjdk-ev3/pull/46#issuecomment-528930026_
> What is the idea for:
https://ci.adoptopenjdk.net/view/ev3dev/job/eljtest/
>
>and:
https://ci.adoptopenjdk.net/view/ev3dev/job/eljpipes/
>
>?

The `eljtest` test folder is intended to contain replacements for the old test jobs.
The `eljpipes` pipelines folder is intended for chaining of tests after builds and optionally for scheduling all builds at once (so they could be ordered).

JakubVanek, 2019-09-06:
Finished

closed 2019-09-06 by JakubVanek