ev3dev-lang-java/ev3dev-lang-java #675: Use TemplateProject with Java 13 and Gradle

jabrena, 2019-08-22:
In order to use latest features from Java, it is necessary to upgrade Template Project to work with:

- Java 13
- Latest Gradle: https://github.com/gradle/gradle/issues/8681
- Review plugins in order to be compatible.

JakubVanek, 2019-09-06:
Plugin PR: https://github.com/ev3dev-lang-java/gradle-plugin/pull/12 (merged)
Library PR: https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/679
Examples PR: https://github.com/ev3dev-lang-java/examples/pull/5
Template project PR: https://github.com/ev3dev-lang-java/template-project-gradle/pull/35

JakubVanek, 2019-09-06:
EDIT: no PRs yet; my removed comment linked PRs for the Java 11 move.

jabrena, 2019-09-06:
I am going to close this one, with Java 11, is enough.

closed 2019-09-06 by jabrena