ev3dev-lang-java/ev3dev-lang-java #676: Remove -Xverify:none and -noverify flags from Gradle plugin

jabrena, 2019-09-01:
On Java 14, the JVM flags -Xverify:none and -noverify was deprecated.

It is necessary to remove from the default configuration.

Message from JVM:

```
OpenJDK Client VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
```

JakubVanek, 2019-09-06:
Plugin PR: https://github.com/ev3dev-lang-java/gradle-plugin/pull/13 (merged)
Library PR: https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/686
Examples PR: https://github.com/ev3dev-lang-java/examples/pull/6
Template project PR: https://github.com/ev3dev-lang-java/template-project-gradle/pull/36

JakubVanek, 2019-09-25:
Pending review

jabrena, 2019-12-01:
Fixed

closed 2019-12-01 by jabrena