ev3dev-lang-java/ev3dev-lang-java #687: Archive JDK9 & JDK10 builds

JakubVanek, 2019-09-13:
It is necessary to archive the following builds:
* JDK9
  - build URL: https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-9/1/
  - Java version: `9.0.4+11-ev3`
  - Target OS: `ev3dev-stretch`, probably also `ev3dev-jessie`
* JDK10
  - https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-10/1/
  - Java version: `10.0.2+13-ev3`
  - Target OS: `ev3dev-stretch`, probably also `ev3dev-jessie`

JakubVanek, 2019-09-25:
Using Git directly is not possible.
```sh
root@scw-zen-feynman:~/openjdk-ev3-archive# git push fork master
Warning: Permanently added the RSA host key for IP address '140.82.118.3' to the list of known hosts.
Enter passphrase for key '/root/.ssh/id_rsa': 
Counting objects: 19, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (19/19), done.
Writing objects: 100% (19/19), 907.75 MiB | 4.38 MiB/s, done.
Total 19 (delta 3), reused 0 (delta 0)
remote: Resolving deltas: 100% (3/3), done.
remote: error: GH001: Large files detected. You may want to try Git Large File Storage - https://git-lfs.github.com.
remote: error: Trace: d1180c81c8bdff16a03c2ac7a6f00210
remote: error: See http://git.io/iEPt8g for more information.
remote: error: File stretch-jdk10/jdk-ev3.tar.gz is 193.83 MB; this exceeds GitHub's file size limit of 100.00 MB
remote: error: File stretch-jdk10/jmods-ev3.tar.gz is 154.28 MB; this exceeds GitHub's file size limit of 100.00 MB
remote: error: File stretch-jdk10/jri-ev3.tar.gz is 113.50 MB; this exceeds GitHub's file size limit of 100.00 MB
remote: error: File stretch-jdk9/jdk-ev3.tar.gz is 189.79 MB; this exceeds GitHub's file size limit of 100.00 MB
remote: error: File stretch-jdk9/jmods-ev3.tar.gz is 149.85 MB; this exceeds GitHub's file size limit of 100.00 MB
remote: error: File stretch-jdk9/jri-ev3.tar.gz is 108.32 MB; this exceeds GitHub's file size limit of 100.00 MB
To github.com:JakubVanek/openjdk-ev3-archive
 ! [remote rejected] master -> master (pre-receive hook declined)
error: failed to push some refs to 'git@github.com:JakubVanek/openjdk-ev3-archive'
```

JakubVanek, 2019-09-25:
I think it should be possible to use GitHub releases. However with them I can't create a PR and I don't have write access to the repository, so I can't do the upload.

JakubVanek, 2019-09-25:
Debian packages: https://drive.google.com/drive/folders/187hS5iBhUrILgcY8MhKWJ4BX0IT8jtZg?usp=sharing

jabrena, 2020-12-08:
POC In progress for OpenJDK:

```
split -b 50m jri-ev3.tar.gz "jri-ev3.tar.part."
cat jri-ev3.tar.part.* > new-jri-ev3.tar.gz
tar -xf new-jri-ev3.tar.gz 
cmp jri-ev3.tar.gz new-jri-ev3.tar.gz 
```

jabrena, 2020-12-27:
A better way without any process to split files:
https://github.com/ev3dev-lang-java/openjdk-ev3/releases/tag/openjdk9