ev3dev-lang-java/ev3dev-lang-java #693: Incorrect EV3GyroSensor initialization

CubixSystem, 2019-11-23:
Hello

My problem: `EV3GyroSensor` truncate negative values

`EV3GyroSensor` initialization use simplified `GenericMode` constructor signature
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/dcde52d2aa45cc37fde5659632bd783b575b6d6c/src/main/java/ev3dev/sensors/ev3/EV3GyroSensor.java#L44-L52

https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/dcde52d2aa45cc37fde5659632bd783b575b6d6c/src/main/java/ev3dev/sensors/GenericMode.java#L25-L37

According [java 11 docs](https://docs.oracle.com/en/java/javase/11/docs/api/java.base/java/lang/Float.html#MIN_VALUE) `Float.MIN_VALUE` is
>A constant holding the smallest positive nonzero value of type float

I think `Float.MIN_VALUE` need to be replaced by `-Float.MAX_VALUE`

Not sure, what is right way to fix this problem

1. This is `EV3GyroSensor` specific bug and should be fixed in its constructor
1. This is more general `GenericMode` bug and should be fixed  in its constructor, thus this change affects **all** consumers

P.S. Thank you very much for maintaining this project

JakubVanek, 2019-11-24:
Hi @CubixSystem,

indeed, I think that is what happened - we thought that `Float.MIN_VALUE` behaves similarly how it does for integers. I believe that this affects all `GenericMode` constructors, as the default should have applied no limits.

The limiting behavior was originally implemented for a defensive measure in EV3UltrasonicSensor and EV3IRSensor. It however shouldn't have applied to other sensors.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/dcde52d2aa45cc37fde5659632bd783b575b6d6c/src/main/java/ev3dev/sensors/ev3/EV3IRSensor.java#L32-L53

Thanks,

Jakub

jabrena, 2019-12-02:
Hi @CubixSystem,

Can you test in your project the following Snapshot?
https://jitpack.io/#ev3dev-lang-java/ev3dev-lang-java/2.6.0-SNAPSHOT

CubixSystem, 2019-12-02:
Hi @jabrena, thanks for new release

When I use new version
```gradle
dependencies {
    implementation("org.slf4j:slf4j-simple:1.7.29")
    implementation("com.github.ev3dev-lang-java:ev3dev-lang-java:2.6.0-SNAPSHOT")
}
```

Imports for `lejos` in my code stops working
 
[image]

When I rollback to previous version all working fine
```gradle
dependencies {
    implementation("org.slf4j:slf4j-simple:1.7.29")
    implementation("com.github.ev3dev-lang-java:ev3dev-lang-java:2.5.3")
}
```

I'm pretty new to Java ecosystem and can't track the source of the problem

JakubVanek, 2019-12-02:
I can confirm the issue with latest template-project-gradle when switching to the snapshot version.

jabrena, 2019-12-02:
The issue is about the nature of the dependency. We change from:

```
    compile("org.slf4j:slf4j-api:1.7.25")
    compile("com.github.ev3dev-lang-java:lejos-commons:0.7.3")
    compile("net.java.dev.jna:jna:4.5.2")
``` 

to

```
    implementation("org.slf4j:slf4j-api:1.7.25")
    implementation("com.github.ev3dev-lang-java:lejos-commons:0.7.3")
    implementation("net.java.dev.jna:jna:4.5.2")
```

JakubVanek, 2019-12-02:
We have changed it because `compile` was deprecated (and was probably going to be removed).
However, looking at https://stackoverflow.com/a/44493379:

> 
> Dependencies appearing in the api configurations will be transitively exposed to consumers of the library, and as such will appear on the compile classpath of consumers.
> 
> Dependencies found in the implementation configuration will, on the other hand, not be exposed to consumers, and therefore not leak into the consumers' compile classpath. This comes with several benefits: [...]

JakubVanek, 2019-12-02:
We just need to switch `implementation` for `api`.

jabrena, 2019-12-02:
I will commit and tag again

JakubVanek, 2019-12-02:
This is because we need to have the dependencies tracked as `runtime` too by Gradle. Without this, automatic library uploading in the plugin wouldn't work:
https://github.com/ev3dev-lang-java/gradle-plugin/blob/d61763b8bf8ab422012dfbcd2390f21c765a5df3/src/main/groovy/com/github/elj/gradle/DeploymentTask.groovy#L62-L76

JakubVanek, 2019-12-02:
Description of the different configurations can be found here: https://docs.gradle.org/current/userguide/java_plugin.html#tab:configurations

JakubVanek, 2019-12-02:
I think that Lombok a special case: https://stackoverflow.com/a/6117891
> Lombok does indeed code against internal API, as Sean Patrick Floyd said. However, as lombok is ONLY involved in the compilation phase, its misleading to claim Lombok will only run on a sun VM. [...]

I.e. the Java compiler needs to see these classes in order to be able to let Lombok do the annotation processing. During runtime, there are no Lombok classes.

jabrena, 2019-12-02:
With `implementation`, it is not working. With `compile` it will work.
Now we have a plugin coupled with a the dependency configuration.
What alternatives at plugin level, do we have to decouple it?

JakubVanek, 2019-12-02:
With `api` it will work as well (and in the future too), because that is the transitive-dependency-providing brother of `implementation`.

`compile` will work, but only until some future Gradle version. `api` and `implementation` are its replacements (I am thinking that they indeed were introduced in order to create this transitive and non-transitive split).

An alternative for the plugin (if it had not to access `runtimeClasspath`) would be would have to directly write the JAR URLs (or local paths) somewhere. However not wanting to do so (in the installer.sh script) was the motivation to introduce this functionality in the first place.

jabrena, 2019-12-02:
We are going to create another issue about the plugin, but first, we need to fix this issue and not using the slim features, the user will deploy on the robot.

jabrena, 2019-12-02:
Hi @CubixSystem,

Can you review the new snapshot?
https://jitpack.io/#ev3dev-lang-java/ev3dev-lang-java/2.6.1-SNAPSHOT

jabrena, 2019-12-02:
Hi @CubixSystem, 

@JakubVanek & me, fixed the issue with the dependency.
https://jitpack.io/#ev3dev-lang-java/ev3dev-lang-java/2.6.2-SNAPSHOT

Can you confirm that the issue is behind us?

CubixSystem, 2019-12-02:
@jabrena @JakubVanek thanks for fix!

I tested with `2.6.1-SNAPSHOT` and all works fine

jabrena, 2019-12-02:
Can you recheck with the ultimate fix: `2.6.2-SNAPSHOT`

CubixSystem, 2019-12-02:
Done, all works as expected
No changes from my side was made (except bumping version)

jabrena, 2019-12-02:
Can you share the plugin configuration?

CubixSystem, 2019-12-02:
Here it is https://gist.github.com/CubixSystem/c5384bcecc394b50ac3e4c70cb2bac50

jabrena, 2019-12-02:
Nope, the required file is: `config.gradle` not the `build.gradle`

JakubVanek, 2019-12-02:
I'd like to add clarification to the relation between plugin & projects: the plugin does not have effect on the dependency resolution. Rather, it is a consumer of the `runtimeClasspath` API provided by Gradle that collects the JAR dependencies that are needed during the runtime. The libraries that are returned by Gradle are then uploaded to the brick. Using `compileOnly` would not make the plugin broken, but it would prevent it from seeing these dependencies and uploading them.

jabrena, 2019-12-02:
Maybe, this comment is a good candidate for this folder:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/tree/master/docs/adr

CubixSystem, 2019-12-02:
> Nope, the required file is: `config.gradle` not the `build.gradle`
Ah, sorry
My config os super simple
```
// Project configuration
brick.pref {
    // Main class //
    mainClass = "my.Gyro"

    // Brick connection parameters //
    sshHost = "192.168.2.3"
    sshUser = "robot"
    sshPassword = <password_here>
}
```

commit bd5540db (2019-12-02): fix(sensors): do not truncate negative values for gyro

Previously, negative values from sensors were truncated to zero.
This happened because Float.MIN_VALUE in Java is the smallest _positive_
value that a float can have. It is therefore necessary to use
-Float.MAX_VALUE as a lower bound to disable limiting.

Fixes: #693
closed 2019-12-02 by jabrena