ev3dev-lang-java/ev3dev-lang-java #695: Deploy on multiple bricks simultaneously using gradle tasks

mrkrizic, 2019-11-25:
Is there a way to deploy a program onto two bricks without having to change the config.gradle file for each additional brick? I want to test remote method calls between two or more bricks, but changing the config.gradle file to deploy on a specific brick seems very inefficient. Can I add multiple configs, preferrably with different main classes to be used in the gradle tasks?

JakubVanek, 2019-11-26:
Hi @mrkrizic,

the gradle plugin does not support this feature yet. However, there is a way how to work it around. In `build.gradle`, the config file is loaded using this syntax:
```gradle
apply from: './config.gradle'
```
With the help of Gradle [`-P` switch][gradle], it should be possible to parameterize it:
```gradle
apply from: "./${brickname}/config.gradle"
```
(run as:)
```shell
gradle build -P brickname=alonso
```
This way, it will load config from the `alonso` subdirectory.

To provide support for multiple bricks, it is necessary to add a wrapper script (example for Linux shell):
```sh
for name in alonso barbara charlie; do
    gradle build -P brickname=$name
done
```

Similar scripts should be possible to write with Batch and Powershell scripts too. 

Regards,

Jakub
[gradle]: https://docs.gradle.org/current/userguide/build_environment.html#sec:project_properties

JakubVanek, 2019-11-26:
EDIT: it is necessary to use the `deploy` gradle subtask instead of the `build` one. The JAR file will be overwritten each time.

JakubVanek, 2019-11-26:
Hmmm, looking back at this, a better convention than per-brick directories might be a common directory for configs an per-brick configs: `<project>/bricks/alonso.gradle`.

jabrena, 2019-12-01:
Wow,

the idea from @mrkrizic is pretty nice!

mrkrizic, 2019-12-10:
Hello sorry for the late response. Thank you for considering this feature. I have created two seperate config files and put them into a folder named configs. I have a problem running the gradle task from the command line as it does not translate ${brickname} to the input I provide with -P.
Currently I am still commenting the one I don't need out as I don't have that much time to look into it. If I make any progress I will get back to you.

closed 2020-04-18 by jabrena