ev3dev-lang-java/ev3dev-lang-java #697: Prepare release 2.6.x

jabrena, 2019-12-02:
Reviewing all changes from 2.5.3 until latest commit

jabrena, 2019-12-02:
Reviewed

JakubVanek, 2019-12-02:
Regarding JVM: there is no problem with JDK13, but currently the only JRI version in the ev3dev repository is the 11.0.4 release. However there is also something to be done: 11.0.5 was released and it should be ready ([1], [2]) for repackaging & upload. I think that going in this direction would be better.

[1]: https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-11/14/
[2]: https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster-11/13/

JakubVanek, 2019-12-06:
OpenJDK 11.0.5 is now available in the ev3dev repository (https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/704).

JakubVanek, 2020-03-21:
OpenJDK 11.0.6 is now uploaded to the ev3dev repository.

jabrena, 2020-03-21:
Next time, I will publish the artifact in the ev3dev repository.
I will create an issue with the idea to write the documentation to do that process.

JakubVanek, 2020-03-21:
Yes, I think that is a good idea. The documentation is already available here: https://docs.google.com/document/d/1rsBDU68_ETXm9b_Lb4b6oPVBbNwT9f4xEua2w3-Pm6A/edit#heading=h.m4x8bw6tk4he

You will also need to send your public GPG & SSH keys to @dlech so that you can access the SFTP upload server and have the packages automatically imported. (This is documented also here: https://www.ev3dev.org/docs/devtools/packaging-for-ev3dev/)

JakubVanek, 2020-03-22:
It seems that next JDK11 release is getting close. https://wiki.openjdk.java.net/display/JDKUpdates/JDK11u:

> OpenJDK 11.0.7
> 
>  * Wednesday, November 27 2019: jdk11u-dev repo open (tag: 11.0.7+0)
>  * Tuesday, January 28 2020: First merge from jdk11u-dev to jdk11u (tag: 11.0.7+1)
>  * Tuesday, February 25 2020: Rampdown; last merge from jdk11u-dev to jdk11u
>  * Tuesday, March 24 2020: Last tag before code freeze
>  * Tuesday, April 14 2020: GA; OpenJDK 11.0.7 released (tag: jdk-11.0.7-ga)

JakubVanek, 2020-03-22:
I have slightly restructured the documentation. It now includes a first time setup section which goes through what needs to be done before the upload: https://docs.google.com/document/d/1rsBDU68_ETXm9b_Lb4b6oPVBbNwT9f4xEua2w3-Pm6A/edit#heading=h.6lt8jp6ideb0

JakubVanek, 2020-03-22:
I have additionally translated the manual to Markdown: https://github.com/ev3dev-lang-java/openjdk-ev3/pull/52

jabrena, 2020-03-24:
This weekend, I will review the PR.

JakubVanek, 2020-03-25:
Just received this broadcast email from jdk-updates mailing list:
> Hi,
> 
> Tag jdk-11.0.7+9 was pushed. jdk11u is closed now.
> Please don't push to jdk11u anymore. The closed security
> changes for 11.0.7 will be prepared on top of this tag.
> 
> Best regards,
>   Goetz.
> 
> See also project plan at https://wiki.openjdk.java.net/display/JDKUpdates/JDK11u

jabrena, 2020-04-18:
Reorganize the project

closed 2020-04-18 by jabrena