ev3dev-lang-java/ev3dev-lang-java #698: Review integration points from Gradle plugin

jabrena, 2019-12-02:


JakubVanek, 2019-12-02:
I don't think this is a bug - this is a feature. I have intentionally integrated dependency lookup to the plugin to make it more user friendly (i.e. usable as a replacement for an IDE plugin too). Below is a log from uploading the `examples` project to a brick with deleted `java` folder:

```
kuba@linuxtardis-home:~/lego/ev3/elj/elj-example-stuff$ ./gradlew deploy

> Task :demo:deploy
/ Connecting...
/ Ensuring on-brick directory structure...
/ Checking on-brick Java libraries...
Uploading file: ev3dev-lang-java-2.5.3.jar
Uploading file: lejos-navigation-0.2.0.jar
Uploading file: RPLidar4J-v0.3.2-SNAPSHOT.jar
Uploading file: logback-classic-1.2.3.jar
Uploading file: jlayer-1.0.1.jar
Uploading file: slf4j-api-1.7.25.jar
Uploading file: lejos-commons-0.7.3.jar
Uploading file: jna-4.5.2.jar
Uploading file: logback-core-1.2.3.jar
Uploading file: rxtx-2.1.7.jar
Uploading file: lombok-1.16.16.jar
/ Uploading program JAR...
Uploading file: demo-0.1.0-SNAPSHOT.jar
/ Uploading program launcher...
Uploading file: launcher.sh
Uploading file from stream: [default splash]

> Task :demo-2.4.16:deploy
/ Connecting...
/ Ensuring on-brick directory structure...
/ Checking on-brick Java libraries...
/ Uploading program JAR...
Uploading file: demo-2.4.16-0.1.0-SNAPSHOT.jar
/ Uploading program launcher...
Uploading file: launcher.sh
Uploading file from stream: [default splash]

> Task :ev3dev-lang-java:deploy
/ Connecting...
/ Ensuring on-brick directory structure...
/ Checking on-brick Java libraries...
/ Uploading program JAR...
Uploading file: ev3dev-lang-java-0.6.1.jar
/ Uploading program launcher...
Uploading file: launcher.sh
Uploading file from stream: [default splash]

> Task :helloWorldKotlin:deploy
/ Connecting...
/ Ensuring on-brick directory structure...
/ Checking on-brick Java libraries...
Uploading file: kotlin-stdlib-1.3.50.jar
Uploading file: kotlin-stdlib-common-1.3.50.jar
Uploading file: annotations-13.0.jar
/ Uploading program JAR...
Uploading file: helloWorldKotlin-0.7.0.jar
/ Uploading program launcher...
Uploading file: launcher.sh
Uploading file from stream: [default splash]

> Task :mbeans:deploy
/ Connecting...
/ Ensuring on-brick directory structure...
/ Checking on-brick Java libraries...
/ Uploading program JAR...
Uploading file: mbeans-0.1.0.jar
/ Uploading program launcher...
Uploading file: launcher.sh
Uploading file from stream: [default splash]

> Task :opencv:deploy
/ Connecting...
/ Ensuring on-brick directory structure...
/ Checking on-brick Java libraries...
/ Uploading program JAR...
Uploading file: opencv-2.1.0-SNAPSHOT.jar
/ Uploading program launcher...
Uploading file: launcher.sh
Uploading file from stream: [default splash]

> Task :rosjava-helloworld:deploy
/ Connecting...
/ Ensuring on-brick directory structure...
/ Checking on-brick Java libraries...
Uploading file: lombok-1.18.8.jar
Uploading file: commons-httpclient-3.1.jar
Uploading file: commons-logging-1.2.jar
Uploading file: rosjava-0.2.2.jar
Uploading file: sensor_msgs-1.12.7.jar
Uploading file: rosjava_test_msgs-0.2.1.jar
Uploading file: rosgraph_msgs-1.11.2.jar
Uploading file: nav_msgs-1.11.9.jar
Uploading file: tf2_msgs-0.5.20.jar
Uploading file: geometry_msgs-1.12.7.jar
Uploading file: actionlib_msgs-1.12.7.jar
Uploading file: std_msgs-0.5.11.jar
Uploading file: message_generation-0.3.3.jar
Uploading file: commons-codec-1.9.jar
Uploading file: apache_xmlrpc_server-0.2.2.jar
Uploading file: apache_xmlrpc_client-0.2.2.jar
Uploading file: apache_xmlrpc_common-0.2.2.jar
Uploading file: dnsjava-2.1.1.jar
Uploading file: com.springsource.org.apache.commons.httpclient-3.1.0.jar
Uploading file: com.springsource.org.apache.commons.logging-1.1.1.jar
Uploading file: com.springsource.org.apache.commons.net-2.0.0.jar
Uploading file: guava-12.0.jar
Uploading file: netty-3.5.2.Final.jar
Uploading file: com.springsource.org.apache.commons.codec-1.3.0.jar
Uploading file: com.springsource.org.apache.commons.io-1.4.0.jar
Uploading file: commons-pool-1.6.jar
Uploading file: com.springsource.org.apache.commons.lang-2.4.0.jar
Uploading file: gradle_plugins-0.3.3.jar
Uploading file: ws-commons-util-1.0.1.jar
Uploading file: junit-3.8.2.jar
Uploading file: jsr305-1.3.9.jar
Uploading file: xml-apis-1.0.b2.jar
/ Uploading program JAR...
Uploading file: rosjava-helloworld-0.1.0.jar
/ Uploading program launcher...
Uploading file: launcher.sh
Uploading file from stream: [default splash]

> Task :video:deploy
/ Connecting...
/ Ensuring on-brick directory structure...
/ Checking on-brick Java libraries...
Uploading file: jcodec-javase-0.2.3.jar
Uploading file: jcodec-0.2.3.jar
Uploading file: jsr305-3.0.2.jar
/ Uploading program JAR...
Uploading file: video-0.3.0.jar
/ Uploading program launcher...
Uploading file: launcher.sh
Uploading file from stream: [default splash]

BUILD SUCCESSFUL in 2m 50s
49 actionable tasks: 42 executed, 7 up-to-date
```

JakubVanek, 2019-12-02:
Indeed, for this feature to be useful, it requires to have dependencies exported on the `runtimeClasspath`. However if they are not there (user chooses to use `compileOnly`), nothing bad will happen, but no libraries will be uploaded.

jabrena, 2019-12-02:
In my IDE, `api` is not recognized. How to enable api in my IDE?

JakubVanek, 2019-12-02:
Ah, I see now why I have missed it: it is a part of the `java-library` plugin that extends the `java` plugin with library concepts (taken from https://stackoverflow.com/a/53726023)

JakubVanek, 2019-12-02:
I have created a PR which includes the `api` label and the `java-library` plugin: https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/699

Build has now succeeded with Gradle 5.6.1 on Ubuntu OpenJDK 11.0.4.

jabrena, 2019-12-02:
Using `api`, it is necessary to change the `Gradle plugin`?

jabrena, 2019-12-02:
Now, the dependencies from the library has the original behaviour.

JakubVanek, 2019-12-02:
Yaay, if it regards the upload functionality, that still works. After re-deleting `java` folder:
```
kuba@linuxtardis-home:~/lego/ev3/elj/elj-tpg-stuff$ ./gradlew deploy

> Task :compileJava
Note: /home/kuba/lego/ev3/elj/elj-tpg-stuff/src/main/java/example/robotics/ev3/actuator/LEDPatternsExample.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.

> Task :deploy
/ Connecting...
/ Ensuring on-brick directory structure...
/ Checking on-brick Java libraries...
Uploading file: slf4j-simple-1.7.25.jar
Uploading file: ev3dev-lang-java-2.6.2-SNAPSHOT.jar
Uploading file: slf4j-api-1.7.25.jar
Uploading file: lejos-commons-0.7.3.jar
Uploading file: jna-4.5.2.jar
/ Uploading program JAR...
Uploading file: template_project_gradle-2.6.0.jar
/ Uploading program launcher...
Uploading file: launcher.sh
Uploading file from stream: [default splash]

BUILD SUCCESSFUL in 36s
6 actionable tasks: 6 executed
```

jabrena, 2019-12-02:
jajajaja, I need to do `Yoga` sometimes with you! :p

closed 2019-12-02 by jabrena

JakubVanek, 2019-12-02:
Hmmm, interesting: this is the POM of the broken version: https://jitpack.io/com/github/ev3dev-lang-java/ev3dev-lang-java/2.6.0-SNAPSHOT/ev3dev-lang-java-2.6.0-SNAPSHOT.pom
And this is the POM of the fixed version: https://jitpack.io/com/github/ev3dev-lang-java/ev3dev-lang-java/2.6.2-SNAPSHOT/ev3dev-lang-java-2.6.2-SNAPSHOT.pom

Looking at it, marking the dependency as `implementation` will indeed only hide it from the compilation classpath in subprojects (which we do not want with ev3dev-lang-java's sublibraries, which form an integral part of the API). It will still be present in the runtime classpath in subprojects, because the library code needs it for running (but not for interaction via the API).

JakubVanek, 2019-12-02:
TL;DR: by using `implementation` only (and not `api`), I have told Gradle not to do compilation with the subdependencies. `api` will, on the other hand, include subdependencies in compilation of projects.