ev3dev-lang-java/ev3dev-lang-java #703: Make 'template-project-gradle' repository template

CubixSystem, 2019-12-03:
Hello

I think it will be useful to make https://github.com/ev3dev-lang-java/template-project-gradle a  template repository

From https://github.blog/2019-06-06-generate-new-repositories-with-repository-templates/
>To get started, all you need to do is mark a repository as a template, and you’ll immediately be able to use it to generate new repositories with all of the template repository’s files and folders.

Benefits
>Whenever you view a template repository, you’ll see a brand new “Use this template” button that enables you to get started with that codebase in seconds.

Thanks,
Oleg

jabrena, 2020-01-11:
Interesting idea, I will review the idea.

JakubVanek, 2020-03-14:
I had a chance to test this. It is just a checkbox in the repository configuration. It then allows one to "generate" a new local repository based on the picked one.

- Template repo: https://github.com/JakubVanek/c4ev3-workspace
- Repository generated using "Use this template": https://github.com/JakubVanek/clone-test

The resulting repo does not show up as a fork, but it shows as "generated from ..."

I believe this could be useful for https://github.com/ev3dev-lang-java/template-project-gradle/ (which has a similar purpose to the repository above). It is not some miracle that will draw new users in, but it is a nice-to-have in that the cloning process feels smooth.

jabrena, 2020-04-18:
Hi @CubixSystem, thanks for the suggestion but I prefer to minimize the lock-in features.

closed 2020-04-18 by jabrena

LouisCAD, 2020-04-20:
@jabrena How is the suggestion a lock-in feature?

jabrena, 2020-04-20:
Basically the feature is from Github only and I don´t many time to in this kind of features basically.