ev3dev-lang-java/ev3dev-lang-java #707: Build Java 14 for EV3 for Debian Stretch & Debian Buster

jabrena, 2020-01-11:
It is necessary to build & test OpenJDK 14 for EV3:

-[x] Build: https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-14/
-[x] Build: https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster-14/
-[ ] Test:
-[ ] Test:

JakubVanek, 2020-01-13:
JDK14ea on Buster:
```
robot@ev3dev:~$ java -version
openjdk version "14" 2020-03-17
OpenJDK Runtime Environment (build 14+31-ev3-unreleased)
OpenJDK Client VM (build 14+31-ev3-unreleased, mixed mode, sharing)
```
Fresh deployment of template project to the brick:
```
$ ~/apps/gradle-5.6.1/bin/gradle deployRun

Welcome to Gradle 5.6.1!

Here are the highlights of this release:
 - Incremental Groovy compilation
 - Groovy compile avoidance
 - Test fixtures for Java projects
 - Manage plugin versions via settings script

For more details see https://docs.gradle.org/5.6.1/release-notes.html

Starting a Gradle Daemon (subsequent builds will be faster)

> Task :deploy
/ Connecting...
/ Ensuring on-brick directory structure...
/ Checking on-brick Java libraries...
Uploading file: slf4j-simple-1.7.25.jar
Uploading file: ev3dev-lang-java-2.5.3.jar
Uploading file: slf4j-api-1.7.25.jar
Uploading file: lejos-commons-0.7.3.jar
Uploading file: jna-4.5.2.jar
/ Uploading program JAR...
Uploading file: template_project_gradle-2.6.0.jar
/ Uploading program launcher...
Uploading file: launcher.sh
Uploading file from stream: [default splash]
---> Running "time java -Xms64m -Xmx64m -XX:+UseSerialGC -jar "/home/robot/java/programs/template_project_gradle-2.6.0.jar"" <---
Hello World

real    0m7.000s
user    0m3.985s
sys     0m0.648s

BUILD SUCCESSFUL in 2m 18s
7 actionable tasks: 6 executed, 1 up-to-date
```

JakubVanek, 2020-01-13:
Please note however that in order to test the template project, it is necessary to move current `/usr/bin/java` to a different temporary location and create a symbolic link from `/usr/bin/java` to wherever the JRI 14 is extracted.

jabrena, 2020-04-18:
Thanks for the advice Jakub.

Good luck in your new projects

closed 2020-04-18 by jabrena