ev3dev-lang-java/ev3dev-lang-java #708: Extend project support until 2021

jabrena, 2020-01-11:


JakubVanek, 2020-01-12:
Hi,

I think that cutting off support quickly/completely is not a good idea, as the product can live on long after it has been "replaced" with something new and better. For example, the university where I want to study at still uses NXTs alongside with EV3s. Why should they throw them away if they work and EV3s do have considerable cost? For mapping, AI/ML & LIDAR they might not be the best, but for (basic?) control engineering & acting as physical frontends for more complex stuff they're OK.

It'd also be nice to have some enumeration of what kinds of support this affects:
* solving of user problems
* accepting new features
* accepting any commits
* etc.

Best regards,

Jakub

jabrena, 2020-04-18:
Thanks for the comment, I will review it.

JakubVanek, 2020-04-23:
It seems that after all, the successor product has arrived already. The PyBricks implementation is apparently going to run on non-EV3 platforms and it is going to have an unified API over these platforms. https://pybricks.com/downloads/

* LEGO MINDSTORMS EV3
  - ev3dev version is available now
  - there was even an experimental port to EV3RT ([source](https://www.facebook.com/Pybricks/posts/549767495658735?comment_id=549839085651576&reply_comment_id=557264548242363&__cft__[0]=AZU8hsyqiHRKK3KRrX1WXu1yuF8u0NOKlnfbpCO4o5R0j8bbeZ44DypVuVjPU3nYNgJbbauOwUYo3KNkpDyd2DVV4VFoge_DJuIp6f16W4QxjMzHWJk-FMJe1fs-XymE9pVVr750t5BytDPGsg9X9ZdX4Jhmd__pD7ZHD_HDgY3T9A&__tn__=R]-R))
* Powered Up / BOOST / Control+ is apparently planned

At this point I understand why LEGO is invested in MicroPython. I don't know about a popular microcontroller-oriented implementation of a JVM that would be actively developed (this excludes NXJ/TinyVM). Also, Java requires compilation to an intermediate bytecode and the final binary. On the other hand, MicroPython:
- is actively developed
- supports many platforms
- comes with a beginner-friendly language
- does not require special compilers, just a text editor

Using ev3dev for the OS seems like an independent and good choice - it probably made the development easier for them (rewritten drivers, up-to-date Linux environment) and it also establishes a relationship with the community.

JakubVanek, 2020-04-23:
On the other hand, I don't think that EV3 is going to be discontinued/replaced soon - I think that Powered Up is a parallel product branch aiming at a lower price point & easier setup, while being more limited than the EV3 line. However I guess that the next-gen full programmable brick will use the Powered Up style connectors (perhaps with a conversion cable to the old RJ11-style ones).

JakubVanek, 2020-04-26:
Interesting (https://github.com/pybricks/pybricks-api):
> Each Pybricks firmware comes with the Pybricks package. All modules, classes, methods, and functions in that package have optimized implementations in C for high performance and reduced memory footprint.

They will be able to avoid some sort of "classloading" & warmup for the library code this way.

jabrena, 2020-04-27:
Sorry @JakubVanek, I don´t understand this comment. 
This is a Java project, not a C project.

JakubVanek, 2020-04-27:
> Sorry @JakubVanek, I don´t understand this comment.
> This is a Java project, not a C project.

Yes, however it depends on what your goal is - whether you want to write everything in Java (also a worthwhile goal) or if you just want to provide a Java API.

JakubVanek, 2020-04-27:
> > Sorry @JakubVanek, I don´t understand this comment.
> > This is a Java project, not a C project.
> 
> Yes, however it depends on what your goal is - whether you want to write everything in Java (also a worthwhile goal) or if you just want to provide a Java API.

The idea behind my comment is that by conceding on having everything implemented in a high-level language like Python (and thus giving up easy editing and perhaps introspection/reflection and maybe extensibility; => single language goal), they were likely able to provide a faster load times and higher runtime performance (=> API goal).

jabrena, 2020-04-27:
Hi @JakubVanek,

I am not sure @JakubVanek.
But this set of comments are not related with issue.

closed 2020-05-09 by jabrena

jabrena, 2020-05-24:
https://www.lego.com/en-dk/product/lego-mindstorms-ev3-31313?fbclid=IwAR1NGikpPazhWfvRh_KGU6mEFWadh7ss_yeNrPsng6mSnUhh_WbVPli-AV0

https://www.stonewars.de/geruechte/lego-mindstorms-51515-snowflake/?fbclid=IwAR1x83lpVqFps98K1jJpITTFTo_zXTjroeQWRau-tZtN8A_3rcoBhzj5FwU

jabrena, 2020-06-12:
[image]

jabrena, 2020-06-12:
https://www.lego.com/en-us/aboutus/news/2020/june/lego-mindstorms-robot-inventor?fbclid=IwAR3g8wv7MeQaIH61iS7O05kyvh6T8oj829gWfkqSqpXGXJLSFuc0ur3U4Lk