ev3dev-lang-java/ev3dev-lang-java #709: Permission denied for task uploadGradleLibraries

mrkrizic, 2020-02-27:
Hello,

I want to deploy my program on a BrickPi, but I get a ClassNotFoundException. I figured out that my gradle libraries did not deploy and started the uploadGradleLibraries task. The task failed, so I ran it again using --stacktrace. 

The error message is: "Caused by: 3: Permission denied"

I set the permissions for all files in the /robot directory to (O)0777. I still keep getting this error and I don't know what the task is doing before returning an error. 

Has somebody ever received this error and knows how I can fix it? I am currently using version 1.6.8 of the plugin.

Thank you in advance.

mrkrizic, 2020-03-02:
Ok I found a "simple" fix. I reinstalled the OS and the error went away. It was still not possible to upload the program using deployRun, but creating a shadowJar and uploading it manually solved this problem as well.

closed 2020-03-02 by mrkrizic

JakubVanek, 2020-03-02:
Hi @mrkrizic,

I think that it might be caused by the issue that previous versions of the plugin haven't created all subdirectories of the required path.
```sh
mkdir -p /home/robot/java/
```

I don't know what could cause this otherwise. Some improbable options include connecting via a different user, extra permissions set through `setfacl`/`getfacl` and apparmor/selinux rules (however this is disabled in the initial ev3dev image AFAIK).

Alternatively, I'd suggest upgrading to the 2.0.0 version of the plugin, as it makes the deployment easier (no more manual uploadGradleLibraries). 

Regarding BrickPi - I think it is necessary to run `setupEverything` on the first deployment to a given brick to install the Java runtime and accompanying libraries.

Best regards,

Jakub