ev3dev-lang-java/ev3dev-lang-java #710: Unknown Platform detection for Debian stretch Raspberry Pi 3 model B BrickPi3

refond, 2020-03-09:
I installed ev3dev 2.3.5 using the provided debian stretch image on a RaspberryPi 3 model B assembled with a BrickPi 3.

I had not luck with the "setupEverything" gradlew task, most certainly due the platform detection problem I believe I am facing.
For that reason I installed the openjdk-11-jdk through debian backport as well as libopencv2.4-java and librxtx-java.

Then I could successfully used deploy, run, undeploy... gradle tasks from template-project-gradle as well as example project except I only successfully ran "Hello World" :)

When using ev3dev-lang-java libraries I get the following error:

```
robot@ev3dev:~/examples$ ./ev3dev-lang-java-0.6.1.sh 
(...removed)

              POWERED BY:

                 ev3dev
                OpenJDK
             Ev3dev-lang-java
 INFO  ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
 DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release
 DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
 INFO  ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
 DEBUG ev3dev.hardware.EV3DevPlatforms - Detecting platform with the battery approach
 DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
 TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/lego-ev3-battery
 DEBUG ev3dev.hardware.EV3DevPlatforms - Detecting platform with the battery approach
 TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/pistorms-battery
 DEBUG ev3dev.hardware.EV3DevPlatforms - Detecting platform with the battery approach
 TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/brickpi-battery
 DEBUG ev3dev.hardware.EV3DevPlatforms - Detecting platform with the battery approach
 TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/brickpi3-battery
 DEBUG ev3dev.hardware.EV3DevPlatforms - Detecting platform with the battery approach
 TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/.
 TRACE ev3dev.hardware.EV3DevPlatforms - Detected platform: UNKNOWN
 INFO  e.a.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: A
 TRACE ev3dev.hardware.EV3DevDevice - Retrieving devices in path: 
 TRACE ev3dev.utils.Sysfs - ls /sys/class/lego-port
Exception in thread "main" java.lang.ExceptionInInitializerError
Caused by: java.lang.RuntimeException: The path doesn't exist: /sys/class/lego-port
	at ev3dev.utils.Sysfs.getElements(Sysfs.java:107)
	at ev3dev.hardware.EV3DevDevice.detect(EV3DevDevice.java:57)
	at ev3dev.actuators.lego.motors.BaseRegulatedMotor.<init>(BaseRegulatedMotor.java:87)
	at ev3dev.actuators.lego.motors.EV3LargeRegulatedMotor.<init>(EV3LargeRegulatedMotor.java:26)
	at ev3dev.misc.BumperCar.<clinit>(BumperCar.java:19)
robot@ev3dev:~/examples$ 
```
I only have the BumperCar log file but obtained the same result with the simpler BatteryDemo.

Here are some hardware information:

```
robot@ev3dev:~/examples$ sudo inxi -Fx
System:    Host: ev3dev Kernel: 4.14.114-ev3dev-2.3.5-rpi2 armv7l (32 bit gcc: 6.4.1) Console: tty 0
           Distro: ev3dev-stretch
Machine:   No /sys/class/dmi; using dmidecode: no smbios data available. Old system?
CPU:       Quad core ARMv7 rev 4 (v7l) (-MCP-) (ARM)bmips: 
           clock speeds: max: 1200 MHz 1: 600 MHz 2: 600 MHz 3: 600 MHz 4: 600 MHz
Graphics:  Card: Failed to Detect Video Card!
           Display Server: N/A driver: N/A tty size: 166x28 Advanced Data: N/A for root out of X
Network:   Card: Standard Microsystems SMSC9512/9514 Fast Ethernet Adapter usb-ID: 001-003
           IF: N/A state: N/A speed: N/A duplex: N/A mac: N/A
Drives:    HDD Total Size: NA (-)
           ID-1: /dev/mmcblk0 model: N/A size: 63.9GB
Partition: ID-1: / size: 59G used: 1.9G (4%) fs: ext4 dev: /dev/root
Sensors:   None detected - is lm-sensors installed and configured?
Info:      Processes: 94 Uptime: 2:30 Memory: 56.6/927.2MB Init: systemd runlevel: 5 Gcc sys: N/A
           Client: Shell (sudo) inxi: 2.3.5 
``` 
I read that lm-sensors is not working for ARM architecture so the Sensors section is not relevant.
I'll provide more relevant information as necessary.
Cheers, Patrick

jabrena, 2020-03-10:
Can you review the sdcard configuration?
When you burn the distro, you need to touch the file `config.txt`

My clue:

` TRACE ev3dev.hardware.EV3DevPlatforms - Detected platform: UNKNOWN`

Juan Antonio

refond, 2020-03-10:
The config.txt file at: `/boot/flash/config.txt` has everything commented out except: 
```
# uncomment if you are using BrickPi3
dtoverlay=brickpi3
```
Anything else that should be configured there ?

refond, 2020-03-10:
I tried to get information about the battery as described at:
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/brickpi3.html#battery
But I do not find the expected sysfs device node `/sys/class/power_supply/brickpi3-battery/` 

```
robot@ev3dev:/$ ls -l /sys/class/power_supply/
total 0
robot@ev3dev:/$
```

Looking at  `/sys/devices/platform` I found a `brickpi3_battery` but without expected `scope` nor `voltage_now` Sysfs Attributes (although I am not comfortable with drivers stuff...). Here is how it looks on my system:

```
robot@ev3dev:/$ ls -l /sys/devices/platform
total 0
drwxr-xr-x  4 root root    0 Mar 10 18:46 Fixed MDIO bus.0
drwxr-xr-x  3 root root    0 Mar 10 18:46 alarmtimer
drwxr-xr-x  4 root root    0 Mar 10 18:46 board
drwxr-xr-x  3 root root    0 Mar 10 18:43 brickpi3_battery
drwxr-xr-x  3 root root    0 Mar 10 18:46 clocks
drwxr-xr-x  4 root root    0 Mar 10 18:46 fixedregulator_3v3
drwxr-xr-x  4 root root    0 Mar 10 18:46 fixedregulator_5v0
drwxr-xr-x  4 root root    0 Mar 10 18:46 leds
drwxr-xr-x  3 root root    0 Mar 10 18:46 phy
drwxr-xr-x  2 root root    0 Mar 10 18:46 power
drwxr-xr-x  4 root root    0 Mar 10 18:46 reg-dummy
drwxr-xr-x  3 root root    0 Mar 10 18:46 regulatory.0
drwxr-xr-x 27 root root    0 Mar 10 18:46 soc
drwxr-xr-x  3 root root    0 Mar 10 18:46 timer
-rw-r--r--  1 root root 4096 Mar 10 18:46 uevent
robot@ev3dev:/$
```
Then within `/sys/devices/platform/brickpi3_battery/`
```
robot@ev3dev:/$ ls -l /sys/devices/platform/brickpi3_battery/
total 0
-rw-r--r-- 1 root root 4096 Mar 10 18:49 driver_override
-r--r--r-- 1 root root 4096 Mar 10 18:49 modalias
lrwxrwxrwx 1 root root    0 Mar 10 18:49 of_node -> ../../../firmware/devicetree/base/brickpi3_battery
drwxr-xr-x 2 root root    0 Mar 10 18:43 power
lrwxrwxrwx 1 root root    0 Mar 10 18:49 subsystem -> ../../../bus/platform
-rw-r--r-- 1 root root 4096 Mar 10 18:49 uevent
robot@ev3dev:/$ 
```
I have no clue where to go on now :(

refond, 2020-03-10:
I discovered a **brickpi3 firmware problem** by running `dmesg` as advised at:
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/brickpi3.html

```
[    3.669467] brickpi3 spi0.1: Address: 1
[    3.673711] brickpi3 spi0.1: Mfg: Dexter Industries
[    3.678398] brickpi3 spi0.1: Board: BrickPi3
[    3.685258] brickpi3 spi0.1: HW: 3.2.1
[    3.693037] brickpi3 spi0.1: FW: 1.0.1
[    3.696414] brickpi3 spi0.1: Unsupported firmware version, expecting 1.4.x
```
I'll post how to update the firmware when I'll find out how.

jabrena, 2020-03-11:
`EV3dev` detects the battery: `/sys/devices/platform/brickpi3_battery/` but I don´t know when the project changed the path from: `/sys/class/power_supply/brickpi3-battery` to `/sys/devices/platform/brickpi3_battery/`

I checked the documentation and the path is: `/sys/class/power_supply/lego-ev3-battery/`
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/ev3.html#battery

@dlech Hi, do you know why in Brickpi3, the battery appear in `/sys/devices/platform/brickpi3_battery/`?

Juan Antonio

JakubVanek, 2020-03-11:
Hi everyone,

I don't have any experience with BrickPi3, but for Stretch, the battery sysfs interface is still [documented](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/brickpi3.html#battery) to be at `/sys/class/power_supply/brickpi3-battery/`. Driver source code [confirms this](https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-stretch/brickpi3/brickpi3_battery.c#L84).

I think that the `/sys/devices` path is an auxiliary directory structure in this case. The `brickpi3-battery` (PSU) directory resides in `/sys/class/power_supply` and simultaneously it should be a subdirectory in the `/sys/devices/platform/brickpi3_battery` directory (device itself that has the PSU interface registered).

My main idea is that it is caused by the firmware version mismatch - the battery driver may be loaded, but perhaps the hardware battery interface cannot be found.

Best regards,

Jakub

dlech, 2020-03-11:
> I'll post how to update the firmware when I'll find out how.

Run `sudo update-brickpi3-fw`

> Hi, do you know why in Brickpi3, the battery appear in `/sys/devices/platform/brickpi3_battery/`?

This is the platform device that gets matched with a driver to create the actual power supply device.

refond, 2020-03-11:
Hi, I finally managed to fix the problem by performing a Brickpi3 firmware update.
@JakubVanek I guess you are right regarding
> My main idea is that it is caused by the firmware version mismatch

```
[    4.330756] brickpi3 spi0.1: Address: 1
[    4.340737] brickpi3 spi0.1: Mfg: Dexter Industries
[    4.352062] brickpi3 spi0.1: Board: BrickPi3
[    4.372806] brickpi3 spi0.1: HW: 3.2.1
[    4.385747] brickpi3 spi0.1: FW: 1.4.6
[    4.387971] brickpi3 spi0.1: ID: 4889C6C7504B5354392E314BFF0F0C38
```
I did it the lazy way, installing 2019.04.05_Raspbian_For_Robots_by_Dexter_Industries_Stretch.zip
image on a new SDCard. Here is the steps I followed in case it may help someone else: 

Followed instructions at: https://www.dexterindustries.com/howto/install-raspbian-for-robots-image-on-an-sd-card/
Downloaded image for RaspberryPi3: "Using Raspbian for Robots:   from Sourceforge here."
https://sourceforge.net/projects/dexterindustriesraspbianflavor/

Unlike stated resizing took place automatically 3.5Gb => 64Gb
Then with screen, mouse and keyboard connected to the Raspberry I had to:

- set up the network wi-fi using `sudo raspi-config` and going to configure network wifi.
- Updating the system: 
```
sudo apt-get update
sudo apt-get upgrade
```
A first diagnostic using the test program available from the Raspian for robot desktop confirmed the issue:
```
BrickPi3 Troubleshooting Script log

Checking for hardware, and checking hardware and firmware version.
==================================================================
BrickPi3 firmware needs to be version 1.4.x but is currently version 1.0.1
```

- Updating the firmware using distribution provided script:
`sudo bash /home/pi/Dexter/BrickPi3/Firmware/brickpi3samd_flash_firmware.sh`

A second diagnostic reusing the test program confirmed the fix:

```
BrickPi3 Troubleshooting Script log

Checking for hardware, and checking hardware and firmware version.
==================================================================
Manufacturer    :  Dexter Industries
Board           :  BrickPi3
Serial Number   :  4889C6C7504B5354392E314BFF0F0C38
Hardware version:  3.2.1
Firmware version:  1.4.6
Battery voltage :  10.089
9v voltage      :  9.185
5v voltage      :  5.187
3.3v voltage    :  3.336
```

Going back to Ev3dev distribution I could happily get battery voltage and NXT motors run in minutes using the very nice `template_project_gradle` while hacking from `example` project !

@jabrena , @JakubVanek , @dlech Thanks a lot for your support and for the software,
Cheers!

jabrena, 2020-03-12:
Enjoy the project!

closed 2020-03-12 by jabrena