ev3dev-lang-java/ev3dev-lang-java #711: Problems following the installation documentation.

themichaelbarrett, 2020-04-15:
(I'm sorry if this is the wrong place for this - I couldn't see anywhere else to ask a question).

I have tried following the Getting Started documentation three times, twice on OSX and once on Linux, and every time I get stuck at the same point.

I can successfully run the "./gradlew testConnection" command.

However, "./gradlew deployAndBrickRun" fails with "Task 'deployAndBrickRun' not found in root project 'template_project_gradle'.".

In addition, in Eclipse the ev3dev and leJos imports can not be resolved, leading to many errors.

Any pointers on how to get past this point would be appreciated.

JakubVanek, 2020-04-15:
Hi,

I'm sorry for your experience. There were recent changes in the [template-project-gradle](https://github.com/ev3dev-lang-java/template-project-gradle/) and the [gradle-plugin](https://github.com/ev3dev-lang-java/gradle-plugin).

First time setup is described here: [gradle-plugin#What's next](https://github.com/ev3dev-lang-java/gradle-plugin#whats-next)
Available gradle tasks are decribed here: [gradle-plugin#All provided gradle tasks](https://github.com/ev3dev-lang-java/gradle-plugin#all-provided-gradle-tasks)

I believe the correct command in this case would be:
```sh
./gradlew deployRun
```

The behaviour of Eclipse seems a bit strange. I think a possible cause could be that the plugin and the libraries require Java 11 (both for IDE JRE, compiler JDK and brick JRI).

Best regards,

Jakub

themichaelbarrett, 2020-04-15:
Hi,
No apologies needed - your documentation was very clear, and I learned a lot from following it.

deployRun did indeed work.

I did have to manually select the JRE installed by SDKMAN in Eclipse. It now shows 11.0.6.hs-adpt, which I think is correct.

However, that hasn't solved the import problem. I'll keep plugging away on that one.

Michael

themichaelbarrett, 2020-04-15:
I think I have fixed the Eclipse issue - though to be honest I don't really understand what I've done.

Inside the file gradle/wrapper/gradle-wrapper.properties, there was a reference to "gradle-5.6.1-bin.zip". Using the listing at https://services.gradle.org/distributions/, I updated this to "gradle-6.3-bin.zip". 

All the import statements in Eclipse now seem to work fine.

jabrena, 2020-04-18:
Hi @themichaelbarrett,

Enjoy!

I will upgrade the project with the dependency this month.

Juan Antonio

closed 2020-04-18 by jabrena