ev3dev-lang-java/ev3dev-lang-java #712: Intellij license

jjegarcia, 2020-04-17:
Hi,
I wonder how i will be out to get an intellij open license for me work in the project. I have my own EV3 and have have a devian flashed on my memory card. Could I use netbeans otherwise?
Any hint will be highly appreciated.
Jenaro Garcia

JakubVanek, 2020-04-18:
Hi,

there is the free (and open-source) IntelliJ IDEA Community edition which should work without any problems: https://www.jetbrains.com/idea/download. I don't know about the Ultimate license for OSS, but it seems that getting it would be complicated - there is not enough development activity in this project and it is only for the core devs.

I'm not familiar with NetBeans, but if you're able to import a Gradle-based project, it should work identically. Eclipse should work too, but according to https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/711#issuecomment-614076841 there may be some extra steps necessary.

Best regards,

Jakub

jabrena, 2020-04-18:
Hi @jjegarcia,

yes, you could use Netbeans if you feel comfortable with that IDE although the project doesn´t so much support for that IDE

Juan Antonio

jjegarcia, 2020-04-18:
thanks a lot both for your prompt reply. I look forward to start doing something useful with the EV3 using your project

jabrena, 2020-04-18:
You are welcome.

closed 2020-04-18 by jabrena