ev3dev-lang-java/ev3dev-lang-java #713: java.lang.UnsupportedClassVersionError: com/github/elj/gradle/GradlePlugin has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0 > com/github/elj/gradle/GradlePlugin has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0

jjegarcia, 2020-04-20:
Has anyone got an idea of how to overcome this issue coming from Intellij Build?

java.lang.UnsupportedClassVersionError: com/github/elj/gradle/GradlePlugin has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0
> com/github/elj/gradle/GradlePlugin has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0

JakubVanek, 2020-04-20:
Yup, you will need to use Java 11 or newer for the Gradle JVM (it should be somewhere in the Settings menu).

Best regards,

Jakub

jjegarcia, 2020-04-20:
I have got installed version 12:
jenarogarcia@MGC000010485 ~ $ java -version
openjdk version "12.0.2" 2019-07-16
OpenJDK Runtime Environment AdoptOpenJDK (build 12.0.2+10)
Eclipse OpenJ9 VM AdoptOpenJDK (build openj9-0.15.1, JRE 12 Mac OS X amd64-64-Bit Compressed References 20190915_201 (JIT enabled, AOT enabled)
OpenJ9   - 0f66c6431
OMR      - ec782f26
JCL      - 9984ea0dd9 based on jdk-12.0.2+10)

JakubVanek, 2020-04-20:
IntelliJ bundles its a Java 8 runtime with itself, so it might still be using an older one. In fact, it has to be using Java 8: https://stackoverflow.com/a/35866015 (the reported 52.0 == Java 8).

jjegarcia, 2020-04-20:
I am using java 8 for Android Studio in this computer as this is a requirement for my work. However , I do not understand why Intellij and android studio should be linked. But this is the only explanation for it. I may have to try another computer otherwise to use IntelliJ only

JakubVanek, 2020-04-20:
I hope that would not be necessary - it should suffice to set the Gradle JVM to the newer one in the IntelliJ settings.

jabrena, 2020-04-20:
I recommend for Java projects use Intellij, not Android Studio.
In some cases, Android Studio offer specific Android classes when you want to resolve a specific class.

- Intellj -> JVM software
- Android Studio -> Android software

jjegarcia, 2020-04-20:
> I hope that would not be necessary - it should suffice to set the Gradle JVM to the newer one in the IntelliJ settings.

Do you now where in the settings it may be? I am looking in preferences with no luck

jjegarcia, 2020-04-20:
this is what I get using Java_home:
jenarogarcia@MGC000010485 ~ $ /System/Library/Frameworks/JavaVM.framework/Versions/A/Commands/java_home ; exit;
/Library/Java/JavaVirtualMachines/jdk1.8.0_221.jdk/Contents/Home
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...none found.

[Process completed]

jjegarcia, 2020-04-20:
start wondering weather sdkman is unable to switch java versions. I believe  gradlew.bat is the setting file. I wonder if I can hardcode the location of sdkman generated version (build 12.0.2+10)

JakubVanek, 2020-04-20:
I don't currently have IntelliJ installed, but I think it looked something like this:
[image]

jabrena, 2020-04-21:
Hi @jjegarcia, you can have multiple Java versions installed with `sdkman` but later you need to indicate your Java IDE, what JVM are you going to use.

Review the documentation:
https://ev3dev-lang-java.github.io/docs/support/getting_started/laptop.html

jjegarcia, 2020-04-21:
Thanks again

closed 2020-04-21 by jabrena