ev3dev-lang-java/ev3dev-lang-java #714: Remove Gradle plugin from main library

jabrena, 2020-04-20:
It is necessary to remove the Gradle plugin from main library to reduce the learning curve.

JakubVanek, 2020-04-20:
Technically, recompiling Gradle plugin with Java 8 would be enough to eliminate that.

JakubVanek, 2020-04-21:
Hi,

could you please re-review the pull request? The JVM that is used to run Gradle and the JVM that is used for program compilation are not necessarily the same (as can be seen with IntelliJ). Gradle indeed supports running on Java 8 and newer. Even if the build fails later (because the compiled project targets Java 11 and uses Java 11 libraries), it fails with a proper error message. Also, because the Gradle plugin is only part of the build system, it does not affect other libraries during run-time.

Best regards,

Jakub

JakubVanek, 2020-04-21:
Regarding https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/711#issuecomment-614076841: I don't know what is the cause, but I suspect that the import would fail even without having the gradle-plugin.

jabrena, 2020-05-09:
Identified commit: `git show d7ff265`

JakubVanek, 2020-05-09:
Hi @jabrena,

the plugin is also used in [template-project-gradle](https://github.com/ev3dev-lang-java/template-project-gradle/) (commit [1289d0a65f083d8f4c3b433023b7f3a8f03d52ee](https://github.com/ev3dev-lang-java/template-project-gradle/commit/1289d0a65f083d8f4c3b433023b7f3a8f03d52ee)) and [examples](https://github.com/ev3dev-lang-java/examples) (commit [1c38a6368dd7c7b7eab8c68e7c18a46ac347d4a0](https://github.com/ev3dev-lang-java/examples/commit/1c38a6368dd7c7b7eab8c68e7c18a46ac347d4a0)). Removing it only from this repository will have no effect on the typical user, because they usually pull the built library artifact from JitPack.

IMHO the better way forward would be for you to refactor the plugin (perhaps keeping only some features, or moving to pure Maven) to become more familiar with its internals, so that it would be possible to support it in this project without external contributors.

Without the plugin you will also need to eventually update the [installer](https://github.com/ev3dev-lang-java/installer/blob/master/installer.sh) for Debian Buster and new Jenkins URLs ([new](https://github.com/ev3dev-lang-java/gradle-plugin/blob/d61763b8bf8ab422012dfbcd2390f21c765a5df3/src/main/resources/setup.sh#L107) vs [old](https://github.com/ev3dev-lang-java/installer/blob/950496b8626971f15a8aad5f24c281c40033a755/installer.sh#L23)). Currently, the gradle-plugin uses its own specialized script for this purpose that supports Debian Stretch and Buster: https://github.com/ev3dev-lang-java/gradle-plugin/blob/master/src/main/resources/setup.sh. The old setup scripts relied on the external installer.

Best regards,

Jakub

jabrena, 2020-05-09:
Hi @JakubVanek, 

This is the problem, I accepted the change in the project with the promise to create an Intellij plugin but that development never comes so, I will have to revert the change gradually.

Juan Antonio

jabrena, 2020-05-09:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/716

closed 2020-05-09 by jabrena