ev3dev-lang-java/ev3dev-lang-java #719: Test last builds from AdoptOpenjdk CI

jabrena, 2020-05-09:
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/

JakubVanek, 2020-05-09:
Fix for JDK11 build: https://github.com/ev3dev-lang-java/openjdk-ev3/pull/53

JakubVanek, 2020-05-09:
JDK15 build fixes: https://github.com/ev3dev-lang-java/openjdk-ev3/pull/54
JDK14 changes (docs only): https://github.com/ev3dev-lang-java/openjdk-ev3/pull/55

jabrena, 2020-05-09:
When you have all builds in Green, design a Strategy to add tests from:
https://github.com/AdoptOpenJDK/openjdk-tests

In OpenJDK 10, I did in a semi manual way. With Native hardware, we should pass the tests.

JakubVanek, 2020-05-09:
> When you have all builds in Green, design a Strategy to add tests from:
> https://github.com/AdoptOpenJDK/openjdk-tests

I have left the AdoptOpenJDK GitHub org, so I don't have permissions on Jenkins and we will have to wait for next scheduled rebuild (13.4. for JDK11, 11.4. for JDK15).

JakubVanek, 2020-05-09:
> When you have all builds in Green, design a Strategy to add tests from:
> https://github.com/AdoptOpenJDK/openjdk-tests

Also please don't count on me doing this soon, if you want to do a new release in near future, please move it to the next milestone.

jabrena, 2020-05-10:
I will use this issue to review the builds in this period, May-June 2020 but I will create a new issue to add OpenJDK test without any time pressure. :)

jabrena, 2021-07-18:
Lack of time

closed 2021-07-18 by jabrena