ev3dev-lang-java/ev3dev-lang-java #721: Remove gradle-plugin from the examples repository

JakubVanek, 2020-05-09:
The repository at https://github.com/ev3dev-lang-java/examples still uses the gradle plugin, it may be worth evaluating the pros and cons of keeping it there.

Commit history:
```
$ git log --pretty=format:"%h %ad | %s%d [%an]" --graph --date=short
*   197b44b 2019-12-01 | Merge pull request #7 from JakubVanek/feature/plugin_update (HEAD -> master, origin/master, origin/HEAD) [Juan Antonio Breña Moral]
|\  
| * 1c7c546 2019-12-01 | Upgrade to gradle plugin 2.0.0 [Jakub Vaněk]
| * 321cc32 2019-09-06 | Upgrade to gradle plugin 1.6.10 - remove -noverify (origin/noverify) [Jakub Vaněk]
| * 13b849d 2019-09-04 | Add missing rosjava dependencies to fix build [Jakub Vaněk]
| * 67389d4 2019-09-04 | Update Lombok to fix Gradle build crash [Jakub Vaněk]
| * 052ac29 2019-09-04 | Fix package-info classes which have crashed newer Lombok [Jakub Vaněk]
| * c25f9b4 2019-09-04 | Update Kotlin to 1.3.50 fix build Gradle crash [Jakub Vaněk]
| * 783a0c2 2019-09-04 | Update to gradle-plugin 1.6.9 [Jakub Vaněk]
| * cb1eba4 2019-09-04 | Update to Java 11 [Jakub Vaněk]
| * 19fdb61 2019-09-04 | Update to Gradle 5.6.1 [Jakub Vaněk]
| * 3f786ef 2019-03-31 | Use new dependency specification syntax; upgrade plugin to 1.6.8 [Jakub Vaněk]
* |   0b44281 2019-01-31 | Merge pull request #3 from ev3dev-lang-java/pluginUpgrade [Juan Antonio Breña Moral]
|\ \  
| |/  
| * 11bd51c 2019-01-30 | Upgrade gradle plugin to 1.6.5 [Jakub Vaněk]
|/  
*   1c38a63 2019-01-14 | Merge pull request #2 from ev3dev-lang-java/plugin [Juan Antonio Breña Moral]
|\  
| * b1346fe 2019-01-14 | Move splash to gradle-plugin (upgrade to 1.6.4) [Jakub Vaněk]
| * 33f3646 2019-01-13 | Add note about changing SSH parameters [Jakub Vaněk]
| * 0494226 2019-01-13 | Add info about mainclass changing + do not use MANIFEST stuff [Jakub Vaněk]
| * 2f1ed0a 2019-01-13 | Put shell wrappers /home/robot/examples [Jakub Vaněk]
| * 2a24f4a 2019-01-13 | Add usage instructions for the new plugin integration [Jakub Vaněk]
| * 8daddcd 2019-01-13 | Remove obsolete deploy.gradle [Jakub Vaněk]
| * 8982240 2019-01-13 | Adapt video project for ev3dev-lang-java plugin [Jakub Vaněk]
| * d4888e7 2019-01-13 | Adapt rosjava-helloworld project for ev3dev-lang-java plugin [Jakub Vaněk]
| * be35149 2019-01-13 | Adapt opencv project for ev3dev-lang-java plugin [Jakub Vaněk]
| * 087a287 2019-01-13 | Adapt mbeans project for ev3dev-lang-java plugin [Jakub Vaněk]
| * 97bc641 2019-01-13 | Adapt HelloWorldKotlin project for ev3dev-lang-java plugin [Jakub Vaněk]
| * 3cf562f 2019-01-13 | Adapt ev3dev-lang-java project for ev3dev-lang-java plugin [Jakub Vaněk]
| * 199e327 2019-01-13 | Adapt demo project for ev3dev-lang-java plugin [Jakub Vaněk]
| * e868f7b 2019-01-13 | Adapt demo-2.4.16 project for ev3dev-lang-java plugin [Jakub Vaněk]
| * 1913769 2019-01-13 | Adapt root project for ev3dev-lang-java plugin [Jakub Vaněk]
| * 76cb3d7 2019-01-13 | Extend gitignore with build and .gradle directories [Jakub Vaněk]
|/  
* 0688aca 2018-12-09 | Upgrading ev3dev-lang-java version Fixing issue with packages [Juan Antonio Brena Moral]

```

jabrena, 2020-05-09:
Hi @JakubVanek,

As I said previously, **I am going to remove the usage of that Plugin from the main repos because it is not a unified solution**. I will find a way from Gradle to execute scripts in the same way than in Maven. But now, it is not a priority.

Take a look the new Maven repository. 
https://github.com/ev3dev-lang-java/template-project-maven
I invested few hours to add that support and it is easy to maintain.

Please, put the focus here if you want to collaborate for OpenJDK15 (March 2021):
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/
https://en.wikipedia.org/wiki/Java_version_history

JakubVanek, 2020-05-09:
> As I said previously, **I am going to remove the usage of that Plugin from the main repos because it is not a unified solution**.

I respect that, I created this issue as I thought that you intentionally didn't proceed with this last repository because the example projects require multiple libraries. FatJars should work OK.

jabrena, 2020-05-09:
@JakubVanek,

Put the focus in the areas (Makefiles & Bash Scripting) where you maximize your skills:
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/
https://github.com/ev3dev-lang-java/openjdk-ev3

I try to be clear and honest. 

Juan Antonio

JakubVanek, 2020-05-09:
> Take a look the new Maven repository.
> https://github.com/ev3dev-lang-java/template-project-maven
> I invested few hours to add that support and it is easy to maintain.

It looks pretty slick indeed. I didn't pursue the Maven plugin because I didn't have the idea about the bash-scripts, I would essentially replicate the current Gradle setup (i.e. new targets, configuration through XML, etc.). I didn't see enough benefits for the effort (learning maven plugin infrastructure, mapping the ideas etc.) when the Gradle project in my eyes already provided everything necessary (although in an non-longterm-maintainable fashion.

JakubVanek, 2020-05-09:
However, with bash scripts, M$ Window$ complicate things a little.
 * OpenSSH needs to be installed separately.
 * Bash needs to be installed separately or scripts need to be rewritten using cmd.exe or PowerShell.

Both of these programs may get installed with Git for Windows though, they ship basic MinGW environment.

jabrena, 2020-05-09:
Sorry, Win10 has a Linux subsystem inside.

With few changes, I will emulate the same behaviour in Gradle to execute scripts and unify that scripts using Installer :)

JakubVanek, 2020-05-09:
> Sorry, Win10 has a Linux subsystem inside.
> 
> With few changes, I will emulate the same behaviour in Gradle to execute scripts and unify that scripts using Installer :)

Yup, that ships full Linux userland, that would solve this; however it also needs to be enabled separately. The last thing that may get in the way is if IDEs (if not helped by custom plugins) can launch scripts inside bash.exe (it would however make sense that the answer is yes).

JakubVanek, 2020-05-09:
> With few changes, I will emulate the same behaviour in Gradle to execute scripts and unify that scripts using Installer :)

The slim JARs with dependency upload target may be harder to get working, but otherwise it should be possible to match the functionality:
 * if manifest contains the main class, shell script wrapper w/ splash should be well doable
 * if the user has SSH available, they can run SSH commands directly instead of running them through Maven
 * configuration is solved by it being easily editable bash scripts
 * installer can be solved similarly (just upload setup.sh and go)

jabrena, 2021-01-06:
Fixed

closed 2021-01-06 by jabrena