ev3dev-lang-java/ev3dev-lang-java #728: SSH handshake

jabrena, 2020-05-17:
It is necessary review SSL problems:

```
#####################
# MAVEN EV3 SCRIPTS #
#####################

-Djavax.net.debug=all
Deploy artifact to EV3
sshpass -p maker scp -P 22 ./target/template-project-maven-0.1.0-SNAPSHOT-jar-with-dependencies.jar robot@192.168.1.151:/home/robot/
Run artifact
sshpass -p maker ssh -p 22 robot@192.168.1.151 java -jar template-project-maven-0.1.0-SNAPSHOT-jar-with-dependencies.jar -Djavax.net.debug=all
/tmp/BROKER_CLIENT_PING_4490452626536460128
2020-05-17 14:30:51 [main] WARN com.github.broker.GitClientWrapper - https://github.com/broker-game/broker-dev-environment7: Secure connection to https://github.com/broker-game/broker-dev-environment7 could not be established because of SSL problems
org.eclipse.jgit.api.errors.TransportException: https://github.com/broker-game/broker-dev-environment7: Secure connection to https://github.com/broker-game/broker-dev-environment7 could not be established because of SSL problems
        at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:222)
        at org.eclipse.jgit.api.CloneCommand.fetch(CloneCommand.java:292)
        at org.eclipse.jgit.api.CloneCommand.call(CloneCommand.java:176)
        at com.github.broker.GitClientWrapper.cloneRepository(GitClientWrapper.java:49)
        at com.github.broker.BrokerClient.connect(BrokerClient.java:85)
        at example.PingApp.main(PingApp.java:18)
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/broker-game/broker-dev-environment7: Secure connection to https://github.com/broker-game/broker-dev-environment7 could not be established because of SSL problems
        at org.eclipse.jgit.transport.TransportHttp.handleSslFailure(TransportHttp.java:675)
        at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:576)
        at org.eclipse.jgit.transport.TransportHttp.openFetch(TransportHttp.java:359)
        at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:105)
        at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:91)
        at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1260)
        at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:211)
        ... 5 more
Caused by: javax.net.ssl.SSLHandshakeException: Received fatal alert: handshake_failure
        at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:131)
        at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:117)
        at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:307)
        at java.base/sun.security.ssl.Alert$AlertConsumer.consume(Alert.java:291)
        at java.base/sun.security.ssl.TransportContext.dispatch(TransportContext.java:180)
        at java.base/sun.security.ssl.SSLTransport.decode(SSLTransport.java:164)
        at java.base/sun.security.ssl.SSLSocketImpl.decode(SSLSocketImpl.java:1151)
        at java.base/sun.security.ssl.SSLSocketImpl.readHandshakeRecord(SSLSocketImpl.java:1062)
        at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:402)
        at java.base/sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:567)
        at java.base/sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185)
        at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1587)
        at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1515)
        at java.base/java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:527)
        at java.base/sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:334)
        at org.eclipse.jgit.transport.http.JDKHttpConnection.getResponseCode(JDKHttpConnection.java:85)
        at org.eclipse.jgit.util.HttpSupport.response(HttpSupport.java:208)
        at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:507)
        ... 10 more
2020-05-17 14:30:57 [main] INFO com.github.broker.BrokerClient - 1589725852723_PING_PING.json
Exception in thread "main" java.lang.NullPointerException
        at com.github.broker.GitClientWrapper.upgradeRepository(GitClientWrapper.java:68)
        at com.github.broker.BrokerClient.produce(BrokerClient.java:100)
        at example.PingApp.main(PingApp.java:23)
[ERROR] Command execution failed.
org.apache.commons.exec.ExecuteException: Process exited with an error: 1 (Exit value: 1)
    at org.apache.commons.exec.DefaultExecutor.executeInternal (DefaultExecutor.java:404)
    at org.apache.commons.exec.DefaultExecutor.execute (DefaultExecutor.java:166)
    at org.codehaus.mojo.exec.ExecMojo.executeCommandLine (ExecMojo.java:804)
    at org.codehaus.mojo.exec.ExecMojo.executeCommandLine (ExecMojo.java:751)
    at org.codehaus.mojo.exec.ExecMojo.execute (ExecMojo.java:313)
    at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo (DefaultBuildPluginManager.java:137)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:210)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:156)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:148)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:117)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:81)
    at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build (SingleThreadedBuilder.java:56)
    at org.apache.maven.lifecycle.internal.LifecycleStarter.execute (LifecycleStarter.java:128)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:305)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:192)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:105)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:957)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:289)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:193)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0 (Native Method)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke (NativeMethodAccessorImpl.java:62)
    at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke (DelegatingMethodAccessorImpl.java:43)
    at java.lang.reflect.Method.invoke (Method.java:566)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:282)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:225)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:406)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:347)

```

jabrena, 2020-05-17:
```
wget https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-14/lastSuccessfulBuild/artifact/build/jdk-ev3.tar.gz
sudo tar -zxvf jdk-ev3.tar.gz -C /opt
sudo mv /opt/jdk/ /opt/jdk-14
sudo update-alternatives --install /usr/bin/java java /opt/jdk-14/bin/java 2014
java -version
openjdk version "14" 2020-03-17
OpenJDK Runtime Environment (build 14+36-ev3-unreleased)
OpenJDK Client VM (build 14+36-ev3-unreleased, mixed mode, sharing)
```

JakubVanek, 2020-05-17:
Related to https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/661

I think this might be relatedto using the distribution-provided CA certificates. Would manual installation of JRI from Jenkins fix the problem? If yes, then there's a bug in packaging; otherwise there's a bug in the JRI generation process and it may indeed be a missing jmod.

jabrena, 2020-05-17:
I used the complete installation and it works, I will test next week with Java 14 JRI from Jenkins.
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-14/lastSuccessfulBuild/artifact/build/jri-ev3.tar.gz

Currently, where the package install the JRI?

JakubVanek, 2020-05-17:
The Debian re-packaging was removed from Jenkins; it has to be done locally ([documentation](https://github.com/ev3dev-lang-java/openjdk-ev3/blob/master/docs/Packaging_signing_uploading.md)). I would then upload the locally-built package to ev3dev repository directly, there is no other hosting. As only Java 11 is uploaded there, it will be necessary to build Java 14 package separately.

The latest ev3dev package can be found directly on the repository server at [archive.ev3dev.org/debian/pool/main/j/jri-11-ev3/](http://archive.ev3dev.org/debian/pool/main/j/jri-11-ev3/).

jabrena, 2020-05-17:
It could be interesting to generate the package from Jenkins and later upload manually to EV3DEV.
In that way, the package is generated automatically and the unique manual process is the upload process. Can you enable the re-packaging again?

JakubVanek, 2020-05-17:
> It could be interesting to generate the package from Jenkins and later upload manually to EV3DEV.
> In that way, the package is generated automatically and the unique manual process is the upload process. Can you enable the re-packaging again?

Luckily this is controlled by Jenkinsfile which can be changed in the repository without admin permissions on AdoptOpenJDK Jenkins, so yes, I will try it.

jabrena, 2020-05-17:
Take your time, currently with a complete JDK, it is fixed :)

JakubVanek, 2020-05-19:
CA certs seem to be handled similarly on [Arch Linux](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=java8-openjdk#n190) as they are [here](https://github.com/ev3dev-lang-java/openjdk-ev3/blob/64e88ef5b66233ac322bcacf1d75acde671bba44/packaging/debian/rules#L55).

JakubVanek, 2020-05-19:
Exception is thrown here: https://github.com/openjdk/jdk11u/blob/f957822b4172adf4f3d08431622bcfefecb864a3/src/java.base/share/classes/sun/security/ssl/TransportContext.java#L307

JakubVanek, 2020-05-19:
Further diagnosis could be done with the `-Djavax.net.debug=all` switch: https://stackoverflow.com/questions/54750922/java-ssl-handshake-failure

The answer also mentions another possible cause: that only the US export-gradle encryption suites are enabled by default. https://stackoverflow.com/a/57752847
GitHub is using TLS1.2+TLS1.3: https://www.ssllabs.com/ssltest/analyze.html?d=github.com

jabrena, 2020-05-19:
But why using the complete JDK, the test passed?

JakubVanek, 2020-05-19:
> But why using the complete JDK, the test passed?

I am not sure, but I suspect some configuration file may not get copied to the debian package (https://github.com/ev3dev-lang-java/openjdk-ev3/blob/64e88ef5b66233ac322bcacf1d75acde671bba44/packaging/debian/rules#L53).

jabrena, 2020-05-19:
Testing XD

jabrena, 2020-05-24:
Process to test the JRI:

```
wget https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-14/lastSuccessfulBuild/artifact/build/jri-ev3.tar.gz
sudo tar -zxvf jdk-ev3.tar.gz -C /opt
sudo mv /opt/jdk/ /opt/jdk-14
sudo update-alternatives --install /usr/bin/java java /opt/jdk-14/bin/java 2014
java -version
```

jabrena, 2020-05-24:
Verified instalation:

```
robot@ev3dev:/opt$ sudo update-alternatives --config java  
There are 3 choices for the alternative java (providing /usr/bin/java).

  Selection    Path                              Priority   Status
------------------------------------------------------------
* 0            /opt/jri-14/bin/java               2024      auto mode
  1            /opt/jdk-14/bin/java               2014      manual mode
  2            /opt/jri-14/bin/java               2024      manual mode
  3            /usr/lib/jvm/jri-11-ev3/bin/java   1106      manual mode

```

jabrena, 2020-05-24:
Hi @JakubVanek,

I can confirm that JRI 14, has the same issue:

```
2020-05-24 16:23:13 [main] WARN com.github.broker.GitClientWrapper - https://github.com/broker-game/broker-dev-environment10: Secure connection to https://github.com/broker-game/broker-dev-environment10 could not be established because of SSL problems
org.eclipse.jgit.api.errors.TransportException: https://github.com/broker-game/broker-dev-environment10: Secure connection to https://github.com/broker-game/broker-dev-environment10 could not be established because of SSL problems
        at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:222)
        at org.eclipse.jgit.api.CloneCommand.fetch(CloneCommand.java:292)
        at org.eclipse.jgit.api.CloneCommand.call(CloneCommand.java:176)
        at com.github.broker.GitClientWrapper.cloneRepository(GitClientWrapper.java:49)
        at com.github.broker.BrokerClient.connect(BrokerClient.java:85)
        at example.PingApp.main(PingApp.java:20)
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/broker-game/broker-dev-environment10: Secure connection to https://github.com/broker-game/broker-dev-environment10 could not be established because of SSL problems
        at org.eclipse.jgit.transport.TransportHttp.handleSslFailure(TransportHttp.java:675)
        at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:576)
        at org.eclipse.jgit.transport.TransportHttp.openFetch(TransportHttp.java:359)
        at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:105)
        at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:91)
        at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1260)
        at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:211)
        ... 5 more
Caused by: javax.net.ssl.SSLHandshakeException: Received fatal alert: handshake_failure
        at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:131)
        at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:117)
        at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:311)
        at java.base/sun.security.ssl.Alert$AlertConsumer.consume(Alert.java:291)
        at java.base/sun.security.ssl.TransportContext.dispatch(TransportContext.java:184)
        at java.base/sun.security.ssl.SSLTransport.decode(SSLTransport.java:167)
        at java.base/sun.security.ssl.SSLSocketImpl.decode(SSLSocketImpl.java:1462)
        at java.base/sun.security.ssl.SSLSocketImpl.readHandshakeRecord(SSLSocketImpl.java:1370)
        at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:437)
        at java.base/sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:567)
        at java.base/sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:171)
        at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1595)
        at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1523)
        at java.base/java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:527)
        at java.base/sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:308)
        at org.eclipse.jgit.transport.http.JDKHttpConnection.getResponseCode(JDKHttpConnection.java:85)
        at org.eclipse.jgit.util.HttpSupport.response(HttpSupport.java:208)
        at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:507)
        ... 10 more
2020-05-24 16:23:16 [main] INFO com.github.broker.BrokerClient - 1590337394597_PING_PING.json
Exception in thread "main" java.lang.NullPointerException
        at com.github.broker.GitClientWrapper.upgradeRepository(GitClientWrapper.java:68)
        at com.github.broker.BrokerClient.produce(BrokerClient.java:100)
        at example.PingApp.lambda$main$0(PingApp.java:26)
        at java.base/java.util.stream.Streams$RangeIntSpliterator.forEachRemaining(Streams.java:104)
        at java.base/java.util.stream.IntPipeline$Head.forEach(IntPipeline.java:593)
        at example.PingApp.main(PingApp.java:26)

real    1m29.757s
user    1m1.757s
sys     0m6.610s
[ERROR] Command execution failed.
org.apache.commons.exec.ExecuteException: Process exited with an error: 1 (Exit value: 1)
    at org.apache.commons.exec.DefaultExecutor.executeInternal (DefaultExecutor.java:404)
    at org.apache.commons.exec.DefaultExecutor.execute (DefaultExecutor.java:166)
    at org.codehaus.mojo.exec.ExecMojo.executeCommandLine (ExecMojo.java:804)
    at org.codehaus.mojo.exec.ExecMojo.executeCommandLine (ExecMojo.java:751)
    at org.codehaus.mojo.exec.ExecMojo.execute (ExecMojo.java:313)
    at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo (DefaultBuildPluginManager.java:137)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:210)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:156)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:148)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:117)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:81)
[... cut ...]
2020-05-24 16:28:54 [main] INFO com.github.broker.GitClientWrapper - 
2020-05-24 16:28:55 [main] INFO com.github.broker.BrokerClient - 1590337735121_PING_PING.json
2020-05-24 16:29:13 [main] INFO com.github.broker.GitClientWrapper - 
2020-05-24 16:29:13 [main] INFO com.github.broker.BrokerClient - 1590337753658_PING_PING.json
2020-05-24 16:29:28 [main] INFO com.github.broker.GitClientWrapper - 
2020-05-24 16:29:28 [main] INFO com.github.broker.BrokerClient - 1590337768411_PING_PING.json
```

Can you review what is java module is missing in the JRI?

JakubVanek, 2020-05-24:
Hi,

yup, I will look into it. Could you also please try running the java command with `-Djavax.net.debug=all`? I think that might give us clearer picture about what is happening there.

Jakub

JakubVanek, 2020-05-24:
JRI14 contains the following modules:
```
java.base@14
java.compiler@14
java.datatransfer@14
java.desktop@14
java.instrument@14
java.logging@14
java.management@14
java.management.rmi@14
java.naming@14
java.net.http@14
java.prefs@14
java.rmi@14
java.scripting@14
java.se@14
java.security.jgss@14
java.security.sasl@14
java.sql@14
java.sql.rowset@14
java.transaction.xa@14
java.xml@14
java.xml.crypto@14
jdk.jartool@14
jdk.jdwp.agent@14
jdk.management.agent@14
jdk.unsupported@14
```

JDK14 contains the following modules:
```
java.base@14
java.compiler@14
java.datatransfer@14
java.desktop@14
java.instrument@14
java.logging@14
java.management@14
java.management.rmi@14
java.naming@14
java.net.http@14
java.prefs@14
java.rmi@14
java.scripting@14
java.se@14
java.security.jgss@14
java.security.sasl@14
java.smartcardio@14
java.sql@14
java.sql.rowset@14
java.transaction.xa@14
java.xml@14
java.xml.crypto@14
jdk.accessibility@14
jdk.attach@14
jdk.charsets@14
jdk.compiler@14
jdk.crypto.cryptoki@14
jdk.crypto.ec@14
jdk.dynalink@14
jdk.editpad@14
jdk.hotspot.agent@14
jdk.httpserver@14
jdk.incubator.foreign@14
jdk.incubator.jpackage@14
jdk.internal.ed@14
jdk.internal.jvmstat@14
jdk.internal.le@14
jdk.internal.opt@14
jdk.jartool@14
jdk.javadoc@14
jdk.jcmd@14
jdk.jconsole@14
jdk.jdeps@14
jdk.jdi@14
jdk.jdwp.agent@14
jdk.jfr@14
jdk.jlink@14
jdk.jshell@14
jdk.jsobject@14
jdk.jstatd@14
jdk.localedata@14
jdk.management@14
jdk.management.agent@14
jdk.management.jfr@14
jdk.naming.dns@14
jdk.naming.rmi@14
jdk.net@14
jdk.nio.mapmode@14
jdk.rmic@14
jdk.scripting.nashorn@14
jdk.scripting.nashorn.shell@14
jdk.sctp@14
jdk.security.auth@14
jdk.security.jgss@14
jdk.unsupported@14
jdk.unsupported.desktop@14
jdk.xml.dom@14
jdk.zipfs@14
```

JakubVanek, 2020-05-24:
It seems that the culprit is the `jdk.crypto.ec` module: https://stackoverflow.com/a/55517159

jabrena, 2020-05-24:
Hi @JakubVanek, 

this is the Output:

```
robot@ev3dev:~$ java -jar template-project-maven-0.1.0-SNAPSHOT-jar-with-dependencies.jar -Djavax.net.debug=all
/tmp/BROKER_CLIENT_PING_15382006138865850816
2020-05-24 16:41:45 [main] WARN com.github.broker.GitClientWrapper - https://github.com/broker-game/broker-dev-environment10: Secure connection to https://github.com/broker-game/broker-dev-environment10 could not be established because of SSL problems
org.eclipse.jgit.api.errors.TransportException: https://github.com/broker-game/broker-dev-environment10: Secure connection to https://github.com/broker-game/broker-dev-environment10 could not be established because of SSL problems
        at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:222)
        at org.eclipse.jgit.api.CloneCommand.fetch(CloneCommand.java:292)
        at org.eclipse.jgit.api.CloneCommand.call(CloneCommand.java:176)
        at com.github.broker.GitClientWrapper.cloneRepository(GitClientWrapper.java:49)
        at com.github.broker.BrokerClient.connect(BrokerClient.java:85)
        at example.PingApp.main(PingApp.java:20)
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/broker-game/broker-dev-environment10: Secure connection to https://github.com/broker-game/broker-dev-environment10 could not be established because of SSL problems
        at org.eclipse.jgit.transport.TransportHttp.handleSslFailure(TransportHttp.java:675)
        at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:576)
        at org.eclipse.jgit.transport.TransportHttp.openFetch(TransportHttp.java:359)
        at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:105)
        at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:91)
        at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1260)
        at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:211)
        ... 5 more
Caused by: javax.net.ssl.SSLHandshakeException: Received fatal alert: handshake_failure
        at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:131)
        at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:117)
        at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:311)
        at java.base/sun.security.ssl.Alert$AlertConsumer.consume(Alert.java:291)
        at java.base/sun.security.ssl.TransportContext.dispatch(TransportContext.java:184)
        at java.base/sun.security.ssl.SSLTransport.decode(SSLTransport.java:167)
        at java.base/sun.security.ssl.SSLSocketImpl.decode(SSLSocketImpl.java:1462)
        at java.base/sun.security.ssl.SSLSocketImpl.readHandshakeRecord(SSLSocketImpl.java:1370)
        at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:437)
        at java.base/sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:567)
        at java.base/sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:171)
        at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1595)
        at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1523)
        at java.base/java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:527)
        at java.base/sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:308)
        at org.eclipse.jgit.transport.http.JDKHttpConnection.getResponseCode(JDKHttpConnection.java:85)
        at org.eclipse.jgit.util.HttpSupport.response(HttpSupport.java:208)
        at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:507)
        ... 10 more
2020-05-24 16:41:47 [main] INFO com.github.broker.BrokerClient - 1590338506502_PING_PING.json
Exception in thread "main" java.lang.NullPointerException
        at com.github.broker.GitClientWrapper.upgradeRepository(GitClientWrapper.java:68)
        at com.github.broker.BrokerClient.produce(BrokerClient.java:100)
        at example.PingApp.lambda$main$0(PingApp.java:26)
        at java.base/java.util.stream.Streams$RangeIntSpliterator.forEachRemaining(Streams.java:104)
        at java.base/java.util.stream.IntPipeline$Head.forEach(IntPipeline.java:593)
        at example.PingApp.main(PingApp.java:26)
```

jabrena, 2020-05-24:
Can you create a PR for Java 14 to add both modules in the JRI?

```
jdk.crypto.cryptoki@14
jdk.crypto.ec@14
```

JakubVanek, 2020-05-24:
SSL Labs test confirms the issue, the preferred key exchange is through elliptic curves. Additionally, since Java 11 supports TLS 1.3 (according to the handshake simulation), it locked out itself from using the less-secure non-EC suites.
[image]

JakubVanek, 2020-05-24:
> `jdk.crypto.cryptoki@14`

This module may not make much sense - opencryptoki is a software for supporting smart card operations (PKCS #11 according to https://docs.oracle.com/en/java/javase/11/docs/api/jdk.crypto.cryptoki/module-summary.html).

JakubVanek, 2020-05-24:
The fix can be tested by copying the jlink command from https://github.com/ev3dev-lang-java/openjdk-ev3/blob/master/scripts/zip.sh and generating a new JRI this way:
```sh
cd /tmp
wget https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-14/lastSuccessfulBuild/artifact/build/jmods-ev3.tar.gz
tar -xf jmods-ev3.tar.gz
sudo /opt/jdk-14/bin/jlink \
   --module-path /tmp/jmods \
   --endian little \
   --compress 0 \
   --no-header-files \
   --no-man-pages \
   --add-modules "java.se,jdk.jdwp.agent,jdk.unsupported,jdk.management.agent,jdk.jartool,jdk.crypto.ec" \
   --output /opt/jri-14-tls
sudo update-alternatives --install /usr/bin/java java /opt/jri-14-tls/bin/java 2025
```

JakubVanek, 2020-05-24:
PR: https://github.com/ev3dev-lang-java/openjdk-ev3/pull/57

jabrena, 2020-05-24:
Hi @JakubVanek,

Is it possible if you test this time the issue with the build?

JakubVanek, 2020-05-24:
I currently cannot (VPS with Ubuntu 20.04 has some problems with running jlink in QEMU & I don't want to download hundreds of MBs through uplink at home), but tomorrow around this time the periodic build of https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-14/ will have finished, so then I can look into that.

jabrena, 2020-05-24:
Yes no problem, next week is Ok. 
The good stuff, we have oriented the issue and we learnt a new stuff from Java :)

JakubVanek, 2020-05-25:
To be exact, the problem is *SSL handshake; SSH is only used as a transport for commands & messages on stdout/stderr.

JakubVanek, 2020-05-25:
JRI14 is ready: [jri-14-ev3_14.0.0~36-1_armel.deb](https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-14/lastSuccessfulBuild/artifact/build/pkg/jri-14-ev3_14.0.0%7E36-1_armel.deb). I couldn't find the `PingApp` mainclass in the https://github.com/broker-game/ organization, so I cannot test the issue using the original problematic code. However the JRI debpkg now contains the `jdk.crypto.ec@14` module.

jabrena, 2020-05-25:
Hi @JakubVanek,

I have just uploaded an example here:
https://github.com/broker-game/template-project-maven

in src/main/resources/ you will find a config.properties where you have to change the values:
https://github.com/broker-game/template-project-maven/blob/master/src/main/resources/config.properties

```
BROKER=https://github.com/broker-game/broker-dev-environment
APPLICATION=PINGPONG20
NODE=PING
FULLNAME=YOUR FULL NAME
EMAIL=YOUR EMAIL
USER=DEMO
PASSWORD=DEMO
``` 

From a terminal in Intellij, execute: `mvn -Pbrick clean package` and voila :)

JakubVanek, 2020-05-25:
The old error is gone, but there is now a different one:
```
root@3da050cadd75:/build# java -jar template-project-maven-0.1.0-SNAPSHOT-jar-with-dependencies.jar 
/tmp/BROKER_CLIENT_PING_6086654752791614609
2020-05-25 18:38:57 [main] WARN com.github.broker.GitClientWrapper - https://github.com/broker-game/broker-dev-environment: cannot open git-upload-pack
org.eclipse.jgit.api.errors.TransportException: https://github.com/broker-game/broker-dev-environment: cannot open git-upload-pack
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:222)
	at org.eclipse.jgit.api.CloneCommand.fetch(CloneCommand.java:292)
	at org.eclipse.jgit.api.CloneCommand.call(CloneCommand.java:176)
	at com.github.broker.GitClientWrapper.cloneRepository(GitClientWrapper.java:49)
	at com.github.broker.BrokerClient.connect(BrokerClient.java:85)
	at example.PingApp.main(PingApp.java:20)
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/broker-game/broker-dev-environment: cannot open git-upload-pack
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:593)
	at org.eclipse.jgit.transport.TransportHttp.openFetch(TransportHttp.java:359)
	at org.eclipse.jgit.transport.FetchProcess.executeImp(FetchProcess.java:105)
	at org.eclipse.jgit.transport.FetchProcess.execute(FetchProcess.java:91)
	at org.eclipse.jgit.transport.Transport.fetch(Transport.java:1260)
	at org.eclipse.jgit.api.FetchCommand.call(FetchCommand.java:211)
	... 5 more
Caused by: javax.net.ssl.SSLException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty
	at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:133)
	at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:324)
	at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:267)
	at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:262)
	at java.base/sun.security.ssl.SSLSocketImpl.handleException(SSLSocketImpl.java:1654)
	at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:443)
	at java.base/sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:567)
	at java.base/sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:171)
	at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1595)
	at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1523)
	at java.base/java.net.HttpURLConnection.getResponseCode(HttpURLConnection.java:527)
	at java.base/sun.net.www.protocol.https.HttpsURLConnectionImpl.getResponseCode(HttpsURLConnectionImpl.java:308)
	at org.eclipse.jgit.transport.http.JDKHttpConnection.getResponseCode(JDKHttpConnection.java:85)
	at org.eclipse.jgit.util.HttpSupport.response(HttpSupport.java:208)
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:507)
	... 10 more
Caused by: java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty
	at java.base/sun.security.validator.PKIXValidator.<init>(PKIXValidator.java:101)
	at java.base/sun.security.validator.Validator.getInstance(Validator.java:181)
	at java.base/sun.security.ssl.X509TrustManagerImpl.getValidator(X509TrustManagerImpl.java:309)
	at java.base/sun.security.ssl.X509TrustManagerImpl.checkTrustedInit(X509TrustManagerImpl.java:183)
	at java.base/sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:198)
	at java.base/sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:132)
	at java.base/sun.security.ssl.CertificateMessage$T13CertificateConsumer.checkServerCerts(CertificateMessage.java:1324)
	at java.base/sun.security.ssl.CertificateMessage$T13CertificateConsumer.onConsumeCertificate(CertificateMessage.java:1215)
	at java.base/sun.security.ssl.CertificateMessage$T13CertificateConsumer.consume(CertificateMessage.java:1158)
	at java.base/sun.security.ssl.SSLHandshake.consume(SSLHandshake.java:396)
	at java.base/sun.security.ssl.HandshakeContext.dispatch(HandshakeContext.java:444)
	at java.base/sun.security.ssl.HandshakeContext.dispatch(HandshakeContext.java:422)
	at java.base/sun.security.ssl.TransportContext.dispatch(TransportContext.java:181)
	at java.base/sun.security.ssl.SSLTransport.decode(SSLTransport.java:167)
	at java.base/sun.security.ssl.SSLSocketImpl.decode(SSLSocketImpl.java:1462)
	at java.base/sun.security.ssl.SSLSocketImpl.readHandshakeRecord(SSLSocketImpl.java:1370)
	at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:437)
	... 19 more
Caused by: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty
	at java.base/java.security.cert.PKIXParameters.setTrustAnchors(PKIXParameters.java:200)
	at java.base/java.security.cert.PKIXParameters.<init>(PKIXParameters.java:120)
	at java.base/java.security.cert.PKIXBuilderParameters.<init>(PKIXBuilderParameters.java:104)
	at java.base/sun.security.validator.PKIXValidator.<init>(PKIXValidator.java:98)
	... 35 more
2020-05-25 18:38:57 [main] INFO com.github.broker.BrokerClient - 1590431937153_PING_PING.json
Exception in thread "main" java.lang.NullPointerException
	at com.github.broker.GitClientWrapper.upgradeRepository(GitClientWrapper.java:68)
	at com.github.broker.BrokerClient.produce(BrokerClient.java:100)
	at example.PingApp.lambda$main$0(PingApp.java:26)
	at java.base/java.util.stream.Streams$RangeIntSpliterator.forEachRemaining(Streams.java:104)
	at java.base/java.util.stream.IntPipeline$Head.forEach(IntPipeline.java:593)
	at example.PingApp.main(PingApp.java:26)
```

JakubVanek, 2020-05-25:
The issue has to do with broken `ca-certificates-java` dependency: https://github.com/AdoptOpenJDK/openjdk-installer/issues/105. This causes the cacerts db to get corrupted.

After putting a dummy link from `/bin/java` to `/usr/lib/jvm/jri-14-ev3/bin/java` and reinstalling Java from scratch, it works.
```
root@3da050cadd75:/build# java -jar template-project-maven-0.1.0-SNAPSHOT-jar-with-dependencies.jar 
/tmp/BROKER_CLIENT_PING_17635740478772410155
2020-05-25 18:49:01 [main] INFO com.github.broker.BrokerClient - 1590432541949_PING_PING.json
2020-05-25 18:49:02 [main] INFO com.github.broker.GitClientWrapper - Waiting for Event in : PINGPONG20
2020-05-25 18:49:02 [main] WARN com.github.broker.GitClientWrapper - Remote origin did not advertise Ref for branch PINGPONG20. This Ref may not exist in the remote or may be hidden by permission settings.
2020-05-25 18:49:03 [main] WARN com.github.broker.GitClientWrapper - https://github.com/broker-game/broker-dev-environment: git-receive-pack not permitted on 'https://github.com/broker-game/broker-dev-environment/'
org.eclipse.jgit.api.errors.TransportException: https://github.com/broker-game/broker-dev-environment: git-receive-pack not permitted on 'https://github.com/broker-game/broker-dev-environment/'
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:147)
	at com.github.broker.GitClientWrapper.push(GitClientWrapper.java:131)
	at com.github.broker.BrokerClient.produce(BrokerClient.java:102)
	at example.PingApp.lambda$main$0(PingApp.java:26)
	at java.base/java.util.stream.Streams$RangeIntSpliterator.forEachRemaining(Streams.java:104)
	at java.base/java.util.stream.IntPipeline$Head.forEach(IntPipeline.java:593)
	at example.PingApp.main(PingApp.java:26)
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/broker-game/broker-dev-environment: git-receive-pack not permitted on 'https://github.com/broker-game/broker-dev-environment/'
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:545)
	at org.eclipse.jgit.transport.TransportHttp.openPush(TransportHttp.java:434)
	at org.eclipse.jgit.transport.PushProcess.execute(PushProcess.java:127)
	at org.eclipse.jgit.transport.Transport.push(Transport.java:1335)
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:137)
	... 6 more
2020-05-25 18:49:03 [main] INFO com.github.broker.BrokerClient - 1590432543901_PING_PING.json
2020-05-25 18:49:04 [main] INFO com.github.broker.GitClientWrapper - Waiting for Event in : PINGPONG20
2020-05-25 18:49:04 [main] WARN com.github.broker.GitClientWrapper - Remote origin did not advertise Ref for branch PINGPONG20. This Ref may not exist in the remote or may be hidden by permission settings.
2020-05-25 18:49:05 [main] WARN com.github.broker.GitClientWrapper - https://github.com/broker-game/broker-dev-environment: git-receive-pack not permitted on 'https://github.com/broker-game/broker-dev-environment/'
org.eclipse.jgit.api.errors.TransportException: https://github.com/broker-game/broker-dev-environment: git-receive-pack not permitted on 'https://github.com/broker-game/broker-dev-environment/'
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:147)
	at com.github.broker.GitClientWrapper.push(GitClientWrapper.java:131)
	at com.github.broker.BrokerClient.produce(BrokerClient.java:102)
	at example.PingApp.lambda$main$0(PingApp.java:26)
	at java.base/java.util.stream.Streams$RangeIntSpliterator.forEachRemaining(Streams.java:104)
	at java.base/java.util.stream.IntPipeline$Head.forEach(IntPipeline.java:593)
	at example.PingApp.main(PingApp.java:26)
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/broker-game/broker-dev-environment: git-receive-pack not permitted on 'https://github.com/broker-game/broker-dev-environment/'
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:545)
	at org.eclipse.jgit.transport.TransportHttp.openPush(TransportHttp.java:434)
	at org.eclipse.jgit.transport.PushProcess.execute(PushProcess.java:127)
	at org.eclipse.jgit.transport.Transport.push(Transport.java:1335)
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:137)
	... 6 more
2020-05-25 18:49:05 [main] INFO com.github.broker.BrokerClient - 1590432545979_PING_PING.json
2020-05-25 18:49:06 [main] INFO com.github.broker.GitClientWrapper - Waiting for Event in : PINGPONG20
2020-05-25 18:49:06 [main] WARN com.github.broker.GitClientWrapper - Remote origin did not advertise Ref for branch PINGPONG20. This Ref may not exist in the remote or may be hidden by permission settings.
2020-05-25 18:49:07 [main] WARN com.github.broker.GitClientWrapper - https://github.com/broker-game/broker-dev-environment: git-receive-pack not permitted on 'https://github.com/broker-game/broker-dev-environment/'
org.eclipse.jgit.api.errors.TransportException: https://github.com/broker-game/broker-dev-environment: git-receive-pack not permitted on 'https://github.com/broker-game/broker-dev-environment/'
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:147)
	at com.github.broker.GitClientWrapper.push(GitClientWrapper.java:131)
	at com.github.broker.BrokerClient.produce(BrokerClient.java:102)
	at example.PingApp.lambda$main$0(PingApp.java:26)
	at java.base/java.util.stream.Streams$RangeIntSpliterator.forEachRemaining(Streams.java:104)
	at java.base/java.util.stream.IntPipeline$Head.forEach(IntPipeline.java:593)
	at example.PingApp.main(PingApp.java:26)
[... cut ...]
Caused by: org.eclipse.jgit.errors.TransportException: https://github.com/broker-game/broker-dev-environment: git-receive-pack not permitted on 'https://github.com/broker-game/broker-dev-environment/'
	at org.eclipse.jgit.transport.TransportHttp.connect(TransportHttp.java:545)
	at org.eclipse.jgit.transport.TransportHttp.openPush(TransportHttp.java:434)
	at org.eclipse.jgit.transport.PushProcess.execute(PushProcess.java:127)
	at org.eclipse.jgit.transport.Transport.push(Transport.java:1335)
	at org.eclipse.jgit.api.PushCommand.call(PushCommand.java:137)
	... 6 more
```

jabrena, 2020-05-25:
It is part of the journey. 

I recommend to change the parameter: `BROKER` you could create a empty repo in your user:
https://github.com/JakubVanek/broker-dev-environment

and the test could finish without any exception. In the last execution appear the typical credential error.

JakubVanek, 2020-05-25:
> and the test could finish without any exception. In the last execution appear the typical credential error.
Yup, that would likely fix it, however the fact that these exceptions appear is enough to confirm that the SSL connection has been successfully made.

jabrena, 2020-05-25:
What is the next step?

JakubVanek, 2020-05-25:
Currently https://github.com/ev3dev-lang-java/openjdk-ev3/pull/61 should resolve the issue at hand, but https://github.com/AdoptOpenJDK/openjdk-support/issues/13 is going to be interesting for us in the near future.

JakubVanek, 2020-05-25:
Actually I think that I will create another PR for generating the cacerts DB from Mozilla cacert database as that would seem to provide the best user experience.

JakubVanek, 2020-05-25:
https://github.com/ev3dev-lang-java/openjdk-ev3/pull/62 is ready for providing newest cacerts.

jabrena, 2020-12-08:
Know issue

closed 2020-12-08 by jabrena