ev3dev-lang-java/ev3dev-lang-java #731: Rebuild JRI11.0.6 for Debian repositories with CAcert fixes

JakubVanek, 2020-05-26:
Reference: https://github.com/ev3dev-lang-java/openjdk-ev3/pull/64#issuecomment-633725341

Currently the `jri-11-ev3` package in ev3dev does not install properly, it leaves itself half-configured and with a damaged trust store (this unfortunately ships in the ev3dev image). I think that it would be a good idea to:
 - make new installations fully succeed (https://github.com/ev3dev-lang-java/openjdk-ev3/pull/61 or https://github.com/ev3dev-lang-java/openjdk-ev3/pull/64 should do so),
 - fix the trust store for existing installations (https://github.com/ev3dev-lang-java/openjdk-ev3/pull/61 should work; https://github.com/ev3dev-lang-java/openjdk-ev3/pull/64 would require clean Java installation).

I want to rebuild the JRI11.0.6 locally after one of the fixes above is merged and then upload the fixed package as a new package revision (`-2`).

Using Mozilla's certificates (https://github.com/ev3dev-lang-java/openjdk-ev3/pull/62) seems somewhat orthogonal to this. Without that PR, the following cacerts file will be (and currently are) used: https://github.com/AdoptOpenJDK/openjdk-build/blob/master/security/cacerts. While they're not the most up-to-date, they seem to work fine for GitHub as discovered in https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/728.

JakubVanek, 2020-05-26:
Clean Java installation equals to running the following:
```sh
sudo apt autoremove --purge jri-11-ev3 ca-certificates-java
sudo apt install jri-11-ev3
```

JakubVanek, 2020-05-26:
On the other hand, the minimal upgrade command to install fixed JRI with its own CA certificates is not that different; we only get away from the need to purge the packages.
```
sudo apt update
sudo apt install jri-11-ev3 # this works as a single-package update too
```

JakubVanek, 2020-05-26:
Countering the previous argument, now that JRI debpkgs are available from Jenkins, the update by an end-user to a newer version can also be made like this:
```sh
wget https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-14/lastSuccessfulBuild/artifact/build/pkg/jri-14-ev3_14.0.0~36-1_armel.deb
sudo dpkg -i jri-14-ev3_14.0.0~36-1_armel.deb
```

When installing on an ev3dev system with damaged system trust store, the clean reinstallation of `ca-certificates-java` would have to be done separately with https://github.com/ev3dev-lang-java/openjdk-ev3/pull/64. With https://github.com/ev3dev-lang-java/openjdk-ev3/pull/61, the system trust store would not be used anymore, instead the Mozilla/OpenJDK cacerts would be used.

JakubVanek, 2020-05-26:
Summary:
https://github.com/ev3dev-lang-java/openjdk-ev3/pull/61 _drop dependency on ca-certificates-java_
 * drops Debian-style integration with system CA certificate store (Java keystore is automatically generated from CA certificates stored in /etc/ssl/certs there)
  - this could confuse some users that would rely on this
  - instead of storing the CA certificates in the system and running `update-ca-certificates`, users would have to update the /usr/lib/jvm/jri-N-ev3/lib/security/cacerts file manually via keytool/other util
 * but it fixes new and existing installations with new packages by ignoring the damaged /etc/ssl/certs/java/cacerts and instead shipping with a separate non-automatically-generated trust store
https://github.com/ev3dev-lang-java/openjdk-ev3/pull/64 _add jri-N-ev3 self-dep to fix ca-certificates-java_
 * maintains Debian-style integration and fixes it so that it works properly
 * it can only fix new installations; for existing installations, the `ca-certificates-java` package has to be removed and then reinstalled

JakubVanek, 2020-05-26:
Java is not the only runtime that can use autogenerated key/trust stores, Mono/C# does this too.

Mechanism: http://manpages.ubuntu.com/manpages/xenial/man8/update-ca-certificates.8.html (Java puts a hook in /etc/ca-certificates/update.d)

JakubVanek, 2020-05-27:
Summary of the 61, 62 & 63 PRs: https://github.com/ev3dev-lang-java/openjdk-ev3/pull/61#issuecomment-634302762

JakubVanek, 2020-05-27:
How the PRs relate to the current status:

JRI debpkg problems:
 * https://github.com/ev3dev-lang-java/openjdk-ev3/pull/61 _drop dependency on ca-certificates-java_
    * this fixes CA certificates in JRI debpkg by not using system-managed `/etc/ssl/certs/java/cacerts` and rather shipping and using the same cacerts as the ones in JDK/JRI tarpkg
       * we lose Debian-style integration with system CA certificate store
         * this may confuse some users if they relied on it
         * instead of storing the CA certificates in the system and running `update-ca-certificates`, users would have to update the /usr/lib/jvm/jri-N-ev3/lib/security/cacerts file manually via keytool/other util
       * on the other hand, simple upgrade of `jri-N-ev3` to any newer major or minor version would fix the problem directly, although _ only for that version_
         * this can be done by downloading JRI debpkg from Jenkins and installing it via DPKG
         * alternatively `apt-get update && apt-get dist-upgrade` would work too
 * https://github.com/ev3dev-lang-java/openjdk-ev3/pull/64 _add jri-N-ev3 self-dep to fix ca-certificates-java_
    * this fixes CA certificates in JRI debpkg by ensuring that `/etc/ssl/certs/java/cacerts` gets generated correctly, but it cannot (yet) fix the broken state
      * it keeps the Debian-style integration and fixes it so that it works properly
      * this can only fix new installation (= ev3dev images because JRI is pre-installed)
      * users can fix the existing trust stores in old versions by running the following commands (even without any of the PRs being merged)
         ```sh
         sudo dpkg --purge --force-all ca-certificates-java
         sudo apt-get download ca-certificates-java
         sudo dpkg -i ./ca-certificates-java*
         ```

Source of CA certificates:
 * https://github.com/ev3dev-lang-java/openjdk-ev3/pull/62 _use CA certificates from Mozilla_
    * this changes the CA certificates that ship with JDK/JRI tarpkg and that would ship with JRI debpkg if #61 is picked
    * with this, Mozilla CA certificates would be used
 * https://github.com/ev3dev-lang-java/openjdk-ev3/pull/65 _use CA certificates from OpenJDK_
    * this changes the CA certificates that ship with JDK/JRI tarpkg and that would ship with JRI debpkg if #61 is picked
    * with this, upstream OpenJDK certificates would be used

jabrena, 2020-05-28:
Hi @JakubVanek, I will do a short summary:

- Current build doesn´t have SSL support

So, we need to build again the package. Is it possible to create a build to add the certificates inside? I read your comments and JRI has some issues linking with `ca-certificates-java`, so add the same certificates that from OpenJDK on the JRI. I am not sure if Mozilla CA certificates but we could begin with that certificates and evolve later. In the solution, we need to add few test to guarantee that we solved the issue.

Juan Antonio

JakubVanek, 2020-05-28:
Hi @jabrena,

> Hi @JakubVanek, I will do a short summary:
> 
>  * Current build doesn´t have SSL support

If "current build" == `jri-11-ev3` package/debpkg currently uploaded to ev3dev repository, then yes, it doesn't support SSL out of the box.

> So, we need to build again the package.

Yes. For this, I'd locally force the OpenJDK source version to `jdk-11.0.6+10` (last version in ev3dev repository) and with that I would locally perform a new full openjdk-ev3 build and then I would package that into a new JRI debpkg.

> Is it possible to create a build to add the certificates inside?

Yup, this is what https://github.com/ev3dev-lang-java/openjdk-ev3/pull/61 should enable. Any build done after the pull request is merged should support SSL properly.

> I read your comments and JRI has some issues linking with `ca-certificates-java`, so add the same certificates that from OpenJDK on the JRI.

I assume that OpenJDK == JDK/JRI tarpkg. In that case, the https://github.com/ev3dev-lang-java/openjdk-ev3/pull/61 should be merged and https://github.com/ev3dev-lang-java/openjdk-ev3/pull/64 should be closed.

> I am not sure if Mozilla CA certificates but we could begin with that certificates and evolve later.

I agree, at worst we can revert such a change. To go this way, https://github.com/ev3dev-lang-java/openjdk-ev3/pull/62 should be merged and https://github.com/ev3dev-lang-java/openjdk-ev3/pull/65 should be closed.

> In the solution, we need to add few test to guarantee that we solved the issue.

There is currently no mechanism for this, but I suppose as a stop-gap measure for _short_ tests like this we could add a new phase to the Jenkins `build` job. To do this properly, a separate Jenkins job with tests should be chained after the build succeeds (this was the idea behind the ["Pipelines"](https://ci.adoptopenjdk.net/view/ev3dev/job/eljpipes/) folder, although that route is not necessary). By doing it like so, we allow jobs doing testing to be scheduled on a less-powerful Jenkins node so that computing resources are not wasted. AdoptOpenJDK Jenkins has a `test`/`ci.role.test` node tag for this purpose.

> I will close tomorrow, previous PR.
> We only need one PR with the solution, not multiple PR.
> https://github.com/ev3dev-lang-java/openjdk-ev3/pulls

If you want to close all pull requests there, then that is in my opinion adding extra unnecessary work.

As far as the current decision goes, you'd only need to close https://github.com/ev3dev-lang-java/openjdk-ev3/pull/64 and https://github.com/ev3dev-lang-java/openjdk-ev3/pull/65 (rejected solutions for two different issues) and merge https://github.com/ev3dev-lang-java/openjdk-ev3/pull/61 and https://github.com/ev3dev-lang-java/openjdk-ev3/pull/62 (accepted solutions for two different issues).

> Juan Antonio

Jakub

jabrena, 2020-05-28:
Hi @JakubVanek,  close the PR that we are not going to merge and merge 2 pending PR into one to analyse better the change. Include tests for the support. Later, we will do together the publication on `ev3dev` repository

JakubVanek, 2020-05-28:
> Hi @JakubVanek,
> 
> close the PR that we are not going to merge and merge 2 pending PR into one to analyse better the change. Include tests for the support. Later, we will do together the publication on `ev3dev` repository

https://github.com/ev3dev-lang-java/openjdk-ev3/pull/66 created, but I will not add the tests yet.

jabrena, 2020-05-28:
Try to review if OpenJDK has some tests for SSH. This is the reason that the software add tests in the software life cycle. Meanwhile, I will review the rest of commits.

JakubVanek, 2020-05-29:
I have decided to go with a minimum viable testing, it is now added to the PR.

JakubVanek, 2020-05-29:
The test is based on https://github.com/ChristopherSchultz/ssltest. The test checks whether any of the TLSv1.3 suites if working with `https://github.com`. If not, it fails, otherwise it succeeds.

jabrena, 2020-05-29:
Hi @JakubVanek, 

OpenJDK has a specific test suite:
https://github.com/AdoptOpenJDK/openjdk-tests

What is the stopper to not use the official OpenJDK test suite with SSL support out of the box?

Juan Antonio

jabrena, 2020-05-29:
Please read the following articles about Testing before replying:

- https://outfresh.com/knowledge-base/6-famous-software-disasters-due-lack-testing/
- https://www.cigniti.com/blog/37-software-failures-inadequate-software-testing/
- https://www.qualified.io/blog/posts/why-developers-hate-coding-skills-tests
- http://tryqa.com/why-is-testing-necessary/

jabrena, 2020-05-29:
And the following articles from **AdoptOpenJDK**:

- https://blog.adoptopenjdk.net/2019/07/the-first-drop-introducing-adoptopenjdk-quality-assurance-aqa-v1-0/
- https://blog.adoptopenjdk.net/2018/08/test-result-summary-service-trss/
- https://blog.adoptopenjdk.net/2017/12/testing-java-help-count-ways/

jabrena, 2020-05-29:
[image]

JakubVanek, 2020-05-29:
I am sorry, but I will not add full testing support to openjdk-ev3. I do not want to resume my activities in it, I just wanted to help with one issue where the cause was unknown.

JakubVanek, 2020-05-29:
> Hi @JakubVanek,
> 
> OpenJDK has a specific test suite:
> https://github.com/AdoptOpenJDK/openjdk-tests
> 
> What is the stopper to not use the official OpenJDK test suite with SSL support out of the box?
> 
> Juan Antonio

One is that I have given up write rights in AdoptOpenJDK Jenkins and I do not want to gain them back.

JakubVanek, 2020-05-29:
> One is that I have given up write rights in AdoptOpenJDK Jenkins and I do not want to gain them back.

To clarify: full testing suite would take a long time and thus it would likely require new jobs in Jenkins.

jabrena, 2020-05-29:
This project was created to develop robots with Java. 
When you develop software with Java, the creation of software include `code + testing`.
The same with the JVM.

Try to add in the build an initial testing support, TRY IT @JakubVanek 
I did for OpenJDK10.

Today it is the SSL, tomorrow will be another stuff because the build itself is not enough.

jabrena, 2020-05-29:
Sorry @JakubVanek,

That distribution only supports LTS versions and and we build all versions. This is the key difference, so it is not an alternative for this issue.

JakubVanek, 2020-05-29:
Oki, however I still do not plan to engage in the project actively.

jabrena, 2020-05-30:
It is a pity to hear it.

I am not sure how many commits do you need to learn the importance about testing but I gave you 3 years to learn that value so now, the tolerance is zero. 

You are welcome to return if you want to help and this is your real intention but here, exist some Engineering rules. I try to be clear in this aspect.

I recommend another article to read:
https://martinfowler.com/bliki/TechnicalDebt.html

Juan Antonio

JakubVanek, 2020-06-15:
Hi,

I've thought about the concept a little today and I don't think that it would bring much value currently anyway. If the tests are added, some of them will certainly fail. Then there are two possibilities - either suppressing the problematic ones or fixing them. The first option reduces the value of the tests to only regression detection. That may indeed be useful, but it only stops the quality from getting lower. The second option brings more benefits, but it requires a commitment to fix the failing tests and an active maintainership. I do not want to take up such a commitment to this project, so a third party would have to do the fixing and that does not seem very likely.

Best regards,

Jakub

jabrena, 2020-06-16:
Hi @JakubVanek, 

if you remember in OpenJDK10 the infra was emulated but now we are compiling in a better environment so the test should pass. The previous comment is about the second option. Try to add the execution. Here existed the previous work:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/466
https://github.com/AdoptOpenJDK/openjdk-tests/issues/520

Try to execute only `make jdk_math`  only as a _Pilot_ and later, we will review the results.

**Note:** I celebrate that you review the case, you are going a good job in this area. :)