ev3dev-lang-java/ev3dev-lang-java #732: EV3 Large Motor is not moving. Sensors work fine.

chh456, 2020-06-14:
Hello,

can somebody help me out with a not moving EV3 large motor? My Java code looks similar to the example section and the sensors seem to work fine.

My setup:
- Intellij
- ev3dev-lang-java:2.5.3 via gradle
- BrickPi3 with an EV3 large motor on Motorport.B
- Touchsensor on S1
- Raspberry Pi

Code:
```
final EV3LargeRegulatedMotor motor = new EV3LargeRegulatedMotor(MotorPort.B);
motor.setSpeed(200);
motor.brake();
motor.forward();
try.. Thread.sleep(10000); ... catch
motor.stop();
```

Standardoutput of the RaspberryPi:
```
[ timestamp] lego-port port5: Removed device 'spi0.1:MB:lego-nxt-motor'
[ timestamp] tacho-motor motor18: Unregistered 'lego-nxt-motor' on 'spi0.1:MB' .
[ timestamp] tacho-motor motor19: Registered 'lego-nxt-motor' on 'spi0.1:MB' .
[ timestamp] lego-port port5: Added new device 'spi0.1:MB:lego-nxt-motor'
```

The motor is not moving but the sensor works fine.

What I tried so far:
- Switched motor ports
- Changed the EV3 large motor
- Changed the cable

Any ideas are welcome and thanks for your time.

jabrena, 2020-06-15:
Hi, 

Few questions:

- What raspberry pi version are you using?
- Do you know what firmware version are you using in BrickPi3?

Juan Antonio

chh456, 2020-06-15:
Thanks for your answer Juan Antonio. It's a Raspberry Pi 3 Model B Rev 1.2 and was recently flashed with ev3dev-stretch-rpi2-generic-2020-04-10.zip.

jabrena, 2020-06-16:
Hi @chh456,

Can you paste the output from the following commands: `ev3dev-sysinfo` and `dmesg`?

Can you paste the whole log from your program and can you paste the program?

I suspect that your firmware in Brickpi was not upgraded.

Juan Antonio

jabrena, 2020-07-03:
Hi @chh456,

did you solve the issue?

Juan Antonio

jabrena, 2020-07-08:
@chh456 any progress?

chh456, 2020-07-09:
Sorry I was not online for some days. Issue was solved by completely starting from scratch and using a new package of batteries. Thanks for your time and help!

closed 2020-07-09 by chh456