ev3dev-lang-java/ev3dev-lang-java #736: Getting started - WiFi "not available" after password

AlanKrueger, 2020-08-08:
I'm following the [Getting started with your Brick](https://ev3dev-lang-java.github.io/docs/support/getting_started/brick.html) docs for EV3 with Stretch.

I'm able to scan for WiFi networks, see mine, and select it, but by the time I'm done entering the password, the network is "not available". The network list will refresh, then I can select it again, but again it will apparently time out before I can get the password entered.

Is this in fact timing out? Any way I can past this issue?

AlanKrueger, 2020-08-09:
I've tried this on a separate WiFi network with a shorter password (all lowercase and no symbols), and that works. My home WiFi network password is a mixture of upper and lower case and symbols, and I can't get that to work.

jabrena, 2020-08-10:
mmmm, Yes, Maybe in the past I had that issue. If you don´t type faster with the virtual keyboard, I recommend to plug an USB Keyboard to EV3 and type directly. In that way, you should not reach the timeout.

Cheers

jabrena, 2020-08-13:
Do you continue with the issue? @AlanKrueger