ev3dev-lang-java/ev3dev-lang-java #737: MQTT -> Connect EV3 to Broker as a Subscriber (HiveMQ)

Kub-sn, 2020-08-09:
Im trying to connect my EV3 to the Broker on my Laptop. 
I am using the gradle-template so what I basically do is deploying my Subscriber Class as a gradle command and then remoteBrickRun. If I execute the class normally on the IDE, the subscription on the Broker is successfull. I am using the Broker from HiveMQ. 

Here is my Code Snippit:

`public class SubscriberEV3 extends Thread{
    static Mqtt5BlockingClient client;
    static Mqtt5BlockingClient clientError;

    public static void main(String... args) throws IOException {
        client = Mqtt5Client.builder()
                .identifier("Subscriber")
                .serverHost(InetAddress.getByName("192.168.0.9"))
                .serverPort(1883)
                .buildBlocking();
        client.connect();`

jabrena, 2020-08-10:
Hi @Kub-sn, what is the problem? do you have any issue in the console?

Juan Antonio

Kub-sn, 2020-08-10:
Here are the error messages:

ev3dev#1|Exception in thread "main" com.hivemq.client.mqtt.exceptions.ConnectionFailedException: io.netty.channel.ConnectTimeoutException: connection timed out: /192.168.0.9:1883
ev3dev#1|	at com.hivemq.client.internal.mqtt.MqttBlockingClient.connect(MqttBlockingClient.java:91)
ev3dev#1|	at com.hivemq.client.mqtt.mqtt5.Mqtt5BlockingClient.connect(Mqtt5BlockingClient.java:64)
ev3dev#1|	at example.SubOnly.main(SubOnly.java:21)
ev3dev#1|Caused by: io.netty.channel.ConnectTimeoutException: connection timed out: /192.168.0.9:1883
ev3dev#1|	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe$1.run(AbstractNioChannel.java:267)
ev3dev#1|	at io.netty.util.concurrent.PromiseTask$RunnableAdapter.call(PromiseTask.java:38)
ev3dev#1|	at io.netty.util.concurrent.ScheduledFutureTask.run(ScheduledFutureTask.java:127)
ev3dev#1|	at io.netty.util.concurrent.AbstractEventExecutor.safeExecute(AbstractEventExecutor.java:163)
ev3dev#1|	at io.netty.util.concurrent.SingleThreadEventExecutor.runAllTasks(SingleThreadEventExecutor.java:404)
ev3dev#1|	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:474)
ev3dev#1|	at io.netty.util.concurrent.SingleThreadEventExecutor$5.run(SingleThreadEventExecutor.java:909)
ev3dev#1|	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
ev3dev#1|	at java.base/java.lang.Thread.run(Thread.java:834)
ev3dev#1|
ev3dev#1|real	1m42.075s
ev3dev#1|user	0m0.317s
ev3dev#1|sys	0m0.295s

> Task :remoteBrickRun
Failed command ev3dev#1 with status 1: time brickrun -- java -jar /home/robot/template_project_gradle-2.6.0.jar 

> Task :remoteBrickRun FAILED

Kub-sn, 2020-08-10:
Here are the logs of the deployAndBrickRun command:

2020-08-10 22:52:01,942 [ 436715]   INFO - s.plugins.gradle.GradleManager - Instructing gradle to use java from C:\Program Files\JetBrains\IntelliJ IDEA Community Edition 2020.1\jbr 
2020-08-10 22:52:01,955 [ 436728]   INFO - s.plugins.gradle.GradleManager - Instructing gradle to use java from C:\Program Files\JetBrains\IntelliJ IDEA Community Edition 2020.1\jbr 
2020-08-10 22:52:02,072 [ 436845]   INFO - xecution.GradleExecutionHelper - Passing command-line args to Gradle Tooling API: --init-script C:\Users\kubiy\AppData\Local\Temp\ijresolvers.gradle 
2020-08-10 22:52:05,383 [ 440156]   INFO - rationStore.ComponentStoreImpl - Saving appFileTypeManager took 11 ms 
2020-08-10 22:52:05,467 [ 440240]   INFO - rationStore.ComponentStoreImpl - Saving Project (name=template_project_gradle, containerState=ACTIVE, componentStore=C:\Users\kubiy\Desktop\EV3-Maven\template-project-gradle) RunManager took 15 ms 
2020-08-10 22:56:27,969 [ 702742]   INFO - rationStore.ComponentStoreImpl - Saving appFileTypeManager took 12 ms, FindSettings took 66 ms, GeneralSettings took 14 ms, XDebuggerSettings took 56 ms 
2020-08-10 22:56:28,023 [ 702796]   INFO - rationStore.ComponentStoreImpl - Saving Module: 'template_project_gradle'ExternalFacetManager took 13 ms 
2020-08-10 22:56:28,214 [ 702987]   INFO - rationStore.ComponentStoreImpl - Saving Project (name=template_project_gradle, containerState=ACTIVE, componentStore=C:\Users\kubiy\Desktop\EV3-Maven\template-project-gradle) RunManager took 36 ms, com.intellij.coverage.CoverageDataManagerImpl took 11 ms, libraryTable took 31 ms 
2020-08-10 22:58:38,972 [ 833745]   INFO - rationStore.ComponentStoreImpl - Saving appDaemonCodeAnalyzerSettings took 12 ms, FindSettings took 57 ms, JavaCodeFoldingSettings took 13 ms 
2020-08-10 22:58:39,121 [ 833894]   INFO - rationStore.ComponentStoreImpl - Saving Project (name=template_project_gradle, containerState=ACTIVE, componentStore=C:\Users\kubiy\Desktop\EV3-Maven\template-project-gradle) Kotlin2JvmCompilerArguments took 18 ms 
2020-08-10 22:59:22,375 [ 877148]   WARN - ConfigurableExtensionPointUtil - ignore deprecated groupId: language for id: preferences.language.Kotlin.scripting 
2020-08-10 22:59:23,755 [ 878528]   INFO - ateSettings.impl.UpdateChecker - failed to load plugin descriptions from : Malformed URL:  
2020-08-10 22:59:54,870 [ 909643]   INFO - rationStore.ComponentStoreImpl - Saving appKeymapManager took 16 ms 
2020-08-10 22:59:54,933 [ 909706]   INFO - rationStore.ComponentStoreImpl - Saving Project (name=template_project_gradle, containerState=ACTIVE, componentStore=C:\Users\kubiy\Desktop\EV3-Maven\template-project-gradle) RunManager took 15 ms 
2020-08-10 23:01:35,520 [1010293]   INFO - j.ide.actions.RevealFileAction - Exit code 1 
2020-08-10 23:01:42,614 [1017387]   INFO - rationStore.ComponentStoreImpl - Saving Project (name=template_project_gradle, containerState=ACTIVE, componentStore=C:\Users\kubiy\Desktop\EV3-Maven\template-project-gradle) RunManager took 14 ms

jabrena, 2020-08-11:
Hi @Kub-sn,

did you `Google` a bit about this exception:

```
ev3dev#1|Exception in thread "main" com.hivemq.client.mqtt.exceptions.ConnectionFailedException: io.netty.channel.ConnectTimeoutException: connection timed out: /192.168.0.9:1883
ev3dev#1|	at com.hivemq.client.internal.mqtt.MqttBlockingClient.connect(MqttBlockingClient.java:91)
ev3dev#1|	at com.hivemq.client.mqtt.mqtt5.Mqtt5BlockingClient.connect(Mqtt5BlockingClient.java:64)
ev3dev#1|	at example.SubOnly.main(SubOnly.java:21)
ev3dev#1|Caused by: io.netty.channel.ConnectTimeoutException: connection timed out: /192.168.0.9:1883
```
 
`ConnectTimeoutException`

Juan Antonio

Kub-sn, 2020-08-11:
Hi @jabrena 

I did google the exception. Basically it means, as you already said, that the connection is not happening fast enough. I tried to increase the timeout on serverside, but I still get the same problem.  

> Remember that CPU/Memory capabilities at EV3 side is reduced, this is the main issue in the connection.

Well I actually have a PiStorms v2 too, so I hooked it up and tried it with that. I had half of a success:

My PiStorms could connect to the online Broker of HiveMQ easily. But I still get the same error if I try to connect to my broker on my notebook.

Do you have an idea what the problem could be?

Kub-sn, 2020-08-12:
I found the problem. Windows Firewall was blocking the necessary ports. I can now connect the Pistorms to the broker on my notebook. I don't know why but it does not work with the Lego EV3 Brick but I don't really care about that now. The Pistorms is just fine for what I need. Still thank you for your help

closed 2020-08-12 by Kub-sn

jabrena, 2020-08-13:
The point with EV3 is the computational capability. EV3 is nice but it is slow.

Enjoy with your project. If you build something interesting, share it!

cheers