ev3dev-lang-java/ev3dev-lang-java #739: Gyro sensor never reads negative

dwalend, 2020-10-12:
Writing code to demo the gyro sensor was really straightforward thanks to the examples. However, the sensor never reads negative and I don't see a way to set the initial value (to some large multiple of 360).

This is a dire problem when turning left. Is there a work-around I haven't found?

(Code is in Scala - it does exactly what you think it does.)
```
import ev3dev.actuators.LCD
import ev3dev.actuators.ev3.EV3Led
import ev3dev.sensors.EV3Key
import ev3dev.sensors.ev3.EV3GyroSensor
import lejos.hardware.{Key, KeyListener}
import lejos.hardware.lcd.GraphicsLCD
import lejos.hardware.port.SensorPort

object WatchGyro {
  def main(args: Array[String]):Unit = {
    watchGyro()
  }

  def watchGyro(): Unit = {

    //Set status lights to red
    val leftLed = new EV3Led(EV3Led.LEFT)
    val rightLed = new EV3Led(EV3Led.RIGHT)
    leftLed.setPattern(2) //red 
    rightLed.setPattern(3) //yellow - can they really be different colors? cool!

    val lcd = LCD.getInstance()

    val gyroSensor = new EV3GyroSensor(SensorPort.S2)
    val gyroSampleProvider = gyroSensor.getAngleMode
    val gyroArray = new Array[Float](gyroSensor.sampleSize())

    var keepGoing = true

    val anyButton = new EV3Key(EV3Key.BUTTON_ALL)
    val anyButtonListener = new KeyListener {
      override def keyPressed(k: Key): Unit = ()

      override def keyReleased(k: Key): Unit = {
        keepGoing = false
      }
    }
    anyButton.addKeyListener(anyButtonListener)
    
    do {
      gyroSampleProvider.fetchSample(gyroArray,0)
      //Write the gyro value to the display
      lcd.setColor(GraphicsLCD.WHITE)
      lcd.fillRect(0, 0, lcd.getWidth, lcd.getHeight)
      lcd.setColor(GraphicsLCD.BLACK)
      lcd.drawString(s"${gyroArray(0)}", 0, 25, 0)
      lcd.refresh()

      System.out.println(s"${gyroArray(0)}")
    } while (keepGoing)

    //Set status lights to green
    leftLed.setPattern(1) //green
    rightLed.setPattern(1)
  }
}
```

dwalend, 2020-10-12:
It's fixed in 2.6.2-SNAPSHOT last December, but I don't see a 2.6.2 (not snapshot) release. 

(Looks like master is on 2.7.0-SNAPSHOT. Do you have plans to publish a non-snapshot with the fix ?)

jabrena, 2020-11-03:
Yes, I have the idea to publish a release version soon.

JakubVanek, 2020-11-26:
I think that a similar problem was resolved in https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/693, but the fix was not yet released.

jabrena, 2020-11-29:
Do you have an example to test?

jabrena, 2020-12-13:
I would like see more examples with Scala for EV3

jabrena, 2020-12-27:
Hi @dwalend, do you have more examples in Scala?

dwalend, 2020-12-27:
My kid's code, not ready to share for a while.

Even in Java code the gyroscope will never read less that 0 in the current release. #693 was fixed in source code more than a year ago.