ev3dev-lang-java/ev3dev-lang-java #740: Two EV3Motor/Sensor instances for the same Motor/Sensor

mrkrizic, 2020-10-15:
Hello, I am trying to read a motor on the same port that has already been instantiated in another class I don't have access to. One class should control the hardware and the other should keep track of the motor's and sensor's state. Both classes should not know about each other and should be able to run on their own as well as simultaneously.
 
Now my problem is that when I instantiate two motors it is looking at the wrong port on the Brick. Each time I run the jar file again the port number it is looking for increases by two. This means I need to restart the brick each time the program is terminated.

I have looked through the EV3 Motor classes and noticed an unused offset parameter, but since it is not used, I doubt this is the issue. Can anybody explain me why this is happening and point me into a direction what I could do to accomplish this? I would like to stay away from dependency injection if possible.

Any help would be appreciated.

Here is the code I wrote to test the behaviour, since my application did not work using both components:

```java
public static void main(String[] args) {
        EV3MediumRegulatedMotor motor1 = new EV3MediumRegulatedMotor(MotorPort.A);
        EV3MediumRegulatedMotor motor2 = new EV3MediumRegulatedMotor(MotorPort.A);
        System.out.println("Motor1="+motor1.getPosition());
        System.out.println("Motor2="+motor2.getPosition());
        EV3LargeRegulatedMotor motor3 = new EV3LargeRegulatedMotor(MotorPort.D);
        EV3LargeRegulatedMotor motor4 = new EV3LargeRegulatedMotor(MotorPort.D);
        System.out.println("Motor3="+motor1.getPosition());
        System.out.println("Motor4="+motor2.getPosition());
        EV3ColorSensor colorSensor1 = new EV3ColorSensor(SensorPort.S3);
        EV3ColorSensor colorSensor2 = new EV3ColorSensor(SensorPort.S3);
        System.out.println("Sensor1=" + colorSensor1.getColorID());
        System.out.println("Sensor2=" + colorSensor2.getColorID());
        EV3TouchSensor sensor1 = new EV3TouchSensor(SensorPort.S2);
        EV3TouchSensor sensor2 = new EV3TouchSensor(SensorPort.S2);
        System.out.println("Sensor1=" + sensor1.isPressed());
        System.out.println("Sensor2=" + sensor2.isPressed());
    }
```

My log output from the test i wrote in case it helps:

```robot@ev3dev:~/build/libs$ java -jar TwoMotorsAtSamePortTest-1.0-SNAPSHOT.jar
[main] INFO ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
[main] INFO ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
[main] INFO ev3dev.actuators.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: A
[main] INFO ev3dev.actuators.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: A
[main] ERROR ev3dev.utils.Sysfs - Problem reading path: /sys/class/tacho-motor/motor8/position
java.io.IOException: Problem reading path: /sys/class/tacho-motor/motor8/position
        at ev3dev.utils.Sysfs.readString(Sysfs.java:77)
        at ev3dev.utils.Sysfs.readInteger(Sysfs.java:90)
        at ev3dev.hardware.EV3DevDevice.getIntegerAttribute(EV3DevDevice.java:106)
        at ev3dev.actuators.lego.motors.BaseRegulatedMotor.getTachoCount(BaseRegulatedMotor.java:117)
        at ev3dev.actuators.lego.motors.BaseRegulatedMotor.getPosition(BaseRegulatedMotor.java:130)
        at TwoMotorsTest.main(TwoMotorsTest.java:14)
Exception in thread "main" java.lang.RuntimeException: Problem reading path: /sys/class/tacho-motor/motor8/position
        at ev3dev.utils.Sysfs.readString(Sysfs.java:80)
        at ev3dev.utils.Sysfs.readInteger(Sysfs.java:90)
        at ev3dev.hardware.EV3DevDevice.getIntegerAttribute(EV3DevDevice.java:106)
        at ev3dev.actuators.lego.motors.BaseRegulatedMotor.getTachoCount(BaseRegulatedMotor.java:117)
        at ev3dev.actuators.lego.motors.BaseRegulatedMotor.getPosition(BaseRegulatedMotor.java:130)
        at TwoMotorsTest.main(TwoMotorsTest.java:14)
Caused by: java.io.IOException: Problem reading path: /sys/class/tacho-motor/motor8/position
        at ev3dev.utils.Sysfs.readString(Sysfs.java:77)
        ... 5 more
robot@ev3dev:~/build/libs$ java -jar TwoMotorsAtSamePortTest-1.0-SNAPSHOT.jar
[main] INFO ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
[main] INFO ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
[main] INFO ev3dev.actuators.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: A
[main] INFO ev3dev.actuators.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: A
[main] ERROR ev3dev.utils.Sysfs - Problem reading path: /sys/class/tacho-motor/motor10/position
java.io.IOException: Problem reading path: /sys/class/tacho-motor/motor10/position
        at ev3dev.utils.Sysfs.readString(Sysfs.java:77)
        at ev3dev.utils.Sysfs.readInteger(Sysfs.java:90)
        at ev3dev.hardware.EV3DevDevice.getIntegerAttribute(EV3DevDevice.java:106)
        at ev3dev.actuators.lego.motors.BaseRegulatedMotor.getTachoCount(BaseRegulatedMotor.java:117)
        at ev3dev.actuators.lego.motors.BaseRegulatedMotor.getPosition(BaseRegulatedMotor.java:130)
        at TwoMotorsTest.main(TwoMotorsTest.java:14)
Exception in thread "main" java.lang.RuntimeException: Problem reading path: /sys/class/tacho-motor/motor10/position
[... 32 lines cut ...]
        ... 5 more
robot@ev3dev:~/build/libs$ java -jar TwoMotorsAtSamePortTest-1.0-SNAPSHOT.jar
[main] INFO ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
[main] INFO ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
[main] INFO ev3dev.actuators.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: A
[main] INFO ev3dev.actuators.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: A
[main] ERROR ev3dev.utils.Sysfs - Problem reading path: /sys/class/tacho-motor/motor14/position
java.io.IOException: Problem reading path: /sys/class/tacho-motor/motor14/position
        at ev3dev.utils.Sysfs.readString(Sysfs.java:77)
        at ev3dev.utils.Sysfs.readInteger(Sysfs.java:90)
        at ev3dev.hardware.EV3DevDevice.getIntegerAttribute(EV3DevDevice.java:106)
        at ev3dev.actuators.lego.motors.BaseRegulatedMotor.getTachoCount(BaseRegulatedMotor.java:117)
        at ev3dev.actuators.lego.motors.BaseRegulatedMotor.getPosition(BaseRegulatedMotor.java:130)
        at TwoMotorsTest.main(TwoMotorsTest.java:14)
Exception in thread "main" java.lang.RuntimeException: Problem reading path: /sys/class/tacho-motor/motor14/position
        at ev3dev.utils.Sysfs.readString(Sysfs.java:80)
        at ev3dev.utils.Sysfs.readInteger(Sysfs.java:90)
        at ev3dev.hardware.EV3DevDevice.getIntegerAttribute(EV3DevDevice.java:106)
        at ev3dev.actuators.lego.motors.BaseRegulatedMotor.getTachoCount(BaseRegulatedMotor.java:117)
        at ev3dev.actuators.lego.motors.BaseRegulatedMotor.getPosition(BaseRegulatedMotor.java:130)
        at TwoMotorsTest.main(TwoMotorsTest.java:14)
Caused by: java.io.IOException: Problem reading path: /sys/class/tacho-motor/motor14/position
        at ev3dev.utils.Sysfs.readString(Sysfs.java:77)
        ... 5 more
```

JakubVanek, 2020-11-26:
Hi @mrkrizic,

I suspect what is happening is this: one of the constructors in EV3LargeRegulatedMotor class hierarchy (BaseRegulatedMotor) is always switching the port mode to the correct one, so the kernel has to recreate the motor device. This results in the original device "handle" being lost (hence the exceptions) and a new one being created.

A workaround for this has been proposed in https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/655, but it needs to be cleaned up first.

With regards,

Jakub

jabrena, 2020-11-27:
Hi @mrkrizic, the library binds with the runtime.

- Why do you need to instanciate 2 objects about the same device?
- Why not use the same object to get information about the same device?

Juan Antonio

jabrena, 2020-12-02:
Well, the whole library is pretty imperative and not designed for that scenario.
Do you have a repo to review that interaction?
it seems pretty interesting, not everyday, people use Actor Model for Robotics.

Juan Antonio

jabrena, 2020-12-02:
@mrkrizic, it is amazing this project:
https://www.youtube.com/watch?v=k5Hi9-SmJ58
https://github.com/jku-isse/factory-in-a-box

mrkrizic, 2020-12-04:
Yes indeed. I wanted to send you a smaller example in the next few days

jabrena, 2020-12-04:
hi @mrkrizic,

Interesting. Please, share with us the project.
I saw that you used Akka & Milo. I would like to learn the use cases and everything.
I saw your customizations for Motors.

Cheers

mrkrizic, 2020-12-11:
Hello @jabrena,

As part of my bachelor's thesis i created a version of the Turntable using my framework. You can find it here: https://github.com/mrkrizic/FIABTurntable

The wiki contains some more information about the project. Feel free to ask any questions I would be glad to answer them.

The hardware wrapper I wrote is used to test the same program on a windows machine first before deploying it on a BrickPi.

jabrena, 2020-12-13:
I will take a look next week. The videos are pretty promising. Cheers

closed 2020-12-13 by jabrena