ev3dev-lang-java/ev3dev-lang-java #741: MyFirstRobot.main() FAILED The path doesn't exist: /sys/class/lego-port

jchristof, 2020-11-26:
> Task :MyFirstRobot.main() FAILED
Creating Motor A & B
2020-11-26 14:56:12 [main] INFO ev3dev.hardware.EV3DevDistros - Providing an EV3DevDistros instance
2020-11-26 14:56:12 [main] WARN ev3dev.hardware.EV3DevDistros - Failed to detect distro, falling back to Stretch.
2020-11-26 14:56:12 [main] INFO ev3dev.hardware.EV3DevPlatforms - Providing a EV3DevPlatforms instance
2020-11-26 14:56:12 [main] INFO ev3dev.actuators.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: B
Exception in thread "main" java.lang.RuntimeException: The path doesn't exist: /sys/class/lego-port
	at ev3dev.utils.Sysfs.getElements(Sysfs.java:107)
	at ev3dev.hardware.EV3DevDevice.detect(EV3DevDevice.java:57)
	at ev3dev.actuators.lego.motors.BaseRegulatedMotor.<init>(BaseRegulatedMotor.java:87)
	at ev3dev.actuators.lego.motors.NXTRegulatedMotor.<init>(NXTRegulatedMotor.java:26)
	at example.MyFirstRobot.main(MyFirstRobot.java:14)

Execution failed for task ':MyFirstRobot.main()'.
> Process 'command '/usr/lib/jvm/java-1.8.0-openjdk-amd64/bin/java'' finished with non-zero exit value 1

I modified this template with the following:
compileJava {
    sourceCompatibility = 8
    targetCompatibility = 8
}

distributionUrl=https\://services.gradle.org/distributions/gradle-6.4.1-all.zip

changed the motor type to nxt motors for my projdct

**./gradlew ev3devInfo**
ev3dev#2|**System info (from `ev3dev-sysinfo`)**
ev3dev#2|```
ev3dev#2|Image file:         ev3dev-stretch-rpi2-generic-2020-04-10
ev3dev#2|Kernel version:     4.14.114-ev3dev-2.3.5-rpi2
ev3dev#2|Brickman:           0.10.3
ev3dev#2|BogoMIPS:           38.40
ev3dev#2|Bluetooth:          
ev3dev#2|Board:              board0
ev3dev#2|BOARD_INFO_HW_REV=A21041
ev3dev#2|BOARD_INFO_MODEL=Raspberry Pi 2 Model B Rev 1.1
ev3dev#2|BOARD_INFO_SERIAL_NUM=000000008fa816ca
ev3dev#2|BOARD_INFO_TYPE=main
ev3dev#2|Board:              board1
ev3dev#2|BOARD_INFO_FW_VER=1.4.3
ev3dev#2|BOARD_INFO_HW_REV=3.2.1
ev3dev#2|BOARD_INFO_MODEL=Dexter Industries BrickPi3
ev3dev#2|BOARD_INFO_SERIAL_NUM=DE91878850533357312E3120FF11293D
ev3dev#2|BOARD_INFO_TYPE=aux
ev3dev#2|```

jchristof, 2020-11-27:
This is not a bug but due to not understanding how to remote deploy and run the project. When BrickPi is installed and the Pi has java installed, remote deploys can be executed correctly.

closed 2020-11-27 by jchristof

jabrena, 2020-11-29:
Oki, enjoy with the project and share your projects for the community.
We can learn too