ev3dev-lang-java/ev3dev-lang-java #742: Review AdoptOpenJDK Status to fix issues

jabrena, 2020-11-27:
It is necessary review AdoptOpenJDK view to fix the issues:
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/

[image]

jabrena, 2020-11-27:
Running Pipelines for Java 11
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-11/
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster-11/

jabrena, 2020-11-27:
Both pipelines fails.
Did you test in local @JakubVanek ?

```
checking for cacerts file... fail
configure: error: Specified cacerts file "/build/openjdk-build/security/cacerts" does not exist
configure exiting with result code 1
```

JakubVanek, 2020-11-27:
@jabrena this is a known issue, please see https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/746.

JakubVanek, 2020-11-27:
* JDK11 build (buster) is OK now: https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster-11/66/
* JDK12 build should succeed, last failure was likely only due to https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/746
* JDK13 build will fail due to out-of-date patches
* JDK14 build should succeed, even the last build was successful
* JDK15 build does not exist yet
* JDK master will fail due to out-of-date patches

JakubVanek, 2020-11-28:
JDK13 fix: https://github.com/ev3dev-lang-java/openjdk-ev3/pull/76
JDK master/tip fix: https://github.com/ev3dev-lang-java/openjdk-ev3/pull/77

JakubVanek, 2020-11-28:
@jabrena The Buster JDKmaster build has locked up: [Buster JDKMaster#71](https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster-bleeding/71/). The JDK build step normally takes around 13 minutes (including a bootcycle build). For current build it already took 45 minutes.

I don't know how to debug this. It might be that the pass 1 JDKmaster build has crashed while compiling pass 2 JDKmaster (this is what gets distributed). 

This is one of the reasons I don't want to touch this anymore  and why I would very much prefer not building custom JVMs.

JakubVanek, 2020-11-28:
What's more, as the builds are not able to exit themselves, they block the `docker-packet-ubuntu1604-armv8-1` executor (the only one that satisfies the current requirements) until someone terminates them manually. In the meantime other AdoptOpenJDK jobs don't get a chance to run on that executor (I remember that there is some docker job that gets blocked).

JakubVanek, 2020-11-28:
Even the restarted build froze: https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster-bleeding/72/console
I'd give it a few more minutes to see if this isn't just a particularly difficult part of the build, but it doesn't seem so.

jabrena, 2020-11-29:
Currently, everything is in order

closed 2020-11-29 by jabrena