ev3dev-lang-java/ev3dev-lang-java #743: Stop Pipelines in AdoptOpenJDK Status about Java 12, 13, 14

jabrena, 2020-11-27:


JakubVanek, 2020-11-27:
One successful JDK14 build has to be kept to have a bootstrap JDK for JDK15.

jabrena, 2020-11-27:
The unique pending stuff is the patch for JDK13

jabrena, 2020-11-27:
And master:
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-bleeding/71/console

```
# ev3dev-lang-java openjdk build metadata
JAVA_ORIGIN="git"
JAVA_REPO="https://github.com/openjdk/jdk.git"
JAVA_BRANCH="jdk-16+26"
JAVA_COMMIT="7aed9b65d05741cb545e5716296450fb8e8bf668"
JAVA_VERSION="16+26-ev3-unreleased"
CONFIG_VM="client"
CONFIG_DEBUG="release"
CONFIG_VERSION="tip"
CONFIG_PLATFORM="ev3"
CONFIG_MODULES="java.se,jdk.jdwp.agent,jdk.unsupported,jdk.management.agent,jdk.jartool,jdk.crypto.ec"
BUILDER_COMMIT="dddd147f6b8c45af6ae1c9dbf6cea25196715f6e"
BUILDER_EXTRA="Jenkins eljbuild/stretch-bleeding#71 with Debian stretch on armel"

[FETCH] Patching the source tree
patching file src/hotspot/cpu/arm/assembler_arm_32.hpp
Hunk #1 succeeded at 1248 (offset -4 lines).
patching file make/autoconf/hotspot.m4
patching file src/hotspot/share/memory/metaspaceShared.cpp
Hunk #1 succeeded at 71 with fuzz 2 (offset -1 lines).
patching file src/hotspot/share/oops/constantPool.cpp
Hunk #1 succeeded at 443 (offset 6 lines).
patching file src/hotspot/cpu/arm/globalDefinitions_arm.hpp
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file src/hotspot/cpu/arm/globalDefinitions_arm.hpp.rej
[Pipeline] }
$ docker stop --time=1 f444def4ab57ed05126ba1d8ae145fa6edee8ec20b2c578490890a697f44c9d5
$ docker rm -f f444def4ab57ed05126ba1d8ae145fa6edee8ec20b2c578490890a697f44c9d5
[Pipeline] // withDockerContainer
```

jabrena, 2020-11-29:
Built and stopped

closed 2020-11-29 by jabrena