ev3dev-lang-java/ev3dev-lang-java #745: Build Java with #Loom support on AdoptOpenJDK

jabrena, 2020-11-27:
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/

JakubVanek, 2020-11-27:
In my opinion, "one-off" mode does not require a build on AdoptOpenJDK. I will upload a local build and prepare a pull request that would make Loom buildable with [openjdk-ev3][1]. I no longer have (and don't want to have) access to AdoptOpenJDK Jenkins.

[1]: https://github.com/ev3dev-lang-java/openjdk-ev3

JakubVanek, 2020-11-27:
Unfortunately, Loom still does not build from source: https://pastebin.com/Q3HfxLdZ. The build can be started via https://github.com/ev3dev-lang-java/openjdk-ev3/pull/72, but it will stop right at the start.

JakubVanek, 2020-11-27:
I cannot therefore build the JDK or submit a full PR, only a partial PR that fixes issues in the build system.

jabrena, 2020-11-27:
Please, remove the SSL Certs, it is not necessary to build loom support.

- @JakubVanek Merge branch 'feature/ssl-certs-from-mozilla' into feature/loom 9d4294a
- @JakubVanek Merge branch 'bugfix/cacerts-from-adopt' into feature/loom d0fc3f4

JakubVanek, 2020-11-27:
@jabrena I have dropped the two commits. However, `Merge branch 'feature/ssl-certs-from-mozilla' into feature/loom 9d4294a` could be considered to be necessary `to build loom support`, as it is necessary for any JDK build now (not that I would necessarily want so). For more info, see https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/746.

jabrena, 2020-11-27:
Merge them again

JakubVanek, 2020-11-27:
I have rebased the PR on top of https://github.com/ev3dev-lang-java/openjdk-ev3/pull/73, so there should be no merge conflicts.

jabrena, 2020-11-27:
we should build this sources:
https://github.com/openjdk/loom

JakubVanek, 2020-11-27:
> we should build from this sources:
> https://github.com/openjdk/loom

This is currently used when `loom` is selected as a JDK version to build: https://github.com/ev3dev-lang-java/openjdk-ev3/blob/master/scripts/config.sh#L324

I'd guess that a x86 build from the same sources would work fine. I'm currently only speculating, but I wouldn't be surprised if the ARM32 port is missing something that Loom requires (C2/Server compiler, some interface, ...).

JakubVanek, 2020-11-27:
@jabrena the build just failed on Jenkins due to incorrect build JDK version. This should be fixed by https://github.com/ev3dev-lang-java/openjdk-ev3/pull/74. This slipped past testing because I only tested the build on my x86 notebook. Loom should be "buildable" (not really, see comments above) on Jenkins after the JDK15 Stretch build finishes (because it depends on its output).

jabrena, 2020-11-27:
```
configure: Found potential Boot JDK using configure arguments
configure: Potential Boot JDK found at /build/jdk-ev3 is incorrect JDK version (openjdk version "14.0.2" 2020-07-14); ignoring
configure: (Your Boot JDK version must be one of: 15 16)
configure: error: The path given by --with-boot-jdk does not contain a valid Boot JDK
configure exiting with result code 1
```

JakubVanek, 2020-11-27:
@jabrena see https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/745#issuecomment-735004978

jabrena, 2020-11-27:
What is the issue?
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/Stretch%20JDK15/2/console
```
+ docker build -t ev3dev-lang-java:jdk-native-stretch --build-arg commit=dddd147f6b8c45af6ae1c9dbf6cea25196715f6e --build-arg extra=Jenkins eljbuild/Stretch JDK15#2 with Debian stretch on armel --build-arg DEBIAN_RELEASE=stretch --build-arg ARCH=armel --build-arg BUILD_TYPE=native -f /home/jenkins/workspace/eljbuild/Stretch JDK15/system/Dockerfile.native /home/jenkins/workspace/eljbuild/Stretch JDK15/system
"docker build" requires exactly 1 argument.
See 'docker build --help'.

Usage:  docker build [OPTIONS] PATH | URL | -
```

JakubVanek, 2020-11-27:
@jabrena see https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/744#issuecomment-735008811 (create anew == delete the current one, create a new one (stretch-15) based on for example stretch-14, change environment variables to match JDK15 and finally update the display name to be "Stretch JDK15")

JakubVanek, 2020-11-27:
The difference is this:
```
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster-14/
https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/Buster%20JDK15/
```

JakubVanek, 2020-11-27:
> we should build from this sources:
> https://github.com/openjdk/loom

confirmed ([link](https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/stretch-loom/23/console)): 
```
[FETCH] Cloning Java repo from Git
[FETCH] - repo url:   https://github.com/openjdk/loom.git
[FETCH] - branch/tag: fibers
```

jabrena, 2020-11-27:
```
Compiling 6 properties into resource bundles for java.base
In file included from ./src/hotspot/share/runtime/continuation.hpp:30:0,
                 from ./src/hotspot/share/runtime/thread.hpp:33,
                 from ./src/hotspot/share/gc/shared/workgroup.hpp:30,
                 from ./src/hotspot/share/memory/heapInspection.hpp:33,
                 from ./src/hotspot/share/gc/shared/collectedHeap.hpp:32,
                 from ./src/hotspot/share/precompiled/precompiled.hpp:37:
./src/hotspot/share/runtime/frame.hpp: In member function 'const ImmutableOopMap* frame::oop_map() const':
./src/hotspot/share/runtime/frame.hpp:125:30: error: 'get_oop_map' was not declared in this scope
       _oop_map = get_oop_map();
                              ^
In file included from ./src/hotspot/share/runtime/continuation.hpp:30:0,
                 from ./src/hotspot/share/runtime/thread.hpp:33,
                 from ./src/hotspot/share/gc/shared/workgroup.hpp:30,
                 from ./src/hotspot/share/memory/heapInspection.hpp:33,
                 from ./src/hotspot/share/gc/shared/collectedHeap.hpp:32,
                 from ./src/hotspot/share/precompiled/precompiled.hpp:37:
./src/hotspot/share/runtime/frame.hpp: In member function 'void frame::oops_do(OopClosure*, CodeBlobClosure*, const RegisterMap*)':
./src/hotspot/share/runtime/frame.hpp:417:108: error: 'DerivedPointerTable' has not been declared
   void oops_do(OopClosure* f, CodeBlobClosure* cf, const RegisterMap* map) { oops_do_internal(f, cf, NULL, DerivedPointerTable::is_active() ?
                                                                                                            ^~~~~~~~~~~~~~~~~~~
Compiling 71 files for COMPILE_CREATE_SYMBOLS
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-cast-function-type'
cc1plus: warning: unrecognized command line option '-Wno-implicit-fallthrough'
cc1plus: warning: unrecognized command line option '-Wno-int-in-bool-context'
lib/CompileJvm.gmk:141: recipe for target '/build/jdk/build/linux-arm-client-release/hotspot/variant-client/libjvm/objs/precompiled/precompiled.hpp.gch' failed
make[3]: *** [/build/jdk/build/linux-arm-client-release/hotspot/variant-client/libjvm/objs/precompiled/precompiled.hpp.gch] Error 1
make/Main.gmk:252: recipe for target 'hotspot-client-libs' failed
make[2]: *** [hotspot-client-libs] Error 2
make[2]: *** Waiting for unfinished jobs....
Creating javadoc element list
Creating ct.sym classes
ERROR: Build failed for target 'bootcycle-images' in configuration 'linux-arm-client-release' (exit code 2) 
Stopping sjavac server

=== Output from failing command(s) repeated here ===
* For target hotspot_variant-client_libjvm_objs_precompiled_precompiled.hpp.gch:
In file included from ./src/hotspot/share/runtime/continuation.hpp:30:0,
                 from ./src/hotspot/share/runtime/thread.hpp:33,
                 from ./src/hotspot/share/gc/shared/workgroup.hpp:30,
                 from ./src/hotspot/share/memory/heapInspection.hpp:33,
                 from ./src/hotspot/share/gc/shared/collectedHeap.hpp:32,
                 from ./src/hotspot/share/precompiled/precompiled.hpp:37:
./src/hotspot/share/runtime/frame.hpp: In member function 'const ImmutableOopMap* frame::oop_map() const':
./src/hotspot/share/runtime/frame.hpp:125:30: error: 'get_oop_map' was not declared in this scope
       _oop_map = get_oop_map();
                              ^
In file included from ./src/hotspot/share/runtime/continuation.hpp:30:0,
                 from ./src/hotspot/share/runtime/thread.hpp:33,
                 from ./src/hotspot/share/gc/shared/workgroup.hpp:30,
                 from ./src/hotspot/share/memory/heapInspection.hpp:33,
                 from ./src/hotspot/share/gc/shared/collectedHeap.hpp:32,
   ... (rest of output omitted)

* All command lines available in /build/jdk/build/linux-arm-client-release/make-support/failure-logs.
=== End of repeated output ===

=== Make failed targets repeated here ===
lib/CompileJvm.gmk:141: recipe for target '/build/jdk/build/linux-arm-client-release/hotspot/variant-client/libjvm/objs/precompiled/precompiled.hpp.gch' failed
make/Main.gmk:252: recipe for target 'hotspot-client-libs' failed
=== End of repeated output ===

Hint: Try searching the build log for the name of the first failed target.
Hint: See doc/building.html#troubleshooting for assistance.

/build/jdk/make/Init.gmk:310: recipe for target 'main' failed
make[1]: *** [main] Error 2
/build/jdk/make/Init.gmk:186: recipe for target 'bootcycle-images' failed
make: *** [bootcycle-images] Error 2
```

JakubVanek, 2020-11-27:
@jabrena the build indeed failed, and a quick search through the loom repository also reveals that it is probably a x86 only feature for now: https://github.com/openjdk/loom/search?q=get_oop_map

jabrena, 2020-11-27:
Can you evaluate how to complex is to patch that function to arm?

JakubVanek, 2020-11-28:
> Can you evaluate how to complex is to patch that function to arm?

Very complex.

For example, this file is missing in other JIT platforms and it contains lots of architecture-specific low-level code:
https://github.com/openjdk/loom/blob/20e146069ca620c24d983baed73d2258485f16c7/src/hotspot/cpu/x86/continuation_x86.inline.hpp

I'd guess this is on scale of months of work.

jabrena, 2020-11-28:
In that case, this issue will have to wait.
Thanks for the effort

Good night

JakubVanek, 2020-11-28:
> In that case, this issue will have to wait.

I am not sure that the support will come. Some features eventually reach the ARM32 codebase (Shenandoah seems to be planning to add support: [link](https://wiki.openjdk.java.net/display/shenandoah/Main#Main-HardwareSupport)), but other may not. IMHO ultimately it depends on whether a third party like BellSoft (who AFAIK maintains the ARM32 port) decides to dedicate the necessary resources to it.

jabrena, 2020-11-29:
Currently, it is not possible.
It is necessary to wait

closed 2020-11-29 by jabrena