ev3dev-lang-java/ev3dev-lang-java #746: Use SSL certs from Mozilla

JakubVanek, 2020-11-27:
It is necessary to resolve https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/728. All openjdk-ev3 builds are going to fail  within a week since two days ago the cacerts file was removed from the openjdk-build repository: https://github.com/AdoptOpenJDK/openjdk-build/commit/9cb3ddcb8f3550aa904f086146be0cdd242d7e8a. The solutions are still the same as outlined in https://github.com/ev3dev/ev3dev/issues/1437#issuecomment-679016949.

I would therefore recommend:
* merging the Mozilla cacerts PR: https://github.com/ev3dev-lang-java/openjdk-ev3/pull/62
* merging the debpkg-tarpkg cacerts unification PR: https://github.com/ev3dev-lang-java/openjdk-ev3/pull/61

JakubVanek, 2020-11-27:
Indeed, first builds have failed yesterday: https://ci.adoptopenjdk.net/view/ev3dev/job/eljbuild/job/buster-12/65/console

jabrena, 2020-11-27:
Apart of Mozilla, what other alternatives exist?

- https://github.com/JakubVanek/elj-jdk-private-dev/tree/bugfix/cacerts-from-adopt
- https://github.com/JakubVanek/elj-jdk-private-dev/tree/feature/ssl-certs-from-mozilla
- https://github.com/JakubVanek/elj-jdk-private-dev/tree/bugfix/alternative-cacerts-fix
- https://github.com/JakubVanek/elj-jdk-private-dev/tree/feature/ssl-certs-from-openjdk

JakubVanek, 2020-11-27:
> Apart of Mozilla, what other alternatives exist?

There are also the stock OpenJDK cacerts. However, this is also what AdoptOpenJDK is leaving. AdoptOpenJDK seems to be switching to Mozilla's certs (please see https://github.com/AdoptOpenJDK/openjdk-build/commit/9cb3ddcb8f3550aa904f086146be0cdd242d7e8a). I don't know other CA certificate programs and as such I don't feel qualified to discuss this.

>   * https://github.com/JakubVanek/elj-jdk-private-dev/tree/bugfix/cacerts-from-adopt
>   * https://github.com/JakubVanek/elj-jdk-private-dev/tree/feature/ssl-certs-from-mozilla
>   * https://github.com/JakubVanek/elj-jdk-private-dev/tree/bugfix/alternative-cacerts-fix
>   * https://github.com/JakubVanek/elj-jdk-private-dev/tree/feature/ssl-certs-from-openjdk

These were described may times by me (e.g. https://github.com/ev3dev/ev3dev/issues/1437#issuecomment-679016949 and related threads in this repository). Only the bottom two PRs directly relate to what SSL certs are used. The top two PRs fix an related, but not the same issue (broken CA certs in JRI, https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/728) and as such I have labelled one of these "unrelated" PRs preferred by me as "not necessary" in the summary above.

> Is it possible to evolve the cacerts in the future?

It depends on which mechanism is intended for this. This does not force the use of Mozilla's certs forever, just a different CACERTFILE has to be provided in `config.sh` (https://github.com/ev3dev-lang-java/openjdk-ev3/pull/62/files#diff-e52229475913ead2a19b9c4be787d753170d054136228f98c3e427452e7bc2ca). If this means for building multiple JDK packages with different cacerts inside, then I do not see much value in that at all. A customized cacerts file (e.g. from different vendor) can always be distributed separately and installed by users by simply replacing the cacerts keystore in the JDK directory.

@jabrena

jabrena, 2020-11-27:
What is the solution adopted by `AdoptOpenJDK`?
https://github.com/AdoptOpenJDK/openjdk-build/commit/9cb3ddcb8f3550aa904f086146be0cdd242d7e8a

jabrena, 2020-11-27:
Oki, in that case, go ahead Mozilla cacerts :)

JakubVanek, 2020-11-27:
> But not merge the debpkg-tarpkg cacerts unification PR: [ev3dev-lang-java/openjdk-ev3#61](https://github.com/ev3dev-lang-java/openjdk-ev3/pull/61) (not necesssary)
> It is necessary to continue with the technical discussion.

Yes, I agree, from this point of view it affects a different part of the project and as such it should be handled separately.

JakubVanek, 2020-11-27:
@jabrena Minimized pull request is now available: https://github.com/ev3dev-lang-java/openjdk-ev3/pull/73

For added transparency I decided to reuse the AdoptOpenJDK code where there was overlap.

jabrena, 2020-11-27:
Running all pipelines now

For JDK14, it is necessary to not be ea version

JakubVanek, 2020-11-27:
> For JDK14, it is necessary to not be ea version

Currently EA is only in the job label, the build system uses ([link][1]) the latest general availability release from the JDK-Updates repository as do other released versions. 

[1]: https://github.com/ev3dev-lang-java/openjdk-ev3/blob/dddd147f6b8c45af6ae1c9dbf6cea25196715f6e/scripts/config.sh#L266

JakubVanek, 2020-11-28:
Issue fixed

closed 2020-11-28 by JakubVanek