ev3dev-lang-java/ev3dev-lang-java #748: Performance too slow for control feedback for a moving robot

dwalend, 2020-11-30:
The java code attached runs in about 100 to 140 milliseconds per iteration. Similar code using the ev3 classroom from LEGO runs each iteration in 2 to 4 milliseconds. 100 milliseconds is enough time for the robot to travel an inch or 3, which becomes significant quickly.

My question is - is the delay in the ev3 os streaming implementation, or in ev3dev-lang-java:2.6.2-SNAPSHOT code that reads and writes those streams? That'll tell me what to try first to speed it up my own code. If it's at the OS level then probably the best I can do is read and write the buffers in parallel. If it's in Java IO then maybe a JNI approach or an NIO loop could help.

I'm using ev3dev-lang-java:2.6.2-SNAPSHOT - on a stock EV3, not overclocked, with the stock ev3dev OS SDD card. It does touch the lejos SampleProvider to use the gyro sensor, but that seems to pass right through to Sysfs.readFloat(...).

Juan Antonio Breña Moral asked me to open an issue for the ev3dev-lang-java project here.

dwalend, 2020-11-30:
Here's the java code:

```
package ev3.timing;

import ev3dev.actuators.ev3.EV3Led;
import ev3dev.actuators.ev3.EV3Led.Direction;
import ev3dev.actuators.lego.motors.EV3LargeRegulatedMotor;
import ev3dev.sensors.EV3Key;
import ev3dev.sensors.ev3.EV3GyroSensor;
import lejos.hardware.port.MotorPort;
import lejos.hardware.port.SensorPort;
import lejos.robotics.SampleProvider;

public class TimingTest {

    //equivalent loop on just ev3 classroom is .3.6 / 100 = 0.0036 per iteration = 3.6 milliseconds - so it's a factor of about 100x
    public static void main(String[] args) {
        EV3LargeRegulatedMotor leftMotor = new EV3LargeRegulatedMotor(MotorPort.B);
        EV3LargeRegulatedMotor rightMotor = new EV3LargeRegulatedMotor(MotorPort.C);
        System.out.println("motors created");

        EV3GyroSensor gyroSensor = new EV3GyroSensor(SensorPort.S2);
        SampleProvider gyroSampleProvider = gyroSensor.getAngleMode();
        float[] gyroArray = new float[gyroSensor.sampleSize()];
        System.out.println("gyro created");

        warmUp(gyroSampleProvider,leftMotor,rightMotor);

        long[] timingResults = new long[10];
        for(int i = 0 ; i < timingResults.length ; i++) {
            long startTime = System.currentTimeMillis();

            //readHeading
            gyroSampleProvider.fetchSample(gyroArray, 0);
            float heading = gyroArray[0];
            //read each motor's odometer
            int leftOdometer = leftMotor.getTachoCount();
            int rightOdometer = rightMotor.getTachoCount();
            //set each motor's speed
            leftMotor.setSpeed(0);
            rightMotor.setSpeed(0);

            long endTime = System.currentTimeMillis();
            timingResults[i] = endTime - startTime;
            //todo set the array value

        }
        long total = 0 ;
        for(long r : timingResults ) {
            System.out.println(r);
            total = total + r;
        }
        System.out.println("average is "+(total/timingResults.length))   ;
    }

    static void warmUp(SampleProvider gyroSampleProvider,EV3LargeRegulatedMotor leftMotor,EV3LargeRegulatedMotor rightMotor) {
        long startTime = System.currentTimeMillis();

        //readHeading
        float[] gyroArray = new float[1];
        gyroSampleProvider.fetchSample(gyroArray, 0);
        //read each motor's odometer
        int leftOdometer = leftMotor.getTachoCount();
        int rightOdometer = rightMotor.getTachoCount();
        //set each motor's speed
        leftMotor.setSpeed(0);
        rightMotor.setSpeed(0);

        long endTime = System.currentTimeMillis();
    }
}
```

dwalend, 2020-11-30:
Here's the ev3 classroom code that runs in 2-4 milliseconds per iteration.

[image]

jabrena, 2020-12-01:
Hi @dwalend, 

many thanks for the issue. 
I will review this week, using the example provided.

Juan Antonio

jabrena, 2020-12-01:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/master/src/main/java/ev3dev/utils/Sysfs.java
https://docs.oracle.com/javase/7/docs/api/java/nio/file/Files.html#readAllLines(java.nio.file.Path,%20java.nio.charset.Charset)

JakubVanek, 2020-12-01:
I will try to hack something together in C++ in the coming days to see what is the limit of the underlying OS.

JakubVanek, 2020-12-01:
I have an initial version of the C++ benchmark available.

"Rolling-release" links (pointing at master, not the commit https://github.com/JakubVanek/ev3dev-lang-cpp-bench/commit/b01a3b22d6d4bbc3cecd23c53c214706d4e23e76):
* Source code: https://github.com/JakubVanek/ev3dev-lang-cpp-bench (it includes the ev3dev-lang-cpp library as a submodule so that it's built with the program).
* Binary: https://github.com/JakubVanek/ev3dev-lang-cpp-bench/raw/main/cxxbench (compiled under Ubuntu 18.04 with `-O3` and CPU-specific optimizations enabled)

However, I've noticed that even though the benchmark sets the speed attribute, it doesn't order the motors to start moving. Is this intentional? I've followed this in the C++ program. I believe the situation with leJOS/ev3dev-lang-java is similar and you'd have to call `forward()` for the speed setting to take full effect.

JakubVanek, 2020-12-01:
> -[ ] Review imports
> -[ ] Create a benchmark with multiple implementations

@jabrena I'd check the ideas proposed in https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/652/files, while not everything is useful, the `writeString` method looks more optimized than the current one.

JakubVanek, 2020-12-01:
It seems that the compiler in Ubuntu 18.04 is too new to compile for ev3dev-stretch, I'll provide a binary built with a "[universal](https://www.musl-libc.org/)" compiler ([download here](https://drive.google.com/file/d/1h6Fl50jRfensIC_CLt73ER-i06X3-ODK/view)) that I was using for different C development.

JakubVanek, 2020-12-01:
EDIT: I'm going to use a `arm-linux-musleabi-cross` compiler from http://musl.cc/, they're available for Windows as well, so then a comparable setup should be available.

dwalend, 2020-12-01:
> I've noticed that even though the benchmark sets the speed attribute, it doesn't order the motors to start moving. Is this intentional?

That's correct. Starting the motors isn't in the control loop, so I left it out of the benchmark. That gives me a reason better than "I don't want to take the wheels off the robot, and don't want it crashing into stuff on my desk."

dwalend, 2020-12-01:
> EDIT: I'm going to use a `arm-linux-musleabi-cross` compiler from http://musl.cc/, they're available for Windows as well, so then a comparable setup should be available.

If that doesn't work out - ev3dev has a Docker recipe for a solid dev environment.

JakubVanek, 2020-12-01:
> > EDIT: I'm going to use a `arm-linux-musleabi-cross` compiler from http://musl.cc/, they're available for Windows as well, so then a comparable setup should be available.
> 
> If that doesn't work out - ev3dev has a Docker recipe for a solid dev environment.

Hmm, I agree, that would be more convenient if that is an option.

JakubVanek, 2020-12-01:
Results for the musl version:
```
robot@ev3dev:~$ ./cxxbench 
motors created
gyro created
3392
785
3280
3283
2666
3552
1109
4874
847
3151
average is 2.6939 ms
```
and this is one of the worse runs, typically it's around `1.4 ms`. The mid-results are off, I've switched the code to microseconds in C++, while in Java they're milliseconds.

JakubVanek, 2020-12-01:
> That's correct. Starting the motors isn't in the control loop, so I left it out of the benchmark. That gives me a reason better than "I don't want to take the wheels off the robot, and don't want it [...]

I think this will not work in the production code -- AFAIK the speed change is committed only after a `forward()` call is made in leJOS / `run-forever` command in invoked in ev3dev.

JakubVanek, 2020-12-01:
Results for the docker-compiled version:
```
robot@ev3dev:~$ ./cxxbench 
motors created
gyro created
1608
1832
1503
2751
1660
1505
2033
1725
2797
1657
average is 1.9071 ms
```

Quickly looking at a few runs, the mean is about `1.9 ms`.

This is with `run-forever` added to my latest version of the benchmark, in Java, there needs to be something like `leftMotor.forward(); rightMotor.forward();` (alt. with synchronization start+end).

JakubVanek, 2020-12-01:
> If it's in Java IO then maybe a JNI approach or an NIO loop could help.

Writing the control loop in C/C++ and then communicating with non-critical logic in Java would most certainly work, although the impact of JIT/GC running next to it would have to be measured. On the other hand, if the logic runs completely in its own thread, it could switch itself to realtime SCHED_RR scheduling. With that, whatever is running next to it shouldn't present a problem.

Using JNI to do just the IO (i.e. replacing Sysfs class with a JNI-backed implementation) is another way to go. This would preserve the existing outer API, but the performance might be slightly lower.

JakubVanek, 2020-12-01:
Results of the last C++ version with SCHED_RR scheduling enabled:
```
robot@ev3dev:~$ sudo chrt -r 50 ./cxxbench 
motors created
gyro created
1468
1548
1306
1168
1506
1460
1281
1186
1599
1299
average is 1.3821 ms
```
The variance now looks much lower too.

JakubVanek, 2020-12-01:
I think the C/C++ version could be hand-optimized to be even faster, my current goal was to make look the same as the Java benchmark. However, the results should be of the same magnitude. There will be the same context switches etc. (= limitation of ev3dev), just the UTF-8 conversion and similar stuff could be cut out.

JakubVanek, 2020-12-02:
Hmm, to be fair to the Java version, proper warmup for microbenchmarking should have much more iterations than one. That being said, the same warmup would then have to be done in the production code too.

JakubVanek, 2020-12-02:
I tried to rewrite the program using the ugliest hacks I know and this is what I got ([binary](https://github.com/JakubVanek/ev3dev-lang-cpp-bench/raw/main/overkill), [source](https://github.com/JakubVanek/ev3dev-lang-cpp-bench/blob/main/overkill.c)):
```
robot@ev3dev:~$ sudo chrt -r 50 ./overkill 
motors created
gyro created
454
292
204
276
243
227
171
173
167
255
average is 0.246200 ms
```
I'd consider that nearing the limit of what is achievable with ev3dev. There are still some places that could be optimized (sprintf for int->string conversion isn't optimal), but something of that style would have to be used anyway for it to be useful.

dwalend, 2020-12-02:
> I tried to rewrite the program using the ugliest hacks I know and this is what I got ([binary]

Up until this one the speeds you saw in C++ lined up with the LEGO classroom scratch. (And thanks for the reminder of why I stopped doing C++. ... You could put assembly code in that for more ugly, but I don't know that you can gain anything.)

The warm-up is short for Java - hotspot normally kicks in after 8 iterations. However, the biggest time-sink on the ev3 seems to be that initial loading of classes. I didn't see much drop-off in longer tests, and - like you mentioned - it's not terribly predictable.

It looks like there's two orders of magnitude potential gain from the base case in Java to reasonable C code.

dwalend, 2020-12-02:
I want to pick on this method in Sysfs.java as a first step:
'        public static String readString(final String filePath) {
		if(log.isTraceEnabled())
			log.trace("cat " + filePath);
		try {
			final Path path = Paths.get(filePath);
			if(existFile(path) && Files.isReadable(path)){
				final String result = Files.readAllLines(path, Charset.forName("UTF-8")).get(0);
				if(log.isTraceEnabled())
					log.trace("value: {}", result);
				return result;
			}
			throw new IOException("Problem reading path: " + filePath);
		} catch (IOException e) {
			log.error(e.getLocalizedMessage(), e);
			throw new RuntimeException("Problem reading path: " + filePath, e);
		}
	}
'

It's opening and closing the file every time. Would it be possible - as part of initializing the sensor - to do the safety checks at init time, and keep the file open ? (And leave it up to the client code to close when done using the sensor?) That'd save all the  IO set-up, and almost all of the memory allocation. 

The trade is that the sensor would pick up some more - and more complicated state.

JakubVanek, 2020-12-02:
I'd consider that nearing the limit of what is achievable with ev3dev. There are still some places that could be optimized (sprintf for int->string conversion isn't optimal), but something of that style would have to be used anyway for it to be useful.

It seems I was wrong again. By removing int<->string conversion altogether, the time is nearly halved:
```
robot@ev3dev:~$ sudo chrt -r 50 ./overkill2 
motors created
gyro created
142
121
110
186
125
118
107
110
107
112
average is 0.123800 ms
```

JakubVanek, 2020-12-02:
Assembly would indeed be the next step, but this seems like too much work.

JakubVanek, 2020-12-02:
> Up until this one the speeds you saw in C++ lined up with the LEGO classroom scratch. (And thanks for the reminder of why I stopped doing C++. ... You could put assembly code in that for more ugly, but I don't know that you can gain anything.)

I have to agree, from this viewpoint the C++ result is a bit of an embarrassment. On the other hand, if this Scratch variant is based on Microsoft MakeCode, I think they might be generating ARMv4T code ahead of time and that could explain their great performance. But if this is based on the stock firmware's VM, then it seems engineers at LEGO optimized it very well.

> The warm-up is short for Java - hotspot normally kicks in after 8 iterations. However, the biggest time-sink on the ev3 seems to be that initial loading of classes. I didn't see much drop-off in longer tests, and - like you mentioned - it's not terribly predictable.

In that case this warmup looks reasonable, I was thinking the threshold was in the order of 1000s of calls. I can confirm I remember classloading time sometimes took a lot of time.

JakubVanek, 2020-12-02:
> I have to agree, from this viewpoint the C++ result is a bit of an embarrassment. On the other hand, if this Scratch variant is based on Microsoft MakeCode, I think they might be generating ARMv4T [...]

On the other hand, I think that part of this might be due to the ev3dev-lang-cpp library and not necessarily due to C++ as a language. IIRC the library works with file names instead of file descriptors and uses C++ iostreams to access the files in sysfs. These do seem to have quite large overhead. However, by using POSIX syscalls directly and caching the file descriptors, the code could be sped up. Interestingly, this is a similar pattern to Java, my only concern here is if the overhead of a JNI call isn't too large + if the JIT compiled code is able to start running fast enough quickly.

JakubVanek, 2020-12-02:
> It's opening and closing the file every time. Would it be possible - as part of initializing the sensor - to do the safety checks at init time, and keep the file open ? (And leave it up to the [...]

I don't see a problem with this and I agree this might produce a measurable drop in execution time.

> The trade is that the sensor would pick up some more - and more complicated state.

Not necessarily -- this could be encapsulated in an `Attribute` class. On the other hand, the sensor classes would have to be (Auto)Closeable.

JakubVanek, 2020-12-02:
> On the other hand, I think that part of this might be due to the ev3dev-lang-cpp library and not necessarily due to C++ as a language.

It seems this is true, I have added a third version of the test - a bastardized version of one of my earlier programs with custom C++ bindings. The resulting code looks mostly similar to the Java version (e.g. no ugly hacks).

```
robot@ev3dev:~$ sudo chrt -r 50 ./ducttape 
- device added: /sys/devices/platform/ev3-ports/ev3-ports:outB/lego-port/port5/ev3-ports:outB:lego-ev3-l-motor/tacho-motor/motor0
- device added: /sys/devices/platform/ev3-ports/ev3-ports:outC/lego-port/port6/ev3-ports:outC:lego-ev3-l-motor/tacho-motor/motor1
- device added: /sys/devices/platform/soc@1c00000/serial8250.0/tty/ttyS0/sensor0
- device added: /sys/devices/platform/ev3-ports/ev3-ports:outB/lego-port/port5/ev3-ports:outB:lego-ev3-l-motor/tacho-motor/motor0
- device added: /sys/devices/platform/ev3-ports/ev3-ports:outC/lego-port/port6/ev3-ports:outC:lego-ev3-l-motor/tacho-motor/motor1
- device added: /sys/devices/platform/soc@1c00000/serial8250.0/tty/ttyS0/sensor0
motors created
gyro created
554
409
320
548
344
572
496
575
571
434
average is 0.4823 ms
```

This is using:
* direct `pread()`/`pwrite()` on file descriptors (they're opened before the first write and then they're stored inside the attribute object directly)
* per-attribute cached read/write buffer
*`std::stof`/`std::stoi`/`std::to_string` to do data type conversions

Program: https://github.com/JakubVanek/ev3dev-lang-cpp-bench/blob/main/custom/program.cpp + libs in parent directory
Executable: https://github.com/JakubVanek/ev3dev-lang-cpp-bench/blob/main/custom/ducttape

JakubVanek, 2020-12-02:
Hmm, there's one optimization I forgot:
```cpp
int16_t rg::lego_gyro::readAngle() {
    return readBinaryInteger<int16_t>(readBinaryData());
}
```
For EV3 sensors, one can usually directly read the integer value directly without parsing anything.

JakubVanek, 2020-12-02:
> > The trade is that the sensor would pick up some more - and more complicated state.
> 
> Not necessarily -- this could be encapsulated in an `Attribute` class. On the other hand, the sensor classes would have to be (Auto)Closeable.

POC in C++: [Attribute.h](https://github.com/JakubVanek/ev3dev-lang-cpp-bench/blob/main/custom/udev/include/Attribute.h) + [sensor.hpp](https://github.com/JakubVanek/ev3dev-lang-cpp-bench/blob/main/custom/devices/include/sensor.hpp). However, this is indeed adding lots of state, at minimum all attributes would have to be closed in `close()` (could be done by putting them into an auxiliary array).

ev3dev-lang-cpp has a stateless solution - a global [LRU cache](https://github.com/ddemidov/ev3dev-lang-cpp/blob/ea157db046a394c9fc904811c605b4a0cd11d127/ev3dev.cpp#L72) for open files. This would avoid keeping the state inside the individual sensors, but it adds the overhead of a cache lookup.

jabrena, 2020-12-02:
Hi @dwalend , @JakubVanek, 

has anyone disabled Logs in that programs to review the performance impact?
what log library did you use for tests?

Some ideas that I see currently:

- Review logging options
- Unify imports to use NIO
- Review PR #652 again
- JNI

**Note:** the whole system is stateless. The usage of a Cache should be considered at the end

Juan Antonio

JakubVanek, 2020-12-02:
> has anyone disabled Logs in that programs to review the performance impact?
> what log library did you use for tests?

In C/C++ only stdout was used. As the logging wasn't in the hot path, it shouldn't have affect the performance there. I haven't tested the Java version yet.

jabrena, 2020-12-02:
@JakubVanek, the issue is about the `Java` library. 
What is the value to do a benchmark in another library?

@dwalend 

Good point:

```java
public static String readString(final String filePath) {
		if(log.isTraceEnabled())
			log.trace("cat " + filePath);
		try {
			final Path path = Paths.get(filePath);
			if(existFile(path) && Files.isReadable(path)){
				final String result = Files.readAllLines(path, Charset.forName("UTF-8")).get(0);
				if(log.isTraceEnabled())
					log.trace("value: {}", result);
				return result;
			}
			throw new IOException("Problem reading path: " + filePath);
		} catch (IOException e) {
			log.error(e.getLocalizedMessage(), e);
			throw new RuntimeException("Problem reading path: " + filePath, e);
		}
	}
``` 

I recommend to import the class:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/master/src/main/java/ev3dev/utils/Sysfs.java

and extend with new approaches and everything.

Using this approach, I don´t see that C++ benchmark.
This issue is about an analysis to improve the performance about Java code.

are everyone in the same pace @dwalend & @JakubVanek ?

Juan Antonio

JakubVanek, 2020-12-02:
> are everyone in the same pace @dwalend & @JakubVanek ?

Yes, the C/C++ part is partially offtopic, the primary benefit to this issue was to show that ev3dev is capable of sub-millisecond loop times. However, if JNI ends up to be used, the code ([here](https://github.com/JakubVanek/ev3dev-lang-cpp-bench/tree/main/custom)) could serve as a starting point for its native backend.

jabrena, 2020-12-02:
Sorry, when I mentioned JNI, I said to interact directly from java with Linux directly, not to communicate with another library installed in EV3.

JakubVanek, 2020-12-02:
> Sorry, when I mentioned `JNI`, I said to interact directly from java with Linux directly, not to communicate with another library installed in EV3 as we did for the Display support.

I see, this could be done via JNA. 
https://github.com/java-native-access/jna/blob/master/www/FrequentlyAskedQuestions.md#how-does-jna-performance-compare-to-custom-jni
https://github.com/java-native-access/jna/blob/master/www/DirectMapping.md

IIRC JNA is used in leJOS as well to do the communication with the EV3 kernel modules.

JakubVanek, 2020-12-02:
In that case, it should be possible to expand https://github.com/ev3dev-lang-java/ev3dev-lang-java/tree/master/src/main/java/ev3dev/utils/io to include [`pread`](https://man7.org/linux/man-pages/man2/pread.2.html) and [`pwrite`](https://man7.org/linux/man-pages//man2/pwrite.2.html) and build an optimized I/O layer on top of that (at the end, NIO will likely have to invoke similar syscalls anyway).

jabrena, 2020-12-02:
JNA, has a weight of 1.4MB
https://mvnrepository.com/artifact/net.java.dev.jna/jna/5.6.0

Remember that we discussed to replace `Logback` to `slf4j-simple` for the same reason.

Lets start from the beginning:

- Logging options & Dependency alternatives
- Unify imports to use NIO
- Review PR #652 again
- JNI
- JNA
- Other ideas using Java technology

We have to provide some time to @dwalend to provide few tests.

JakubVanek, 2020-12-02:
> JNA, has a weight of 1.4MB
> https://mvnrepository.com/artifact/net.java.dev.jna/jna/5.6.0

@jabrena the real cost should be ~0, because JNA is used by the display code.

JakubVanek, 2020-12-02:
Here we go: https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/d8320a391f9ecba756814b12fce903f40c49da9c/build.gradle#L40
It's an older release, but that shouldn't be a big issue. Moreover, this way it's tested that it works on ev3dev.

jabrena, 2020-12-02:
Right!

What is mean: `build an optimized I/O layer on top`? 

Can we call pread and pwrite with java to improve current calls?

JakubVanek, 2020-12-02:
> What is mean: `build an optimized I/O layer on top`?
>
> Can we call `pread` and `pwrite` with `Java` to improve the performance in current calls?

This way the standard Java IO stack could be bypassed completely. Now that I'm thinking about it, it probably is quite optimized and I'm not sure how much it is possible to gain, it would be necessary to test this.

In total, four functions would be needed:

* `open(filename, how)` asks the kernel to open a file for reading/writing and returns file descriptor [already there]
* `pread(fd, buffer, buffer_size, 0)` asks the kernel to read up to `buffer_size` bytes to the specified (native) buffer. [new]
* `pwrite(fd, buffer, buffer_size, 0)` asks the kernel to write up to `buffer_size` bytes from the specified (native) buffer. [new]
* `close(fd)` would close that file. [already there]

The difference between read/write and pread/pwrite is that it is not necessary to call `lseek(fd, 0, SEEK_SET)` before /after each read/write operation to return to the beginning of the file. Thus, it makes these calls more suitable for reading from cached FDs.

Without caching the file descriptors, open+pread (or just read)+close would be needed. With cached file descriptors, open would be called at the beginning and subsequent reads would just call pread.

Where I'm not sure how much this solution is going to help is getting the data from/to Java. JNA seems to handle this via `java.nio.Buffer`: https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/d8320a391f9ecba756814b12fce903f40c49da9c/src/main/java/ev3dev/utils/io/NativeLibc.java#L50. The data in the buffer would then have to be converted to string (quick googling hints that StandardCharsets.US_ASCII.decode() could do it) and then to integer (standard Java).

jabrena, 2020-12-02:
This approach sounds nice for me:

```
The data in the buffer would then have to be converted to string (quick googling hints that StandardCharsets.US_ASCII.decode() could do it) and then to integer (standard Java).
```

JakubVanek, 2020-12-02:
> I miss `DI, Dependency Injection` in this project to replace some pieces of code with new alternatives in an easy way but EV3 is so limited in terms of Memory & CPU

There's always the option of replacing the libraries at the program build time. Without file descriptor caching, something like `ISysfs` should be possible; each JAR / backend implementation would then implement a static factory method for this interface.

JakubVanek, 2020-12-02:
Alternatively, to avoid the overhead of virtual dispatch through an interface, non-interface approach could be used.

There would be an API library containing dummy implementations of sysfs functions. This is what the main library would get built against. However, it would use only the `compileOnlyApi` dependency, not `compile`. The `runtimeOnly` part with real implementation would be provided by a separate library when building the final program/fatjar.

jabrena, 2020-12-02:
@JakubVanek

We are losing the focus.

Focus in this point:

```
The data in the buffer would then have to be converted to string 
(quick googling hints that StandardCharsets.US_ASCII.decode() 
could do it) and then to integer (standard Java).
```

I will create a project to add ideas from @dwalend & @JakubVanek 

Juan Antonio

dwalend, 2020-12-03:
> has anyone disabled Logs in that programs to review the performance impact? what log library did you use for tests?

I used out-of-box. Trace logging is off by default, and switched off by those if statements. (Very easy to hack out altogether in a branch of that sandbox.)

I think a sandbox where we can see everything is a great idea. From here I think there's a few directions to explore:

The most deliberate is to hook up a profiler to see where the Java program is spending its time. It looks like maybe YourKit supports ARM. I'll put half an hour into that tomorrow and see how far I get.

My first guess is that readAllLines() is optimized for reading a large number of lines - allocating a big buffer - overkill for 4 bytes to read the motor odometer. That'd be the best news possible; we could simply write some code tailored precisely to our needs.

My second guess is asking the OS to open the file is really slow and pure overhead. The standard work-around for that is to open the file when the sensor is created and close it at the end. That adds a minor burden on developers; we'd have to close things we opened (or just be sloppy and let the OS clean up; I don't think many ev3 projects will be unplugging sensors while the program is executing, so maybe that's OK). It would be an API change - rejected in PR #652 . 

My third guess is that the more accessible Java IO libraries are not up to the job, but NIO, maybe even an NIO loop, could do it. That's really challenging code to write. 

I was able to write something using multiple threads and volatile variables that let each sensor read happen in parallel - built on top of the library as-is. That gave maybe a factor of 5 speed-up. I don't like the aesthetics of the solution. It seemed like I was burying the problem instead of solving it.

You already have JNA in place? (And that's why using the screen is so slow to start??) Maybe a native code solution isn't so terrible. I was going to argue against it: Ev3 is a toy for kids so keeping it accessible seemed important. Building the library would add a lot of complexity to the project. I could help with a command line to get that docker image to compile the .cpp source. 

@jabrena - is the sandbox you envision just a fork of ev3dev-lang-java ?

dwalend, 2020-12-03:
Yourkit doesn't support ARM with soft float. https://www.yourkit.com/changes/yjp_2019_8.jsp .

JakubVanek, 2020-12-03:
> You already have JNA in place? (And that's why using the screen is so slow to start??)

JNA shouldn't be the problem there. I think (and now also hope) that the problem is in Java2D, but I don't know for sure.

The rendering functions from leJOS worked only on 1bpp buffers. Ev3dev rewrite of the display driver to tinydrm changed this to 32bpp XRGB, this is then converted to native 2bpp somewhere in the kernel. To save time, it was decided to use stock Java2D rendering pipeline instead of porting leJOS rendering functions to XRGB.

JakubVanek, 2020-12-03:
> My second guess is asking the OS to open the file is really slow and pure overhead.

I can measure the raw impact on C code, although the JVM is going to add its own overhead too.

jabrena, 2020-12-03:
To measure the optimization, we will use **JMH**, nothing more.

jabrena, 2020-12-03:
Hi @dwalend 

> is the sandbox you envision just a fork of ev3dev-lang-java ?

No, the idea to improve the class `SysFs.java` and to measure the ideas, we will use **JMH**.

JakubVanek, 2020-12-03:
> My second guess is asking the OS to open the file is really slow and pure overhead. The standard work-around for that is to open the file when the sensor is created and close it at the end. That [...]

Adding a hidden LRU cache in Sysfs could work, although that has been partially rejected. I don't think that letting the OS to close these FDs would be a problem. This change would, however, add the overhead of string hashing/map lookup. String path is already getting built by the sensor classes.

Alternative solution to breaking the existing API would be to create a new, faster API and keep the old one for backwards compatibility.

> You already have JNA in place? (And that's why using the screen is so slow to start??) Maybe a native code solution isn't so terrible.

The benefit of JNA is that native code is actually avoided. It is used in the display code to perform `ioctl`s and do `mmap`ing of `/dev/fb0` for display data. I don't remember what led to this decision, but it was probably to avoid compiling a native JNI library next to the Java one.

jabrena, 2020-12-03:
The idea to develop native code is rejected, this project only use Java code.

dwalend, 2020-12-03:
@jabrena - this one is probably a "go read ..." . (Also, possibly not appropriate for the ticket discussion.)

I cloned the repository and ran `gradle build` . The build hit errors starting the test code. What's the right way past this? Are there extra steps to set up the environment?

```
> Task :test
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:64)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:564)
        at java.instrument/sun.instrument.InstrumentationImpl.loadClassAndStartAgent(InstrumentationImpl.java:513)
        at java.instrument/sun.instrument.InstrumentationImpl.loadClassAndCallPremain(InstrumentationImpl.java:525)
Caused by: java.lang.RuntimeException: Class java/lang/UnknownError could not be instrumented.
        at org.jacoco.agent.rt.internal_28bab1d.core.runtime.ModifiedSystemClassRuntime.createFor(ModifiedSystemClassRuntime.java:140)
FATAL ERROR in native method: processing of -javaagent failed, processJavaStart failed
        at org.jacoco.agent.rt.internal_28bab1d.core.runtime.ModifiedSystemClassRuntime.createFor(ModifiedSystemClassRuntime.java:101)
        at org.jacoco.agent.rt.internal_28bab1d.PreMain.createRuntime(PreMain.java:55)
        at org.jacoco.agent.rt.internal_28bab1d.PreMain.premain(PreMain.java:47)
        ... 6 more
Caused by: java.lang.NoSuchFieldException: $jacocoAccess
        at java.base/java.lang.Class.getField(Class.java:2092)
        at org.jacoco.agent.rt.internal_28bab1d.core.runtime.ModifiedSystemClassRuntime.createFor(ModifiedSystemClassRuntime.java:138)
        ... 9 more
*** java.lang.instrument ASSERTION FAILED ***: "result" with message agent load/premain call failed at ./src/java.instrument/share/native/libinstrument/JPLISAgent.c line: 422
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:64)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
...
```

jabrena, 2020-12-03:
Eyyy

I will review tomorrow, the last builds were passing: 
https://travis-ci.org/github/ev3dev-lang-java/ev3dev-lang-java
https://travis-ci.org/github/ev3dev-lang-java/ev3dev-lang-java/builds

It should be complex to build again.

jabrena, 2020-12-03:
I am triggering the build on CI. 
The command is:

  - ./gradlew test checkstyleMain

dwalend, 2020-12-04:
TL/DR: I was able to speed up Sysfs.readFloat() by only a factor of 2 while preserving the outer API, a factor of 4 with a medium API change. 

Sysfs.readFloat(string) as written consistently ran in 24 ms. The final version reliably runs in 12 ms. Precomputing the path - an API change - drops it to 6 ms. But that's still just a factor of 4. (It is also more than an order of magnitude slower than Jakub's C++ approach.) (I did have to switch to 100 samples to get millisecond resolution, and would need to use the picosecond clock to go further. I'm not yet familiar with that microbenchmark framework, so the usual caveats apply.)

Preventing memory allocation seems to matter most. Perhaps there's a reasonable way to get the sensor value in binary  instead of ASCII, and translate the byte array directly to an Int or a Float value.

I think I'd expect about the same improvements with the Sysfs.writeString(string), but I did not explore that tonight.

I don't know that there's much more speed to find without changing how the code reads a sensor.  When I tried a first pass at keeping the file open I discovered that the gyro readings were not a replenishing stream; it only has one value in it.

I found a lot of boilerplate and some large-file assumptions down in the Java standard library that I could unwind. Here's the final version for the night:

```
    //Skipping much of the boilerplate between InputStream and Channels. 12 ms, but 6 ms with the static path!
    static float fetchSampleCustomChannel() {
        return Float.parseFloat(readStringCustomChannel());
    }

    private static final byte[] buffer = new byte[8];
    static final Path staticPath = Paths.get(pathString);
    public static String readStringCustomChannel() {
        try {
            try (InputStream in = customInputStream(staticPath)) {
                int n = in.read(buffer);
                //todo throw IOException for n = -1
                return new String(buffer, StandardCharsets.US_ASCII);
            }
        } catch (IOException e) {
            throw new RuntimeException("Problem reading path: " + pathString, e);
        }
    }

    static InputStream customInputStream(final Path path) throws IOException {
        ReadableByteChannel rbc = Files.newByteChannel(path);
        return Channels.newInputStream(rbc);
    }
```

jabrena, 2020-12-04:
Many thanks for the effort @dwalend 

I will test this weekend too

JakubVanek, 2020-12-04:
> Preventing memory allocation seems to matter most. Perhaps there's a reasonable way to get the sensor value in binary instead of ASCII, and translate the byte array directly to an int or a float.

There is the [`bin_data`](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensors.html#sysfs) attribute. The format is specified separately - via the `bin_data_format` attribute.

[227 more replies]

closed 2020-12-27 by jabrena