ev3dev-lang-java/ev3dev-lang-java #765: Improve the usage of Sysfs in the interaction with EV3Dev

jabrena, 2020-12-27:
In the previous Issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/748

We improved the performance with current architecture, the usage of `Sysfs`. `Sysfs` in the current form, is a stateless class with static methods. 

Ideas to improve in this iteration:

- [ ] Moving from static methods, to instance methods
- [ ] Add setSysfs to `Mock` Sysfs to improve testing

@dwalend, what other ideas, do you have?
@JakubVanek, what other ideas, do you have? 

In your opinion, what state is necessary to maintain in `Sysfs` and have some kind of concurrency protection?

dwalend, 2020-12-28:
I don't think we should try for concurrency safety or state inside Sysfs. I don't think we can do it correctly and do it well at the same time; I think we'd regret any choice internal to Sysfs. 

I think the best path forward is to step away from Sysfs, and start using DataChannelRereaders and (new) DataChannelRewriters directly in Battery, GenericMode and BaseRegulatedMotor (contains EV3LargeRegulatedMotor's interesting code). (That's what I did for my kid's basic control loop.) 

The nice thing about Battery is that it only needs 2 readers, zero writers. BaseRegulatedMotor needed 4 readers and 5 writers. The GyroSensor had 1 writer and 3 readers.

I should have some free evenings this week after the kids are in bed.

I could do a PR of DataChannelRereader in Battery - and make it Closable - as a first step if that's of interest. 

As a second step I can do a PR for a DataChannelRewriter.

jabrena, 2020-12-28:
Let’s try your idea. 
With Battery, we will explore the model.

Apart of the performance, the first goal in this issue, I am pretty interesting to introduce Mockito. In the project we added tests (One innovation in LEJOS ecosystem), but it is possible to improve.

dwalend, 2020-12-28:
For 
```
    public float getBatteryCurrent() {
        if (CURRENT_PLATFORM.equals(EV3DevPlatform.EV3BRICK)) {
            return Sysfs.readFloat(BATTERY_PATH + "/" + BATTERY_EV3 + "/" + CURRENT);
        } else {
            LOGGER.warn("This method is not available for {} & {}", EV3DevPlatform.PISTORMS, EV3DevPlatform.BRICKPI);
            return -1f;
        }
    }
```

Does that file path exist on every platform (so the code can open the file) or is it missing from PISTORMS and BRICKPI ?

dwalend, 2020-12-28:
Also, most of the codebase looks like it is after the "`IOException`s are annoying" revelation. Should I follow that pattern or throw IOExcpetion s?

dwalend, 2020-12-28:
For Mockito - slipping an alternate filesystem in place is pretty easy. Does Mockito add value to that approach? (I think it at least adds a little.)

jabrena, 2020-12-28:
Hi @dwalend,

> Does that file path exist on every platform (so the code can open the file) or is it missing from PISTORMS and BRICKPI ?

Exactly, this particular method is only available in EV3 Brick.

> Also, most of the codebase looks like it is after the "IOExceptions are annoying" revelation. Should I follow that pattern or throw IOExcpetion s?

Interesting point for a Debate. Currently I return a bad value and I advice in logs?
Maybe it is more strict to stop the program with a runtime exception and explain why.
What do you think?

> For Mockito - slipping an alternate filesystem in place is pretty easy. Does Mockito add value to that approach? (I think it at least adds a little.)

Mockito introduce an standard in the way to test software as a Best practice. In general, people forget `Testing phase` due to multiple factors and this is a good opportunity to to show the advantage of standard testing in Java but in robotics side.

dwalend, 2020-12-28:
`Battery` doesn't seem to get any value from inheriting from `EV3DevDevice` or `Power` . Should I pull those out now, or save it for a different change? (My preference is a different change to keep concerns separate, but it does mean something out there could be calling the various set*s and get*s on battery via promoting to public and reflecting.

PR to follow in a moment.

dwalend, 2020-12-28:
Here's the PR: https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/770

dwalend, 2020-12-28:
Hmm. Instead of using -1f as the bad value try out Float.NaN . I think in Java Exceptions are a better answer, but at least NaN will blow up if someone tries to multiply or add it. 

I'm coming back to Java from Scala. I don't miss the caught Exceptions like I thought I might. Java's the only language that has them I think. They put you into a contravariant world. But really if you're using the wrong robot then the Battery class isn't going to do what you'd hoped and there's no saving that part of the program.

dwalend, 2020-12-29:
Here's a draft DataChannelRewriter : https://github.com/dwalend/ev3dev-lang-java/pull/1/files

jabrena, 2020-12-29:
I have merged the changes. The new version maintains a good readability. I see your strategy, with this kind of inmutable objects, we will increase the performance.

> I think in Java Runtime Exceptions are a better answer, but at least NaN will blow up if someone tries to multiply or add it.

Yes, please, change the code to return a Runtime Exception.

> Battery doesn't seem to get any value from inheriting from EV3DevDevice or Power . 

For EV3DevDevice, it provides some objects like ev3DevProperties
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/master/src/main/java/ev3dev/hardware/EV3DevDevice.java

Power provides the contract for Battery.
https://github.com/ev3dev-lang-java/lejos-commons/blob/master/src/main/java/lejos/hardware/Power.java

To test the new class: DataChannelRewriter, I believe that the class EV3Led could be a good candidate:

- ev3dev.actuators.ev3.EV3Led

We should have a Unit Test for the class: DataChannelRereader.java

Can you try to Mock: DataChannelRereader from BatteryTest?

JakubVanek, 2020-12-29:
> inmutable objects

Unfortunately, DataChannelRereader is currently not immutable. The problem is with the cached [`byteBuffer`](https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/c69ceaba8e4860d52b42aaa7f495c5220c46c428/src/main/java/ev3dev/utils/DataChannelRereader.java#L20) member in that class. This also currently breaks multithreading safety - if two threads read from the same Battery instance (which they will, because it is a singleton that cannot be recreated), there is a possibility that two calls to `<rereader>.readString()` will get interleaved in a way that one or both receive incorrect data.

There is also a caveat with the cached FileChannel that I didn't realize earlier: https://stackoverflow.com/a/53361005/1561345 (state/current file pointer is visible across threads and could cause problems too). I am hoping this could be fixed by using a `read()` function with the offset directly specified instead of relying on `position()`.

The two above fields mean that methods using DataChannelRereader are currently safely callable only from one thread at a time.

My intuition is getting OK with that approach, but I'd add a warning to Javadoc for that method about that. For example, `strerror()` in POSIX/C is documented to be racy: https://man7.org/linux/man-pages/man3/strerror.3.html.

jabrena, 2020-12-29:
Hi @JakubVanek, 

the class is Immutable:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/sysfs_perf2/src/main/java/ev3dev/utils/DataChannelRereader.java

> In object-oriented and functional programming, an immutable object (unchangeable object) is an object whose state cannot be modified after it is created.

https://en.wikipedia.org/wiki/Immutable_object

other stuff is the `Thread Safety` but lets give time to @dwalend to review your comment. 

I put some links as a reference:

- https://stackoverflow.com/questions/11153583/options-to-make-javas-bytebuffer-thread-safe
- https://stackoverflow.com/questions/36974760/how-to-use-bytebuffer-properly-in-multiple-thread-environment

JakubVanek, 2020-12-29:
I think the non-thread-safe nature is intentional (thread safety decisions are shifted to the user). I just consider this unfortunate with singletons (only one thread should to access Battery at a time, or an additional locking wrapper has to be used by users). It may also change semantics of programs that were built with the assumption that multithreaded access is OK (this might be only an imagined problem though, I have no statistics about how the API is really used).

jabrena, 2020-12-29:
> I think the non-thread-safe nature is intentional (thread safety decisions are shifted to the user).

No, **ByteBuffer** has a known bug and It was solved in `Java 13`.
https://www.oracle.com/java/technologies/javase/13-relnote-issues.html
https://bugs.java.com/bugdatabase/view_bug.do?bug_id=JDK-5029431

JakubVanek, 2020-12-29:
Hmmm, regarding the `Power` interface - this is how it worked on real leJOS:
* `Battery` was likely only a compatibility layer, as it is a static class: https://sourceforge.net/p/lejos/ev3/code/ci/master/tree/ev3classes/src/lejos/hardware/Battery.java
* It deferred processing to `Power` implementations: https://sourceforge.net/p/lejos/ev3/code/ci/master/tree/ev3classes/src/lejos/hardware/Power.java
* For EV3, this was implemented by an internal `EV3Battery` class: https://sourceforge.net/p/lejos/ev3/code/ci/master/tree/ev3classes/src/lejos/internal/ev3/EV3Battery.java
* The voltage readout itself was implemented in `EV3AnalogPort` as a static function that reads the value a shared memory buffer provided by the kernel driver: https://sourceforge.net/p/lejos/ev3/code/ci/master/tree/ev3classes/src/lejos/internal/ev3/EV3AnalogPort.java#l343

JakubVanek, 2020-12-29:
> No, **ByteBuffer** has a known bug and It was solved in `Java 13`.
> https://www.oracle.com/java/technologies/javase/13-relnote-issues.html [...]

This could work, but as the issue notes, the intent is likely the following:

> Consider a use case where multiple threads are to operate on a large region of memory by dividing it up into segments, and having each thread operate on its own segment.

jabrena, 2020-12-29:
Sorry, @JakubVanek 

Please Stop doing it. 
This issue is not a chat.

JakubVanek, 2020-12-29:
I am sorry for posting another comment (this time it is related directly to the threading issue). One way out would be to declare the whole library as safe to access only from one thread (IIUC Swing does this similarly, it has its own UI thread and changes can be safely made only from within it). Under that assumption the changes in Battery & DataChannelRereader should work fine and indeed, improve performance. However, I'd consider this an API contract change and as such it should be documented and mentioned in the release notes.

jabrena, 2020-12-29:
Listen me @JakubVanek,

Don´t say sorry, your ideas are valuable in many many cases, but you need to measure your energy. 
If @dwalend is focusing on this issue, he needs to lead it.

Review the backlog, exist new good issues to develop from scratch new stuff.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues

Indeed, Power interface will have more relevance in the future.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/771

Enjoy

Good night

dwalend, 2020-12-30:
Do you two think it's worth making the read() and close() methods in DataChannelRereader synchronized ? (Anything Closable has at least two states.) That'd give the thread safety. The JVM is pretty good at optimizing synchronized away if only one thread uses a lock. 

I personally think it is reasonable to just mention that the calling code is responsible for concurrent access as a caveat in the docs that this code works like most of the rest of the Java ecosystem. Ultimately there's no reasonable way to prevent two separate processes from writing the same file. We could only ever fix so much of the problem adding locks to a library.

We could always measure performance and let that guide us.

Crazy day today. I'll slip these changes in - likely tomorrow evening.

jabrena, 2020-12-30:
Hi @dwalend,

I have uploaded a TestCase to review the concurrency for the new class: `DataChannelRereader.java` working in isolation and used by `Sysfs.java`.

https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/sysfs_perf2/src/test/java/ev3dev/utils/DataChannelRereaderConcurrencyTest.java

The testcases are:

```java
    /**
     * Writer1 -> pairs.txt
     * Reader1 <- pairs.txt
     * Reader2 <- pairs.txt
     *
     * Writer1 -> odds.txt
     * Reader1 <- odds.txt
     * Reader2 <- odds.txt
     */
```

This is case, is not very realistic because `ev3dev` will upgrade the file and later `Sysfs` will read the value. At the end the path is unique so the user should share in the program the same object but in this case, I am checking what happen if exist multiple readers.

When I execute multiple times, I receive the following exception in some case:

```
Caused by: java.io.IOException: Premature end of file 
	at ev3dev.utils.DataChannelRereader.readString(DataChannelRereader.java:59)
	... 14 common frames omitted
```

From the block:

``` java
            do {
                byteBuffer.clear();
                channel.position(0);
                n = channel.read(byteBuffer);
                if (n == -1) {
                    throw new IOException("Premature end of file ");
                }
            } while (n <= 0);
```

In the second scenario, it is more realistic:

```java
    /**
     * Writer1 -> pairs.txt
     * Reader1 <- pairs.txt
     *
     * Writer1 -> odds.txt
     * Reader1 <- odds.txt
     */
```

`ev3dev` will upgrade the file and later `Sysfs` will read the value. At the end the path is unique and only exist a reader but again, I receive the the exception: `"Premature end of file "`

Why appear that exception?

Juan Antonio

JakubVanek, 2020-12-30:
@jabrena I have quickly looked at `DataChannelRereaderConcurrencyTest` and it seems to me that the data race is accidentally avoided there. The reason is that each CompletableFuture has its own DataChannelRereader and thus the buffer is not shared. We cannot achieve this effective per-thread separation with Battery, as threads are not able to create their own Battery instances - they can only get the global singleton instance.

@dwalend Locking doesn't seem strictly necessary. If the buffer is moved to a thread local variable and the file channel is accessed atomically (i.e. only one function call with absolute positioning), the race can be avoided (I think, I may be wrong). I have provided https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/775 with an overview how that could look.

I haven't thought about concurrent writers, the issue I was thinking about is with concurrent readers. There is a vulnerable time window between the moment the NIO read is completed (and data are stored in the buffer) and the moment the data from the buffer are converted to a string. There is also another window between the moment the buffer limits are cleared and the rest of the code.

The scenario that is "haunting" me is that someone tries to read battery voltage from multiple threads (whatever may their motivation be) and they occasionally get an incorrect value and/or exception due to unparseable integer (empty string).

jabrena, 2020-12-30:
Hi @JakubVanek,

I tested in local your alternative, but I receive the same exception in both tests:

``` java
            int n = channel.read(buffer, 0);
            if (n == -1) {
                throw new IOException("Premature end of file " + path);
            }
```

```
Caused by: java.io.IOException: Premature end of file ./pairs.txt
	at ev3dev.utils.DataChannelRereader2.readString(DataChannelRereader2.java:96)
	... 14 common frames omitted
```

In the second test:

/**
     * Writer1 -> pairs.txt
     * Reader1 <- pairs.txt
     *
     * Writer1 -> odds.txt
     * Reader1 <- odds.txt
     */

exist one reader per file, not concurrent readers. Execute in local the test to review the results.

JakubVanek, 2020-12-30:
I tried running `./gradlew test` earlier and I cannot get this to fail (OpenJDK 11.0.9.1 from Ubuntu 18.04.5 x86_64). `./gradlew dockerTest` succeeds too. However, I'm wondering if the problem couldn't be having the file open for both reading and writing at the same time (I think Windows has issues with this). I have added https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/775/commits/346425f055b9a26ffb1a45fcffd546aeee10b441, that might fix the problem.

jabrena, 2020-12-30:
I go for lunch, later I will continue

jabrena, 2020-12-30:
@JakubVanek, Upgrade the Class: `DataChannelRereader2` with your proposals.

I have improved the tests, you should see that it fails in the CI. (I added some missing asserts)
https://github.com/ev3dev-lang-java/ev3dev-lang-java/actions/runs/452838235

With both Test classes, we can follow the cases.

I have just added a test to review the old Sysfs implementation and I noticed that I receive something similar:

```
Caused by: java.lang.IndexOutOfBoundsException: Index 0 out of bounds for length 0
	at java.base/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:64)
	at java.base/jdk.internal.util.Preconditions.outOfBoundsCheckIndex(Preconditions.java:70)
	at java.base/jdk.internal.util.Preconditions.checkIndex(Preconditions.java:248)
	at java.base/java.util.Objects.checkIndex(Objects.java:372)
	at java.base/java.util.ArrayList.get(ArrayList.java:458)
	at ev3dev.utils.SysfsOld.readString(SysfsOld.java:74)
	at ev3dev.utils.SysfsOld.readInteger(SysfsOld.java:94)
```

``` java
            if (existFile(path) && Files.isReadable(path)) {
                final String result = Files.readAllLines(path, Charset.forName("UTF-8")).get(0);
                if (log.isTraceEnabled()) {
                    log.trace("value: {}", result);
                }
                return result;
            }
```

> The scenario that is "haunting" me is that someone tries to read battery voltage from multiple threads (whatever may their motivation be) and they occasionally get an incorrect value and/or exception due to unparseable integer (empty string).

We need to know the reason that testing only with one Writer and one Reader, the behavior is the same.
For some reason exist something when you write and read (no Concurrent scenario) it fails.

Maybe it is something related to the way that open the file.

JakubVanek, 2020-12-30:
> * When Java write and Read concurrently, it is possible to have some issues. (Something to tackle, but this is not blocking because users doesn´t have the same object to do different actions, but should find a solution)

I found the issue - `-1` returned from FileChannel.read() actually means that the file is empty (it was - `cat <file>` returned nothing). Therefore the write in Sysfs class is not atomic and there is a moment when the file is empty. This seems consistent with how I'd imagine the syscall sequence:

1. The file is internally opened for writing by Java and truncated (something like `open(path, O_WRONLY | O_CREAT | O_TRUNC, 0664)` in C), thus setting contents to "". The window after this is where DataChannelRereader crashed.
2. Only then the integer/whatever is written to it, making the file contents valid again.

This is going to crash old Sysfs as well, because even if the read succeeds, parsing empty string as an integer will throw another exception.

This also shouldn't affect real sysfs because there is no file content to be truncated. The buffer that is returned in response to read is always freshly generated.

> * When Java Read concurrently, 0 issues with all implementations. (So, we can live with current DataChannelRereader.java)

This may not be the case. The problem is when one shared DataChannelRereader instance is used concurrently. In the concurrency test, multiple separate DataChannelRereaders are used (this works OK). The untested situation where one DataChannelRereader instance is used by two threads happens when these two threads use Battery (which is a singleton with a private constuctor) in parallel and this is where the race is.

> Do we have to raise and exception in this case?

This seems not to be the case, `-1` does not imply error here (in C/POSIX it would). Quoting from FileChannel.read docs ([link](https://docs.oracle.com/javase/7/docs/api/java/nio/channels/FileChannel.html#read(java.nio.ByteBuffer,%20long))):
> Returns: The number of bytes read, possibly zero, or -1 if the given position is greater than or equal to the file's current size

jabrena, 2020-12-30:
Good research, 

but how to fix the scenario for concurrent Write/Read?
What is your proposal?

In my opinion, the user doesn´t implement in the same program that scenario. 

I share a concurrent read old example:
https://www.youtube.com/watch?v=3hXTeK8lFMc

dwalend, 2020-12-30:
I guess my big opening question before examining the test code is: how do you want this to behave concurrently? I think there are two reasonable approaches: synchronized read() and close() or let the caller do the work. I'm fine with either.

The follow-up is: how would you like to handle times when the JVM catches the OS midway through writing a file while the JVM tries to read it? Returning an empty string from readString() seems reasonable in a very local sense, but will still blow up.

Maybe placing readFloat(), readInt(), maybe even readEnum() in DataChannelRereader would give us an easier way to handle what our code does for those "The JVM can't do this now, but try later" scenarios. I'll see what that looks like next.

dwalend, 2020-12-30:
From https://docs.oracle.com/en/java/javase/11/docs/api/java.base/java/nio/channels/ReadableByteChannel.html#read(java.nio.ByteBuffer)

Returns:
The number of bytes read, possibly zero, or -1 if the channel has reached end-of-stream

Shouldn't it return zero for an empty file, and only -1 if the file doesn't exist?

jabrena, 2020-12-30:
> I guess my big opening question before examining the test code is: how do you want this to behave concurrently? I think there are two reasonable approaches: synchronized read() and close() or let the caller do the work. I'm fine with either. I'm not sure of the value of writing tests of concurrent behavior until we define that expected behavior.

You commented that current approach is followed by the Java Ecosystem

> I personally think it is reasonable to just mention that the calling code is responsible for concurrent access as a caveat in the docs that this code works like most of the rest of the Java ecosystem. 

So, we can continue in current way, so _lets the caller do the work_.

> I'm not sure of the value of writing tests of concurrent behavior until we define that expected behavior.

Review if we have issues in the different scenarios and if some issues detected are scenarios where users use the objects.

jabrena, 2020-12-30:
I have opened the discussion tab:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/discussions/777

dwalend, 2020-12-30:
That whole do/while looks suspicious to me... I'm cleaning it out.

dwalend, 2020-12-30:
Pull request without the do/while at https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/778

jabrena, 2020-12-30:
Many thanks @dwalend,

Tomorrow, I will review in detail before new year's eve dinner

jabrena, 2020-12-30:
Many thanks @JakubVanek,

with the alternative based on `ThreadLocal`
https://docs.oracle.com/en/java/javase/11/docs/api/java.base/java/lang/ThreadLocal.html
https://stackoverflow.com/questions/817856/when-and-how-should-i-use-a-threadlocal-variable

JakubVanek, 2020-12-31:
I have some optimizations for DataChannelRewriter: https://github.com/ev3dev-lang-java/ev3dev-lang-java/commit/7bd73bcfb31a824ed6f4f6c253bd01881b593bb8

* At this stage, the shared buffer is not necessary. The allocations are already made inside `String.getBytes()`, the best thing to do may be to just wrap the returned array in a ByteBuffer. That being said, if custom integer serialization (bypassing strings) is added later, the buffer might be useful again.
* `position()` + `write()` can be folded together - this should be faster & thread-safer (although with `synchronized`, that doesn't matter much).
* `truncate()` can be avoided if the target is in real `/sys`, as it should have no effect there. The truncation is also moved to the beginning of the operation, mimicking the current behaviour of the Sysfs class. This needs some testing though.

As there are two independent file channel operations for each write, I have retained the `synchronized` keyword there. Otherwise, two threads may get interleaved in an unfortunate way and that would produce invalid file contents in emulation (outside of real /sys):

| n | thread A | thread B | file contents |
|-|-|-|-|
| 1 | truncate to zero size | | empty |
| 2 | | truncate to zero size | empty |
| 3 | | write "abcd" to pos 0 | `abcd` |
| 4 | write "xyz" to pos 0 | | `xyzd` |

The result would be "xyzd", which is neither "abcd" or "xyz".

(For comparison, this is what I think would happen in /sys if the truncation was there):
| n | thread A | thread B | kernel action |
|-|-|-|-|
| 1 | truncate to zero size | | noop |
| 2 | | truncate to zero size | noop |
| 3 | | write "abcd" to pos 0 | call attribute store function with "abcd" |
| 4 | write "xyz" to pos 0 | | call attribute store function with "xyz" |

On the other hand, I wouldn't be surprised if "xyzd" occured even with the current Sysfs class, so protecting against this scenario might not be worth it. After all, these are concurrent writes and they already behave strangely. Also, there indeed might be another process changing the file in parallel and there is no easy way of protecting against that (perhaps file locks, but it would have to be mandatory, not just advisory locking).

jabrena, 2020-12-31:
Returning in 1h from lunch

dwalend, 2020-12-31:
@JakubVanek - does "outside of real /sys" mean "outside of a real filesystem"  ? Or is there something else special on the ev3dev OS ?

If a file system will do - what do you two think of testing on a file system? It could be a first step in a long journey to an ev3dev simulator.

jabrena, 2020-12-31:
Hi @dwalend 

> It could be a first step in a long journey to an ev3dev simulator.

The idea is nice, but it is not realistic. Exist many many Mindstorms projects in the past, some of them with a commercial nature and all of them are Ko and it requiere huge mountains of hours. I try to advice you.

jabrena, 2020-12-31:
I am going to review all PR.

@JakubVanek, I don´t see this commit as a PR:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/commit/7bd73bcfb31a824ed6f4f6c253bd01881b593bb8

dwalend, 2020-12-31:
@jabrena - I took 7bd73bc as review commentary for my work.  Not sure if that was Jakub's intent.

jabrena, 2020-12-31:
@JakubVanek, we need to improve the way that you discuss the ideas.

- https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/775/files this PR touch the class from @dwalend, BLOCKED to Merge to compare in local
- https://github.com/ev3dev-lang-java/ev3dev-lang-java/commit/7bd73bcfb31a824ed6f4f6c253bd01881b593bb8 should be promoted as a PR in another class, not the Original class.

Current Status:

- DataChannelRereader supports Multiple Readers, Ok
- DataChannelRereader2 supports Multiple Readers, Ok
- SysfsOld supporst too, but it is so much slower in compare with previous one.

@dwalend In relation to the tests, review the rest of tests, you can "Recreate" a `ev3dev` fs to test the classes. It is not necessary to path DataChannelRereader with if blocks.

dwalend, 2020-12-31:
@jabrena - what's the right way to recreate the ev3dev file system? 

Battery gets its paths from EV3DevPropertyLoader.getEV3DevProperties() , which is pulling them in from a baked-in property file. I'd like to slip in my own properties file for testing there, but I don't see other code doing that, and don't see a way to do it without altering the EV3DevPropertyLoader API.

There's some kind of inheritance interaction between FakeBattery and BaseElement, but I don't see how that changes the paths that Battery will use.

Is there an example I should follow?

JakubVanek, 2020-12-31:
> @JakubVanek, we need to improve the way that you discuss the ideas.
> 
> * https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/775/files this PR touch the class from @dwalend, BLOCKED to Merge to compare in local

This PR should be removed, it uses `ThreadLocal<>` which was rejected. However, it also contains `DataChannelRereaderRaceTest` which shows the problematic behavior solved by adding `synchronized` to DataChannelRereader. I can rebase & change the PR to only include that.

> * [7bd73bc](https://github.com/ev3dev-lang-java/ev3dev-lang-java/commit/7bd73bcfb31a824ed6f4f6c253bd01881b593bb8) should be promoted as a PR in another class, not reusing DataChannelRereader

I can create a PR from it, in its current state it was meant as a sample implementation of the ideas in that comment.

jabrena, 2020-12-31:
@dwalend  SysfsTest.java should be fine

If you feel blocked, upload in a PR the tests and I will help tomorrow. :)

Once we have DataChannelRereader stable and DataChannelRewriter too, we will go faster.

jabrena, 2020-12-31:
@JakubVanek that version was not rejected. I wanted to have more options.

> This PR should be removed, it uses ThreadLocal<> which was rejected.

but it is true that ThreadLocal<> protect Bytebuffer in a different way but in order to merge, you need to reset the class DataChannelRereader

I would like to compare and observe with `Visualvm` the impact with ThreadLocal

Now, we notice that new versions improve the performance.

jabrena, 2020-12-31:
For today, is Ok, have a good dinner with the family in USA & Cech Republic

I have to say that this journey is pretty nice and it is not easy.
You should feel happy

jabrena, 2021-01-03:
Testing in the Brick both alternatives with the concurrent scenario for Readers.

- [ ] DataChannelRereader
- [ ] DataChannelRereader2
- [ ] Sysfs (Old)

Criterias:

- [ ] Performance
- [ ] Memory

Tomorrow, I will share the results.

jabrena, 2021-01-05:
If you like, tomorrow I will clean the branch to continue with the part about the `Writer`

JakubVanek, 2021-01-05:
> It is clear, that the usage of DataChannelRereader improve the performance in near of 300% in compare with the usage of Sysfs.

The error ranges are once again huge. For example, peeking at the individual data values shows that the first sample in Throughput is many times larger than the other values. JMH is capable of measuring more accurate results; wouldn't it be possible to adjust the benchmark parameters?

dwalend, 2021-01-05:
I've got a first pull request at https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/781  . What do you two think of it?

jabrena, 2021-01-06:
Hi @dwalend, I add waiting for the Unit Test for DataChannelRereader to continue.

jabrena, 2021-01-06:
Hi @JakubVanek, 

In the performance comparison, you notice that the usage of `DataChannelRereader` offers a better performance and reviewing the solution, the alternative using `ThreadLocal<>` solves the same issue with Concurrent readers but in terms of memory, every object using a threadlocal, forces to have a specific memory area to maintain the Thread Safety so this is the reason that the first solution is better. Better performance, less memory usage and better code understanding. In general, `ThreadLocal` is not the first option but I felt curious to see the performance results but I have to say that the discussion improved the solution for `DataChannelRereader`.

JakubVanek, 2021-01-06:
> In the performance comparison, you notice that the usage of `DataChannelRereader` offers a better performance and reviewing the solution, the alternative using `ThreadLocal<>` solves the same issue [...]

I agree that the synchronized DataChannelRereader solution is better. However, I still think that performance conclusions shouldn't be drawn from the graph posted above due to the error margins that the data has. For example the first outlier sample result shouldn't be counted to the averages; it should be processed separately.

jabrena, 2021-01-06:
Hi @JakubVanek,

it is interesting. If you are interested in that field, please create a new issue about "Best Practices Performance Testing" and use that issue to add new references and examples using tools from Java Space. It could be pretty valuable to be reused in the future.

JakubVanek, 2021-01-06:
> it is interesting. If you are interested in that field, please create a new issue about "Best Practices Performance Testing" and use that issue to add new references and examples using tools from Java Space. It could be pretty valuable to be reused in the future.

I feel this is just following common sense, I outlined some of the parts in https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/748#issuecomment-751272222 and an example of better error ranges (with JMH) is https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/748#issuecomment-743974830. I have since deleted the forked repository and recreated it, but I can reproduce the configuration if necessary.

jabrena, 2021-01-06:
Hi @JakubVanek,

It is not necessary.

dwalend, 2021-01-06:
> Hi @dwalend, I am waiting for the Unit Test for DataChannelRereader to continue.

Will https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/781/files#diff-c0e0309e22ce34fb394a33b423bdafcf7785541728b2b6b6ef21490144a47b38 do? I'm afraid I might have lost the thread.

[24 more replies]