ev3dev-lang-java/ev3dev-lang-java #779: first time user gotchas using gradle template

maxandersen, 2020-12-30:
Got ev3dev installed etc. java is there and all looks good, but had a few issues with the gradle template

1) host is hard coded to 192.168.1.151 in `config.gradle` - should this not be `ev3dev` so it "just works" ? 

2) `gradle clean deployAndBrickRun` result in error as it tries to run "/home/robot/template_project_gradle-2.6.0-all.jar" but what is actually deployed is named "/home/robot/template_project_gradle-2.6.0.jar"

jabrena, 2020-12-30:
Hi Max, 

Tomorrow I will fix it

maxandersen, 2020-12-30:
well, all the docs until this point uses `ev3dev` or `ev3dev.local` thus I think you would have bigger issues before that :)

jabrena, 2020-12-31:
Hi Max,

I have fixed the template project with a basic configuration.
https://github.com/ev3dev-lang-java/template-project-gradle

I have just tested.

Merry Christmas.

closed 2020-12-31 by jabrena