ev3dev-lang-java/ev3dev-lang-java #786: Button implementation does not match JavaDoc

dwalend, 2021-01-04:
I'm pretty sure this will never return a value that can be decoded to determine which button was pushed:

```
    /**
     * Waits for some button to be pressed. If a button is already pressed, it
     * must be released and pressed again.
     *
     * @param timeout The maximum number of milliseconds to wait
     * @return the ID of the button that has been pressed or in rare cases a
     *     bitmask of button IDs, 0 if the given timeout is reached
     */
    public static int waitForAnyPress(int timeout) {
        ALL.waitForPress();
        return 0;//keys.waitForAnyPress(timeout);
    }

    /**
     * Waits for some button to be pressed. If a button is already pressed, it
     * must be released and pressed again.
     *
     * @return the ID of the button that has been pressed or in rare cases a bitmask of button IDs
     */
    public static int waitForAnyPress() {
        return waitForAnyPress(0);
    }
    ```

dwalend, 2021-01-04:
Not too hard to get what I needed via a KeyListener.

JakubVanek, 2021-01-05:
This is how it worked on leJOS: https://sourceforge.net/p/lejos/ev3/code/ci/master/tree/ev3classes/src/lejos/internal/ev3/EV3Keys.java

dwalend, 2021-01-05:
You sure you want to close this? Maybe at least update the javadoc to say "this method always returns 0" at a minimum.

JakubVanek, 2021-01-05:
My post about leJOS was in support of @dwalend's assessment of the situation. However, I now realized that it can be understood that leJOS behaviour was the same. This is actually the opposite of what I meant; I wanted to show that leJOS implementation was more sophisticated and behaved correctly (= as documented). The reason why I keep posting links to leJOS sources here and there is that it adds additional context about how some things came to be in this library.

JakubVanek, 2021-01-05:
What's more, even the `timeout` argument is ignored. This can create unfortunate effects/bugs in user programs if this function is incorporated into a mainloop as a waiting element.

jabrena, 2021-01-05:
Hi @dwalend,

Sorry, I read the issue and I supposed that it was more about to clarify the usage of Buttons.

In LeJOS for EV3, the main library worked with a runtime coupled and this project interact with a third party integration, so it is not possible to have a 100% behaviour. 

We could upgrade the 100% the Javadocs to add that the comment: "this method always returns 0"

JakubVanek, 2021-01-05:
> In LeJOS for EV3, the main library worked with a runtime coupled and this project interacts with a third party system, so it is not possible to have the "exact" behaviour in this kind of integration.

I don't believe this is the case, it should certainly be possible. If I remember correctly, the leJOS code polls a bitmask of buttons provided by the LEGO kernel. Yes, ev3dev does not give the bitmask to us "prepackaged", but we can still build the bitmask by polling the individual buttons.

dwalend, 2021-01-05:
There are a few ways to implement it. One way is with KeyListeners on the existing infrastructure. Another would be to tap into EV3Key's keyEventReader thread - which is hard-looping on a file and applying a bit mask:

```
        if (keyState == STATE_KEY_UP) {
            keyBits ^= keyType.bitMask;   // clearing the key bit
            broadcastToListeners(keyType, KeyListener::keyReleased);
        } else if (keyState == STATE_KEY_DOWN) {
            keyBits |= keyType.bitMask;   // setting the key bit
            broadcastToListeners(keyType, KeyListener::keyPressed);
``` 

The existing API around keyEventReader returns  void. Opening that up to return something useful should be a possible non-breaking change. What do you think @jabrena  ?

Also, any performance concerns about keyEventReader hard-looping?

JakubVanek, 2021-01-05:
> Also, do either of you have any performance concerns about keyEventReader hard-looping?

I'm in the process of studying the code, so I don't know the answer yet for sure, but it's possible that the device read is actually a blocking operation here (it only returns when an event happens).

dwalend, 2021-01-05:
> it's possible that the device read is actually a blocking operation here (it only returns when an event happens).

That would be pretty fantastic. Would events queue up waiting to be processed?

Is that possible to test with something like 
```
cat /dev/input/by-path/platform-gpio_keys-event
```
?

JakubVanek, 2021-01-05:
Comments indicate that is the case:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/d8320a391f9ecba756814b12fce903f40c49da9c/src/main/java/ev3dev/sensors/EV3Key.java#L49

The structure that gets read should be the following one: https://www.kernel.org/doc/html/latest/input/input.html#event-interface . Its size should be the 16 bytes that get read in that loop.

dwalend, 2021-01-05:
"You can use blocking and nonblocking reads, and also select() on the /dev/input/eventX " - nice!

At risk of entangling with our larger effort - it's not using Sysfs. Is it worth replacing the 
```
            try (final DataInputStream in = new DataInputStream(new FileInputStream(SYSTEM_EVENT_PATH))) {
``` ?

JakubVanek, 2021-01-05:
> That would be pretty fantastic. Would events queue up waiting to be processed?

Oh, I now see why :smile: We could theoretically avoid touching different threads in the `waitForAnyPress()` function and just directly read the event device in the user's thread. I hope that the event queue is maintained for each file descriptor separarely and not for the whole process.

However, I see one possible issue with this: missed events (if the files are not kept open, but that could bring different issues). leJOS solves this by maintaining the last read state in a thread-unsafe static variable (see quote below). This should guarantee that if a key gets pressed (and stays pressed) when the user's code is doing something different, they'll get notified of the button press when they call `waitFor*`. This may not work with the above solution. However, this flaw might be considered "insignificant enough" to be ignored. I've realized now that C4EV3 has this bug too.

> Thread Safety: All methods that return buttons states can be used safely from multiple threads, even while a call to one of the waitFor* methods active. However, it is not safe to invoke waitFor* methods in parallel from different threads. This includes the waitFor* methods of different buttons. For example Button.ENTER.waitForPress() must not be invoked in parallel to Button.ESCAPE.waitForPress() or the static Button.waitForAnyEvent(). In case this is needed, it is strongly recommended that you write your own Thread, which waits for button events and dispatches the events to anyone who's interested.

Regarding Sysfs: We're essentially reading C structures from a file here, and so Sysfs/DataChannelRereader seem unsuited for this (I think that they focus more on string-based files). The file is also already "cached" in the key thread, as it is opened only once over the thread's lifetime.

dwalend, 2021-01-05:
> Oh, I now see why 😄 We could theoretically avoid touching different threads in the `waitForAnyPress()` function and just directly read the event device in the user's thread.

Exactly. And that gets us one fewer thread deeper in the concurrency puzzle.

> Regarding Sysfs: We're essentially reading C structures from a file here, and so Sysfs/DataChannelRereader seem unsuited for this (I think that they focus more on string-based files). The file is also already "cached" in the key thread, as it is opened only once over the thread's lifetime.

I'm pretty comfortable adding other get*() methods beyond getString(). My exploratory branch has getIntFromAscii() to just keep the Integer.parseInt()s from cluttering things up. I think I might have that in Battery, will definitely have it for reading values from the various motors. Sensors with GenericMode need getFloatFromAscii() . getByte() and other primitives seems a natural extension. 

I don't think we'd see a big performance gain for Ev3Key s. The main benefit would be having that layer of code all work the same way. FileInputStream seems pretty innocuous.

jabrena, 2021-01-06:
Hi @dwalend,

In this repository some guys has contributed in the past with his best intention and knowledge, @JakubVanek included.
I agree with you that everything is possible to improve and maybe do a refactoring but again, this is a community project that everyone play/collaborate in his spare time.

Honestly, I am happy with current set of examples for the usage of Button and they are working, you can find here a set of examples.
https://github.com/ev3dev-lang-java/examples/tree/master/ev3dev-lang-java/src/main/java/ev3dev/sensors

That examples match with the behaviour from LeJOS, you could find more on Internet.

- https://github.com/ev3dev-lang-java/examples/blob/master/ev3dev-lang-java/src/main/java/ev3dev/sensors/ButtonExample.java
- https://github.com/ev3dev-lang-java/examples/blob/master/ev3dev-lang-java/src/main/java/ev3dev/sensors/ButtonExample2.java
- https://github.com/ev3dev-lang-java/examples/blob/master/ev3dev-lang-java/src/main/java/ev3dev/sensors/ButtonListenersExample.java

In perspective, I didn´t find in the past many interest about this specific sensor in all LeJOS generations, RCX, NXT & EV3 so **the priority to change any line code is really low**. If you like, we could change the Javadoc, but not the implementation in the short term. 

Exist many opportunities here:

- https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/771 LocalEV3
- https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/772 RemoteEV3
- https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/765 Current active issue

The project evolve adding new features, not refactoring every time.

Juan Antonio

dwalend, 2021-01-06:
In those examples nothing ever uses the int returned from `waitForAnyPress()` . Any code that relies on that return value is broken.

If there's no appetite to fix the code to match the contract in the javadoc - then the easiest fix would be to change the javadoc to describe what really happens in the method, `Deprecate` the methods, and direct people to use KeyListeners instead.

Any interest in a PR for that sort of change?

jabrena, 2021-01-06:
Hi @dwalend,

Go ahead with a change in the Javadocs, but I am not sure to deprecate the methods.

JakubVanek, 2021-01-06:
Yep, for example leJOS menu (https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/774) uses it: https://sourceforge.net/p/lejos/ev3/code/ci/master/tree/EV3Menu/src/lejos/ev3/startup/ExceptionHandler.java#l35 (found via GitHub code search).

jabrena, 2021-01-06:
User programs @JakubVanek

JakubVanek, 2021-01-06:
User program: https://github.com/jonasschultheiss/m121/blob/170f095c6a82c9c3a9ec5230909e2fd682a2f42b/src/Task.java#L211

jabrena, 2021-01-06:
Again, it is not a common case.

dwalend, 2021-01-06:
@jabrena - That method only returns 0. It's never going to be a common case with that bug. Anyone trying thee method will see it doesn't work, then either switch to KeyListeners, give up on using buttons for control, or give up on the library. They are extremely unlikely to commit something broken to a publicly searchable project.

jabrena, 2021-01-06:
It is acceptable @dwalend 

But, as I said, currently it is not the higher priority this particular case and indeed, it is an uncommon case in general.

jabrena, 2021-01-07:
Clarified

closed 2021-01-07 by jabrena