ev3dev-lang-java/ev3dev-lang-java #787: Using the LCD requires sudo

dwalend, 2021-01-04:
Using the LCD requires sudo .. 

```
Debian stretch on LEGO MINDSTORMS EV3!
Last login: Mon Jan  4 01:58:49 2021 from fe80::99:40ae:6557:e588%wlx08beac054001
robot@ev3dev:~$ java -jar Replay.jar
[main] INFO ev3dev.actuators.LCDStretch - Instancing LCD for Stretch
[sudo] password for robot: 
[main] INFO ev3dev.actuators.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: B
[main] INFO ev3dev.actuators.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: C
```

dwalend, 2021-01-04:
@JakubVanek - is there something intrinsic to the way the code works that it needs root access?

JakubVanek, 2021-01-04:
Yes and no. Running Java from console like this requires us to stop the brick menu via `sudo systemctl stop brickman` ("stolen display mode"). This is because we don't support switching to separate display manually. However, if one runs `brickrun -- java -jar LcdTest.jar`, then display access is granted to the Java process by brickrun ("owned display mode"). In that mode sudo shouldn't be necessary. Similar thing happens when java programs are launched viafrom brickman (possible via shell scripts, was built into https://github.com/ev3dev-lang-java/gradle-plugin)

dwalend, 2021-01-04:
That sounds like what I was hoping for. I'll explore tonight. Thanks!

dwalend, 2021-01-04:
brickman seems to do the trick - next I'll get the flashing to stop.

Any downside to using it?

JakubVanek, 2021-01-05:
There is a peculiarity that has to be kept in mind. As specified in brickrun [docs](https://github.com/ev3dev/brickrun/blob/ev3dev-stretch/doc/brickrun.rst), unless `-r`/`--redirect` arguments are passed, then stdin/stdout of the program will be routed to the brick display/keyboard, not to the SSH console (SSH=redirected). However, stderr will always go to the SSH console. You can also see text from stdout on the display only when graphical mode is disabled, this stops being the case when a `GraphicsLCD` is constructed (`switchToGraphicsMode()`/`switchToTextMode()`).

closed 2021-01-05 by jabrena