ev3dev-lang-java/ev3dev-lang-java #791: Display architecture - increase test coverage

JakubVanek, 2021-01-09:
Current unit testing solution for the display stack is inadequate. A revision of the current testing code is necessary.

jabrena, 2021-01-09:
Yep, it is something missing in the library. When we added the nice EV3 Display support, we forget adding it.

Currently, we have this status:

[image]

[image]

- LCDStretch 
- LCD
- LCDStretch.Flusher

[image]

[image]

With that contribution, we could enable again the following configuration to reach 80% of code coverage:

```groovy
jacocoTestCoverageVerification {
    violationRules {
        rule {
            limit {
                //minimum = 0.75 // implement more tests
                minimum = 0.2
            }
        }
    }
}
```

JakubVanek, 2021-01-09:
Given that LCDJessie is implemented as a separate class, it seems that the BitFramebuffer & FramebufferProvider paths are mostly redundant. Removing them might reduce the amount of code that needs testing.

jabrena, 2021-01-09:
Create a PR with this small change. I will merge directly to master.