ev3dev-lang-java/ev3dev-lang-java #800: Problem setting up VSCODE as an IDE Java programming developpment for EV3DEV

ALPHALAB8067, 2021-06-28:
Hello everyone, I am so glad I stumbled upon this chat room. I have been desperately searching for someone to help me clarify a few things about the workings of Ev3Dev. To make the story short, I have been programming my EV3 brick using Eclipse and running the Lejos program and I would like to know if I could replicate the same using VSCODE and the EV3Dev Java program. So far I have installed the EV3Dev on my brick but I am really struggling with integrating the Java in VSCODE while using the Lejos librairies. All the tutorials available cover the use of Pthon and the ones covering the Java language are very scarce and unclear. On the latter, is it all the LeJos librairies that are available to the EV3Dev or just partially? How far can I get with EV3Dev ( with Lejos and Eclipse , I was able to run complex program , controlling it remotely, pid and more) I would really want to know if I could find the same smooth experience of using Lejos integrated to the IDE Eclipse where I could connect/disconnect and upload files to my EV3 brick effortlessy. As the time of me writing this message indicates, I am desperate to figure this out and hopefully someone couldhelp me with my request.  Thanks!

jabrena, 2021-07-18:
Hi @ALPHALAB8067, 

I will try to reply your questions:

1. VSCODE:
you could use VSCODE without any issue. Take a look the following samples:
https://github.com/ev3dev-lang-java/template-project-gradle
https://github.com/ev3dev-lang-java/template-project-maven

Using VSCODE, you could open a terminal window and interact with gradle or maven.

2. Is this project a complete port of all lejos packages?

No, it is a partial port, but many people use this library and have fun.

I hope that my answer help you and sorry by the delay.

Cheers

Juan Antonio

jabrena, 2021-08-29:
No feedback

ALPHALAB8067, 2021-08-30:
Good evening  jabrena, 
I am so sorry ( where are my manners?) for not having replied to your response. I did try  your recommendations  but without any success. I loaded up the Gradle sample  suggested by you in VSCODE which recognized the EV3 , but I was not ever able to download the program to the EV3  and run the code successfuly- I kind have given up  because no matter how many times I tried, nothing worked .  Like I said in my first post, I used in to program my EV3 running Java LeJos in Eclipse and it worked like charm. I am not too familiar with using Gradle or Maven and I must have done something wrong or simply forgotten a step. Is there a way that I could get a step-by-step (literally from Ato Z - beginner's perspective) of all the process needed to run the Java program in my EV3 using the EV3Dev Gradle example. I've looked everywhere (forums, Youtube, web) and every tutorial seems to be directed towards the Python programming languages.  I honestly believe that the EV3DEV is a much more powerful solution but a little less intuitive to use ( from my experience ) 

I don't like to bother people but I would really appreciate it if you could help me and I'll make sure to do a follow-up this time.

Thanks a lot and talk to you soon.

flemmii, 2023-01-22:
Hi, I don't know how to compile the program in VSCode using the docker extension, can you please help me? I already have the ev3dev image but know how to use it. A dockerfile or whatever needed would help.

jabrena, 2023-01-23:
Oki, I will help you

flemmii, 2023-01-24:
[wro.zip](https://github.com/ev3dev-lang-java/ev3dev-lang-java/files/10488644/wro.zip)
Here is my code if you need it, it's not finished yet, if you have any questions I will try to respond as soon as possible. And thank you for helping :)

flemmii, 2023-02-05:
Hi,
I dont know if you forgot this conversation, but I would be happy about an answer :)

jabrena, 2023-02-11:
Hi @flemmii, 

Thie project doesn´t offer support for any particular Java IDE, but in any case, I will reply you.

In VSCODE, you could open a terminal and execute a compilation process and it is working:

```
gradle clean build
```

What is your problem?

Juan Antonio

flemmii, 2023-02-12:
I cant execute my Java or class files on the ev3 and I dont know I can manage, that they are executable on the ev3?

flemmii, 2023-02-12:
[wro.zip](https://github.com/ev3dev-lang-java/ev3dev-lang-java/files/10717024/wro.zip)
This is my newest version of my code, I found out I must to install java on the ev3 and I have done that now, but I can still not execute the .java or .class files. How can I execute them, what do I need to do?

jabrena, 2023-02-12:
For that purpose, it was created this repo: https://github.com/ev3dev-lang-java/template-project-gradle

flemmii, 2023-02-12:
I have done all steps now, but at Step 2.2: ./gradlew testConnection the error: main class was not found or could not get loaded (maybe not the exact error in english, because I translated it). I changed nothing in the program and I have installed the gradle for java extension.

jabrena, 2023-02-12:
Can you use the template project and later reuse it for your project? That repo has some examples that you could use in the brick and understand the commands

flemmii, 2023-02-13:
I use the template project, but the error appears there.

jabrena, 2023-02-13:
Are you updated this file?
https://github.com/ev3dev-lang-java/template-project-gradle/blob/master/config.gradle

flemmii, 2023-02-13:
The file automatically updates when I open it right? If yes then it is updated. And at the Manifest I don't need to change something in the test program, just later in my own program?

jabrena, 2023-02-13:
Are you able to open a ssh connection with your terminal?
https://www.ev3dev.org/docs/tutorials/connecting-to-ev3dev-with-ssh/

flemmii, 2023-02-13:
yes, I am

flemmii, 2023-02-13:
I think it's Kernel: 4.14.117-ev3dev-2.3.5-ev3

jabrena, 2023-02-13:
I will find a brick and I will test the repo.
Tomorrow, I will reply you. This way worked for years 🧐

flemmii, 2023-02-13:
Ok thank you very much :)

jabrena, 2023-02-14:
Today, I have a long day @flemmii . Tomorrow I will reply

flemmii, 2023-02-14:
No problem, take your time

flemmii, 2023-02-15:
Hi, IT WORKS!!! I can download and execute the test as well as my own project now I have only a few questions: 
Is it normal, that if you execute a class for the first time that it needs like 2 seconds? 
At the moment I have one main program and a few sub-test programs, how can I execute the test programs without needing to change the manifest? (At the moment I execute the main program via a .sh file)
I have a problem with loading the sensors, I get this error:
ev3dev#1|Exception in thread "main" java.lang.ExceptionInInitializerError
ev3dev#1|       at WRO.Test.MotorTest.main(MotorTest.java:28)
ev3dev#1|Caused by: java.lang.RuntimeException: The path doesn't exist: /sys/class/lego-sensor
ev3dev#1|       at ev3dev.utils.Sysfs.getElements(Sysfs.java:107)
ev3dev#1|       at ev3dev.hardware.EV3DevDevice.detect(EV3DevDevice.java:57)
ev3dev#1|       at ev3dev.hardware.EV3DevSensorDevice.<init>(EV3DevSensorDevice.java:40)
ev3dev#1|       at ev3dev.sensors.BaseSensor.<init>(BaseSensor.java:22)
ev3dev#1|       at ev3dev.sensors.ev3.EV3ColorSensor.<init>(EV3ColorSensor.java:48)
ev3dev#1|       at WRO.functions.Sensors.<clinit>(Sensors.java:15)
ev3dev#1|       ... 1 more
How can I fix that?

jabrena, 2023-02-15:
Congrats!

The hardware is from 2010 and it is not very powerful but using other full-featured language, the performance is the same. I had an idea to improve that startup time but is a trick over the JVM. Every time that you run your program, you run a JVM instance.

Nope, you need to touch the manifest, indeed this is the way that Java works, nothing more & nothing else. If you continue playing with this project, you will learn more stuff about real world. Think in the brik like a slow computer.

in relation to the sensors, plug the sensor and code the program pointing to that sensor and it will work. Please, not unplug the sensor meanwhile the program is running.

what was the solution to run your program?

Good night from Madrid

Juan Antonio

jabrena, 2023-02-15:
@ALPHALAB8067, apologies with my long absence, I was very sad with latest actions from LEGO.

flemmii, 2023-02-16:
Ok that's sad, my trick at the moment is just to make an object of every class at the beginning because then I have the waiting time for every class all at the start

Ok, then I will look for another way, maybe I will find one

So all sensors need to be plugged in in order that the program can run? Because the motors don't need to be plugged in.

A friend of mine also tried it and at his EV3 brick it worked. 
After I tried it with his ev3 and my PC it worked too. So I think I just did something wrong while installing everything on the ev3 brick.

Thank you very much for your help I am very glad that I now can program with java :)

jabrena, 2023-02-16:
I am going to close this issue, enjoy

closed 2023-02-16 by jabrena