ev3dev-lang-java/ev3dev-lang-java #805: Adopting ev3dev-lang-java at London Uni

cyclingProfessor, 2021-08-18:
### Discussed in https://github.com/ev3dev-lang-java/ev3dev-lang-java/discussions/804

<div type='discussions-op-text'>

<sup>Originally posted by **cyclingProfessor** August 17, 2021</sup>
Hi.
ev3dev-lang-java and particularly the navigation stack - Brilliant work!  Thanks :-)
I have taught robotics for many years at London Uni (RHUL college) to CS first years.  We have always used Lejos, but ev3dev-lang-java is so nice that we are moving across.   Motor sychronization (not speed sync but starting and stopping at the same time) is the issue and we have been following your discussions.  We would really find it so useful to be able ask questions and share answers if you have time.  We are trying to determine where the delay in the system is.  Maybe it is in the Linux kernel?  But we think that it is in the Java side?
We are working on three things:
1) Motor synchronization
2) Adding support to more legacy sensors (because we have loads of these)
3) Making my two helper/example apps (one Android and one for Mapping) work seamlessly with ev3dev-lang-java

We would love to be able to ask you questions and maybe even help with answers.</div>

cyclingProfessor, 2021-08-28:
I will ask the guys to run up that project.  The resampling for the particle filter in Lejos is not correct - but not fatally flawed.

jabrena, 2021-08-29:
> "The resampling for the particle filter in Lejos is not correct" 

It is interesting, that algorithm was implemented for NXT and the same implementation was used in EV3 generation.
In relation to the sync features, I would like to review when your team has a version stable.

nicopao, 2021-08-31:
> It could be possible.

Great! Can you please send your discord tag so I can add you? (or provide some contact info where I can send the discord server invitation) Thank you

jabrena, 2023-02-15:
Closing the issue. The Users doesn’t have activity in github

closed 2023-02-15 by jabrena