ev3dev-lang-java/ev3dev-lang-java #807: How to reset local angle?

mmaxus35, 2021-08-29:
Dear All,

Please beg my naiveness.

I should be able to reset the local angle. I use an elevator system and the initial position of my elevator changes when I restart my system. Therefore, I send the elevator to the top floor and at the top floor the reference/local angle has to be equal to "zero".  

Is there a way to do it?  I could not see any function in the documents.

Thank you.

jabrena, 2021-08-29:
Hi,

Can you upgrade the issue with a picture of your project?

mmaxus35, 2021-08-29:
[image]

The elevator can be started to operate at any point of Middle level. So degree initial angle with respect to top-level changes. Therefore elevator should be sent to the top level. When it reaches there, the local/starting angle of the motor should be 0. In other words, if I sent the elevator to the top level, getPosition should be equal to zero. 

Note: Elevator cannot be started/initialized at the top level once the system is powered on.

I think it helps.

mrkrizic, 2022-11-07:
I think what you want to do is to move the elevator until the top touch sensor detects the elevator and stop the motor. After this you should be able to call resetTachoCount() to set the rotation to 0.

I hope this helps